coolsnowwolf/lede #8616: 5.15.89版本iptables不可用

zopargr, 2022-01-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：mwan3无法正确跟踪，提示没有iptables相关，手动安装mwan3到最新的2.10.13-1
日志有关于iptables的报错
### (2) 路由器型号和固件版本

A：架构X86 版本5.15.89

### (3) 详细日志

A：Thu Jan  6 00:02:39 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:02:39 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.
Thu Jan  6 00:02:44 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:02:44 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.
Thu Jan  6 00:02:49 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:02:49 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.
Thu Jan  6 00:02:54 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:02:54 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.
Thu Jan  6 00:02:59 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:02:59 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.
Thu Jan  6 00:03:05 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:03:05 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.
Thu Jan  6 00:03:10 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:03:10 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.
Thu Jan  6 00:03:15 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:03:15 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.
Thu Jan  6 00:03:20 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:03:20 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.
Thu Jan  6 00:03:25 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:03:25 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.
Thu Jan  6 00:03:30 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:03:30 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.
Thu Jan  6 00:03:34 2022 daemon.err uhttpd[3702]: iptables v1.8.7 (legacy): can't initialize iptables table `mangle': Table does not exist (do you need to insmod?)
Thu Jan  6 00:03:34 2022 daemon.err uhttpd[3702]: Perhaps iptables or your kernel needs to be upgraded.

Beginner-Go, 2022-01-05:
5.15.89？

zopargr, 2022-01-05:
说错了，是5.15.12，还活在昨天5.10.89那一天呢🌝🌝

发自我的iPhone

Beginner-Go, 2022-01-05:
是不是伴随 b站 抖音加载不出内容？

zopargr, 2022-01-05:
> 是不是伴随 b站 抖音加载不出内容？

我这边正常的

Beginner-Go, 2022-01-05:
[把这个patch删除后再编译](https://github.com/coolsnowwolf/lede/commit/e105008c7dec13ce8a037141a8506916ef6b1689)

WYC-2020, 2022-01-06:
https://github.com/WYC-2020/lede/commit/e891529e169aece14de0154669935127a7e6f130
用我这个补丁替换原来哪个补丁，再试下 @zopargr

zopargr, 2022-01-06:
> [WYC-2020/lede@e891529](https://github.com/WYC-2020/lede/commit/e891529e169aece14de0154669935127a7e6f130) 用我这个补丁替换原来哪个补丁，再试下 @zopargr

你好，亲测无效

WYC-2020, 2022-01-06:
我这边一切正常呢 @zopargr

alexliu269, 2022-01-07:
已经更新了，从新拉库编译就好了

zopargr, 2022-01-09:
> 已经更新了，从新拉库编译就好了

解决啦！

closed 2022-01-09 by zopargr