coolsnowwolf/lede #8617: 更新最新5.15源码无法上网

Jason6111, 2022-01-05:
未找到原因 打不开网页

mitsukileung, 2022-01-06:
[把这个补丁删除试试](https://github.com/coolsnowwolf/lede/commit/e105008c7dec13ce8a037141a8506916ef6b1689)

Jason6111, 2022-01-06:
> 把这个补丁删除试试

好的 我去试试

Myoko, 2022-01-06:
最新源码是有点问题！昨天编译个 发现$$R和 PW 都不能用了！

Jason6111, 2022-01-06:
成功了 删除[670-fixup-fw3.patch](https://github.com/coolsnowwolf/lede/commit/e105008c7dec13ce8a037141a8506916ef6b1689) 在把automount[改回]https://github.com/coolsnowwolf/lede/commit/1bf58b7e182f9b21da79fb3789c3204010adf219
希望L大改下 辛苦各位了

Jason6111, 2022-01-06:
> 最新源码是有点问题！昨天编译个 发现$$R和 PW 都不通了！ 退回5.10.87 则一切正常

你可以试下我的办法

mitsukileung, 2022-01-06:
> 成功了 删除[670-fixup-fw3.patch](https://github.com/coolsnowwolf/lede/commit/e105008c7dec13ce8a037141a8506916ef6b1689) 在把automount[改回](https://github.com/coolsnowwolf/lede/commit/1bf58b7e182f9b21da79fb3789c3204010adf219) 希望L大改下 辛苦各位了

automount可以退回，只要修改ntfs3-mount就可以了
WYC-2020大佬提供的方法
[WYC-2020/lede@c2690c1](https://github.com/WYC-2020/lede/commit/c2690c17b1e16514ae0148d5b904cf70f843c6f0)

Jason6111, 2022-01-06:
> > 成功了 删除[670-fixup-fw3.patch](https://github.com/coolsnowwolf/lede/commit/e105008c7dec13ce8a037141a8506916ef6b1689) 在把automount[改回](https://github.com/coolsnowwolf/lede/commit/1bf58b7e182f9b21da79fb3789c3204010adf219) 希望L大改下 辛苦各位了
> 
> automount不用退回，只要修改ntfs3-mount就可以了 WYC-2020大佬提供的方法 [WYC-2020/lede@c2690c1](https://github.com/WYC-2020/lede/commit/c2690c17b1e16514ae0148d5b904cf70f843c6f0)

昨晚试过 但还是没法联网 我退回原因是因为删了670补丁 automount就报错 退回后就没事了

WYC-2020, 2022-01-06:
不能上网？这到没注意，有时间瞧瞧 ，毕竟我不用5.15

mitsukileung, 2022-01-06:
> 昨晚试过 但还是没法联网 我退回原因是因为删了670补丁 automount就报错 退回后就没事了

我意思是删除 670-fixup-fw3.patch ，automount 不用退回，按WYC-2020大佬的方法修改 ntfs3-mount 就可以了
删除 670-fixup-fw3.patch 是无法联网问题，修改 ntfs3-mount 是固件启动时不停循环 ntfs3: unknown parameter 'nls' 是问题
automount 不用退回

Jason6111, 2022-01-06:
> 我意思是删除 670-fixup-fw3.patch ，automount 不用退回，按WYC-2020大佬的方法修改 ntfs3-mount 就可以了 删除 670-fixup-fw3.patch 是无法联网问题，修改 ntfs3-mount 是固件启动时不停循环 ntfs3: unknown parameter 'nls' 是问题 automount 不用退回

嗯 也是不错解决方法 找到问题了 等大佬们更新源吧 嘻嘻

WYC-2020, 2022-01-06:
@Jason6111  是不能删除哪个补丁的，是哪个补丁有问题，更新一下就可以，等会我给你发patch

Jason6111, 2022-01-06:
> @Jason6111 是不能删除哪个补丁的，是哪个补丁有问题，更新一下就可以，等会我给你发patch

好的 感谢

WYC-2020, 2022-01-06:
@Jason6111  https://github.com/WYC-2020/lede/commit/e891529e169aece14de0154669935127a7e6f130 用这个补丁替换原来的，是l大补丁有一个地方有问题而已

Jason6111, 2022-01-06:
> @Jason6111 [WYC-2020/lede@e891529](https://github.com/WYC-2020/lede/commit/e891529e169aece14de0154669935127a7e6f130) 用这个补丁替换原来的，是l大补丁有一个地方有问题而已

好的 现在我去试

mitsukileung, 2022-01-06:
> @Jason6111 [WYC-2020/lede@e891529](https://github.com/WYC-2020/lede/commit/e891529e169aece14de0154669935127a7e6f130) 用这个补丁替换原来的，是l大补丁有一个地方有问题而已

大佬发个PR吧

WYC-2020, 2022-01-06:
@mitsukileung 你们去测试吧，测完你们提就是了，我的仓库和l大的有很多我自己的改变，懒得提，哈哈

Jason6111, 2022-01-06:
> @mitsukileung 你们去测试吧，测完你们提就是了，我的仓库和l大的有很多我自己的改变，懒得提，哈哈

试了还是不行 我先用我的办法吧 还是谢谢 辛苦啦

WYC-2020, 2022-01-06:
额，我测试我这边都行呢，哈哈

closed 2022-01-06 by AmadeusGhost