coolsnowwolf/lede #8619: 5.4内核x86编译openwrt-fullconenat错误

netlovehf, 2022-01-06:
make[3]: Entering directory '/home/netlove/lede/package/lean/openwrt-fullconenat'
mkdir -p /home/netlove/lede/dl
SHELL= flock /home/netlove/lede/tmp/.fullconenat-2021-12-04-7fe02604.tar.xz.flock -c '  	/home/netlove/lede/scripts/download.pl "/home/netlove/lede/dl" "fullconenat-2021-12-04-7fe02604.tar.xz" "857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" "" || (  /home/netlove/lede/scripts/dl_github_archive.py --dl-dir="/home/netlove/lede/dl" --url="https://github.com/llccd/netfilter-full-cone-nat.git" --version="7fe02604e073945a3b081c18efbb4d3c799a31f6" --subdir="fullconenat-2021-12-04-7fe02604" --source="fullconenat-2021-12-04-7fe02604.tar.xz" --hash="857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/netlove/lede/tmp/dl && cd /home/netlove/lede/tmp/dl && rm -rf fullconenat-2021-12-04-7fe02604 && [ \! -d fullconenat-2021-12-04-7fe02604 ] && git clone  https://github.com/llccd/netfilter-full-cone-nat.git fullconenat-2021-12-04-7fe02604 && (cd fullconenat-2021-12-04-7fe02604 && git checkout 7fe02604e073945a3b081c18efbb4d3c799a31f6 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fullconenat-2021-12-04-7fe02604 && git log -1 --format='\''@%ct'\''` && rm -rf fullconenat-2021-12-04-7fe02604/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fullconenat-2021-12-04-7fe02604 | 	xz -zc -7e > /home/netlove/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz && mv /home/netlove/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz /home/netlove/lede/dl/ && rm -rf fullconenat-2021-12-04-7fe02604; );  )    '
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_42645540f7ac912202af65068e987065_6664517399ebbbc92a37c5bb081b5c53_check
. /home/netlove/lede/include/shell.sh; xzcat /home/netlove/lede/dl/fullconenat-2021-12-04-7fe02604.tar.xz | tar -C /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604
cp -fpR ./files/Makefile /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_42645540f7ac912202af65068e987065_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.configured_*
rm -f /home/netlove/lede/staging_dir/target-x86_64_musl/stamp/.openwrt-fullconenat_installed
(cd /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/./; if [ -x ./configure ]; then find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r chmod u+w; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r -n1 cp --remove-destination /home/netlove/lede/scripts/config.guess; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r chmod u+w; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r -n1 cp --remove-destination /home/netlove/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.built
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.built_check
make -j1 -C "/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.162" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" M="/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604" EXTRA_CFLAGS="" modules
make[4]: Entering directory '/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.162'
make[5]: *** No rule to make target '/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/xt_FULLCONENAT.o', needed by '__build'.  Stop.
make[4]: *** [Makefile:1734: /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604] Error 2
make[4]: Leaving directory '/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.162'
make[3]: *** [Makefile:68: /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.built] Error 2
make[3]: Leaving directory '/home/netlove/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.74#0.34#1.14
    ERROR: package/lean/openwrt-fullconenat failed to build.
make[2]: *** [package/Makefile:116: package/lean/openwrt-fullconenat/compile] Error 1
make[2]: Leaving directory '/home/netlove/lede'
make[1]: *** [package/Makefile:110: /home/netlove/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/netlove/lede'
make: *** [/home/netlove/lede/include/toplevel.mk:230: world] Error 2

netlovehf, 2022-01-06:
删了源码重新编译，每次都是到这里就错误，dl里的fullconenat-2021-12-04-7fe02604.tar.xz也下载了，还是编译错误，请问是什么原因。

WYC-2020, 2022-01-06:
我这边编译没任何异常呢

netlovehf, 2022-01-06:
make[3]: Entering directory '/home/hzybb/lede/package/lean/openwrt-fullconenat'
mkdir -p /home/hzybb/lede/dl
SHELL= flock /home/hzybb/lede/tmp/.fullconenat-2021-12-04-7fe02604.tar.xz.flock -c '  	/home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "fullconenat-2021-12-04-7fe02604.tar.xz" "857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" "" || (  /home/hzybb/lede/scripts/dl_github_archive.py --dl-dir="/home/hzybb/lede/dl" --url="https://github.com/llccd/netfilter-full-cone-nat.git" --version="7fe02604e073945a3b081c18efbb4d3c799a31f6" --subdir="fullconenat-2021-12-04-7fe02604" --source="fullconenat-2021-12-04-7fe02604.tar.xz" --hash="857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/hzybb/lede/tmp/dl && cd /home/hzybb/lede/tmp/dl && rm -rf fullconenat-2021-12-04-7fe02604 && [ \! -d fullconenat-2021-12-04-7fe02604 ] && git clone  https://github.com/llccd/netfilter-full-cone-nat.git fullconenat-2021-12-04-7fe02604 && (cd fullconenat-2021-12-04-7fe02604 && git checkout 7fe02604e073945a3b081c18efbb4d3c799a31f6 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fullconenat-2021-12-04-7fe02604 && git log -1 --format='\''@%ct'\''` && rm -rf fullconenat-2021-12-04-7fe02604/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fullconenat-2021-12-04-7fe02604 | 	xz -zc -7e > /home/hzybb/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz && mv /home/hzybb/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz /home/hzybb/lede/dl/ && rm -rf fullconenat-2021-12-04-7fe02604; );  )    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/fullconenat-2021-12-04-7fe02604.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/fullconenat-2021-12-04-7fe02604.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/fullconenat-2021-12-04-7fe02604.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
touch /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_85de41ecac59b6a4309a566a8a2f48d2_6664517399ebbbc92a37c5bb081b5c53_check
. /home/hzybb/lede/include/shell.sh; xzcat /home/hzybb/lede/dl/fullconenat-2021-12-04-7fe02604.tar.xz | tar -C /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604
cp -fpR ./files/Makefile /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/
touch /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_85de41ecac59b6a4309a566a8a2f48d2_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.configured_*
rm -f /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.openwrt-fullconenat_installed
(cd /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/./; if [ -x ./configure ]; then find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r chmod u+w; find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r -n1 cp --remove-destination /home/hzybb/lede/scripts/config.guess; find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r chmod u+w; find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r -n1 cp --remove-destination /home/hzybb/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-o [... cut ...]
make[3]: Leaving directory '/home/hzybb/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.50#0.59#12.61
    ERROR: package/lean/openwrt-fullconenat failed to build.
make[2]: *** [package/Makefile:116: package/lean/openwrt-fullconenat/compile] Error 1
make[2]: Leaving directory '/home/hzybb/lede'
make[1]: *** [package/Makefile:110: /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hzybb/lede'
make: *** [/home/hzybb/lede/include/toplevel.mk:230: world] Error 2

netlovehf, 2022-01-06:
> 又编译了一次，没有用make download，结果还是这个错误。发现这几个链接都是无法下载的，但dl里却有fullconenat-2021-12-04-7fe02604.tar.xz这个文件，大小16kb，不知道这个文件有没有下对。 make[3]: Entering directory '/home/hzybb/lede/package/lean/openwrt-fullconenat' mkdir -p /home/hzybb/lede/dl SHELL= flock /home/hzybb/lede/tmp/.fullconenat-2021-12-04-7fe02604.tar.xz.flock -c ' /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "fullconenat-2021-12-04-7fe02604.tar.xz" "857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" "" || ( /home/hzybb/lede/scripts/dl_github_archive.py --dl-dir="/home/hzybb/lede/dl" --url="https://github.com/llccd/netfilter-full-cone-nat.git" --version="7fe02604e073945a3b081c18efbb4d3c799a31f6" --subdir="fullconenat-2021-12-04-7fe02604" --source="fullconenat-2021-12-04-7fe02604.tar.xz" --hash="857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" || ( echo "Checking out files from the git repository..."; mkdir -p /home/hzybb/lede/tmp/dl && cd /home/hzybb/lede/tmp/dl && rm -rf fullconenat-2021-12-04-7fe02604 && [ ! -d fullconenat-2021-12-04-7fe02604 ] && git clone https://github.com/llccd/netfilter-full-cone-nat.git fullconenat-2021-12-04-7fe02604 && (cd fullconenat-2021-12-04-7fe02604 && git checkout 7fe02604e073945a3b081c18efbb4d3c799a31f6 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fullconenat-2021-12-04-7fe02604 && git log -1 --format='\''@%ct'\''` && rm -rf fullconenat-2021-12-04-7fe02604/.git && tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fullconenat-2021-12-04-7fe02604 | xz -zc -7e > /home/hzybb/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz && mv /home/hzybb/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz /home/hzybb/lede/dl/ && rm -rf fullconenat-2021-12-04-7fe02604; ); ) '
> 
> * curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/fullconenat-2021-12-04-7fe02604.tar.xz
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>   Dload  Upload   Total   Spent    Left  Speed
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
>   curl: (22) The requested URL returned error: 404
>   Download failed.
> * curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/fullconenat-2021-12-04-7fe02604.tar.xz
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>   Dload  Upload   Total   Spent    Left  Speed
>   0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
>   curl: (22) The requested URL returned error: 404 Not Found
>   Download failed.
> * curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/fullconenat-2021-12-04-7fe02604.tar.xz
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>   Dload  Upload   Total   Spent    Left  Speed
>   0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
>   curl: (22) The requested URL returned error: 404 Not Found
>   Download failed.
>   No more mirrors to try - giving up.
>   touch /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_85de41ecac59b6a4309a566a8a2f48d2_6664517399ebbbc92a37c5bb081b5c53_check
>   . /home/hzybb/lede/include/shell.sh; xzcat /home/hzybb/lede/dl/fullconenat-2021-12-04-7fe02604.tar.xz | tar -C /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.. -xf -
>   [ ! -d ./src/ ] || cp -fpR ./src/. /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604
>   cp -fpR ./files/Makefile /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/
>   touch /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_85de41ecac59b6a4309a566a8a2f48d2_6664517399ebbbc92a37c5bb081b5c53
>   rm -f /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.configured_*
>   rm -f /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.openwrt-fullconenat_installed
>   (cd /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/./; if [ -x ./configure ]; then find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r chmod u+w; find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r -n1 cp --remove-destination /home/hzybb/lede/scripts/config.guess; find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r chmod u+w; find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r -n1 cp --remove-destination /home/hzybb/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-lin [... cut ...]
>   time: package/lean/openwrt-fullconenat/compile#0.50#0.59#12.61
>   ERROR: package/lean/openwrt-fullconenat failed to build.
>   make[2]: *** [package/Makefile:116: package/lean/openwrt-fullconenat/compile] Error 1
>   make[2]: Leaving directory '/home/hzybb/lede'
>   make[1]: *** [package/Makefile:110: /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
>   make[1]: Leaving directory '/home/hzybb/lede'
>   make: *** [/home/hzybb/lede/include/toplevel.mk:230: world] Error 2

netlovehf, 2022-01-06:
> 

又编译了一次，没有用make download，结果还是这个错误。发现这几个链接都是无法下载的，但dl里却有fullconenat-2021-12-04-7fe02604.tar.xz这个文件，大小16kb，不知道这个文件有没有下对。
make[3]: Entering directory '/home/hzybb/lede/package/lean/openwrt-fullconenat'
mkdir -p /home/hzybb/lede/dl
SHELL= flock /home/hzybb/lede/tmp/.fullconenat-2021-12-04-7fe02604.tar.xz.flock -c ' /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "fullconenat-2021-12-04-7fe02604.tar.xz" "857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" "" || ( /home/hzybb/lede/scripts/dl_github_archive.py --dl-dir="/home/hzybb/lede/dl" --url="https://github.com/llccd/netfilter-full-cone-nat.git" --version="7fe02604e073945a3b081c18efbb4d3c799a31f6" --subdir="fullconenat-2021-12-04-7fe02604" --source="fullconenat-2021-12-04-7fe02604.tar.xz" --hash="857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" || ( echo "Checking out files from the git repository..."; mkdir -p /home/hzybb/lede/tmp/dl && cd /home/hzybb/lede/tmp/dl && rm -rf fullconenat-2021-12-04-7fe02604 && [ ! -d fullconenat-2021-12-04-7fe02604 ] && git clone https://github.com/llccd/netfilter-full-cone-nat.git fullconenat-2021-12-04-7fe02604 && (cd fullconenat-2021-12-04-7fe02604 && git checkout 7fe02604e073945a3b081c18efbb4d3c799a31f6 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=cd fullconenat-2021-12-04-7fe02604 && git log -1 --format='\''@%ct'\'' && rm -rf fullconenat-2021-12-04-7fe02604/.git && tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fullconenat-2021-12-04-7fe02604 | xz -zc -7e > /home/hzybb/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz && mv /home/hzybb/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz /home/hzybb/lede/dl/ && rm -rf fullconenat-2021-12-04-7fe02604; ); ) '

curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/fullconenat-2021-12-04-7fe02604.tar.xz
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
0 0 0 0 0 0 0 0 --:--:-- --:--:-- --:--:-- 0
curl: (22) The requested URL returned error: 404
Download failed.
curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/fullconenat-2021-12-04-7fe02604.tar.xz
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
0 0 0 0 0 0 0 0 --:--:-- 0:00:02 --:--:-- 0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/fullconenat-2021-12-04-7fe02604.tar.xz
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
0 0 0 0 0 0 0 0 --:--:-- 0:00:01 --:--:-- 0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
touch /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_85de41ecac59b6a4309a566a8a2f48d2_6664517399ebbbc92a37c5bb081b5c53_check
. /home/hzybb/lede/include/shell.sh; xzcat /home/hzybb/lede/dl/fullconenat-2021-12-04-7fe02604.tar.xz | tar -C /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604
cp -fpR ./files/Makefile /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/
touch /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_85de41ecac59b6a4309a566a8a2f48d2_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.configured_*
rm -f /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.openwrt-fullconenat_installed
(cd /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/./; if [ -x ./configure ]; then find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r chmod u+w; find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r -n1 cp --remove-destination /home/hzybb/lede/scripts/config.guess; find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r chmod u+w; find /home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r -n1 cp --remove-destination /home/hzybb/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/hzybb/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linu [... cut ...]
make[3]: Leaving directory '/home/hzybb/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.50#0.59#12.61
ERROR: package/lean/openwrt-fullconenat failed to build.
make[2]: *** [package/Makefile:116: package/lean/openwrt-fullconenat/compile] Error 1
make[2]: Leaving directory '/home/hzybb/lede'
make[1]: *** [package/Makefile:110: /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hzybb/lede'
make: *** [/home/hzybb/lede/include/toplevel.mk:230: world] Error 2

>

netlovehf, 2022-01-06:
[image]
[image]

closed 2022-01-06 by AmadeusGhost

lon91ong, 2022-01-11:
为什么关闭了？ 解决了吗？ 我用GitHub Action编译4.14内核的Lienol源码， 也报了同样的错误
具体[参见](https://github.com/lon91ong/Actions-OpenWrt/runs/4757793314)

WYC-2020, 2022-01-11:
只要不是所有人编译失败，个把人编译失败都会关闭