coolsnowwolf/lede #8620: 网易云解锁，go版与node版切换时保存应用不了

lgh200, 2022-01-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：
[image]

sunshineyxc, 2022-01-08:
一样的issue

WYC-2020, 2022-01-10:
我这边一切正常，用kenzok8的luci-app-unblockneteasemusic 覆盖lede @sunshineyxc @lgh200

WYC-2020, 2022-01-10:
https://github.com/WYC-2020/lede/commit/2220a7893f74fae4ff42159f8e135641bc10d1ec
这样修复试下，我这边没问题测试 @lgh200 @sunshineyxc

lgh200, 2022-01-10:
> [WYC-2020/lede@2220a78](https://github.com/WYC-2020/lede/commit/2220a7893f74fae4ff42159f8e135641bc10d1ec) 这样修复试下，我这边没问题测试 @lgh200 @sunshineyxc

之前没问题，是他们最近改出来的问题

WYC-2020, 2022-01-10:
@lgh200 我知道 我刚看了下 从node 切换会go 就会有问题，按我给你的那方法改下 就可以了

lgh200, 2022-01-10:
> 

改天我试试

sunshineyxc, 2022-01-10:
> [WYC-2020/lede@2220a78](https://github.com/WYC-2020/lede/commit/2220a7893f74fae4ff42159f8e135641bc10d1ec) 这样修复试下，我这边没问题测试 @lgh200 @sunshineyxc

/etc/rc.common: line 185: node: not found
Mon, 10 Jan 2022 18:39:07 +0800 # UnblockNeteaseMusic Nodejs Version (http:5200, https:5201)
这是错误代码，Lean不修改这个代码，我自己怎么弄

lgh200, 2022-01-10:
> /etc/rc.common: line 185: node: not found Mon, 10 Jan 2022 18:39:07 +0800 # UnblockNeteaseMusic Nodejs Version (http:5200, https:5201) 这是错误代码，Lean不修改这个代码，我自己怎么弄

自己在本地改就行了

WYC-2020, 2022-01-10:
你这个错误码是没联网吧获取失败 @sunshineyxc

doumeki, 2022-01-10:
不仅LUCI有问题， 好像插件也没用，需要重启下DNSMASQ才能启效

WROIATE, 2022-02-03:
应该是nodejs配置在config里多加了一个字段，但是切换回go的时候没删掉，所以出现了这个情况，你们可以先把config删了再重新再luci上设置就行

yuehongcheng, 2022-07-28:
> https://github.com/WYC-2020/lede/commit/2220a7893f74fae4ff42159f8e135641bc10d1ec 这样修复试下，我这边没问题测试 @lgh200 @sunshineyxc

大佬你好，链接已经失效了。  能否麻烦大佬说一下怎么修复嘛？

coolsnowwolf, 2026-09-05:
网易云解锁切换问题只有截图，没有具体配置、保存后的 UCI 内容或日志；且该第三方应用已多次变更，无法在当前源码中复现，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf