coolsnowwolf/lede #8621: Package kmod-fs-nfs is missing dependencies for the following libraries: nfs_ssc.ko

GreatMichaelLee, 2022-01-06:
5.10.x 依然missing nfs_ssc.ko,能修复吗？

make[3]: Entering directory '/home/builder/lede_x86/package/firmware/prism54-firmware'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/builder/lede_x86/package/firmware/prism54-firmware'
time: package/firmware/prism54-firmware/compile#0.18#0.07#0.24
make[3]: Entering directory '/home/builder/lede_x86/package/kernel/linux'
mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-fs-nfs is missing dependencies for the following libraries:
nfs_ssc.ko
make[3]: *** [modules/fs.mk:395: /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-fs-nfs_5.10.89-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/linux'
time: package/kernel/linux/compile#2.48#0.34#2.81
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$

WYC-2020, 2022-01-06:
不会吧我刚还编译了5.10 只是不是用的lede的仓库，我用的我自己克隆的，不过代码应该和l大的差不了多少 ，虽然我自己修改了一些

GreatMichaelLee, 2022-01-06:
@WYC-2020 @AmadeusGhost 

[image]

WYC-2020, 2022-01-06:
我这里删除了也是没有报的 试下万能大法

breakings, 2022-01-06:
> 我这里删除了也是没有报的 试下万能大法

这个确实得加回去,云编译不存在编译缓存污染问题

WYC-2020, 2022-01-06:
那就不清楚了，反转我本地是没有加回去 一样能编译过，可以自己fork一份想怎么改就怎么改，我就跟到队伍走 @breakings

WYC-2020, 2022-01-06:
@breakings  并且云编译是有缓存一说的哈，主要看开了没

WYC-2020, 2022-01-06:
@GreatMichaelLee 你那里选了什么库，我这边编译5.10选择我常用的库，看config都不会编译kmod-fs-nfs这个模块，怀疑还是哪个包用了哪个模块，你把你config发出来 我给你对比看看
这是我的config
[config.txt](https://github.com/coolsnowwolf/lede/files/7820467/config.txt)

closed 2022-01-06 by AmadeusGhost