coolsnowwolf/lede #8622: luci-app-cpufreq设置里边只有performance模式

weigeloveu, 2022-01-06:
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：luci-app-cpufreq设置里边只有performance模式，频率也不能调节。编译的时候勾选了luci-app-cpufreq。但是看配置文件都是未设置，还要在内核的模块中勾选哪个选项吗？

### (2) 路由器型号和固件版本

A：k3,r22.1.1

### (3) 详细日志

A：
~/lede$ zcat /proc/config.gz | grep CONFIG_CPU_FREQ
CONFIG_CPU_FREQ=y
# CONFIG_CPU_FREQ_STAT is not set
CONFIG_CPU_FREQ_DEFAULT_GOV_PERFORMANCE=y
# CONFIG_CPU_FREQ_DEFAULT_GOV_POWERSAVE is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_USERSPACE is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_ONDEMAND is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_CONSERVATIVE is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_SCHEDUTIL is not set
CONFIG_CPU_FREQ_GOV_PERFORMANCE=y
# CONFIG_CPU_FREQ_GOV_POWERSAVE is not set
# CONFIG_CPU_FREQ_GOV_USERSPACE is not set
# CONFIG_CPU_FREQ_GOV_ONDEMAND is not set
# CONFIG_CPU_FREQ_GOV_CONSERVATIVE is not set
# CONFIG_CPU_FREQ_GOV_SCHEDUTIL is not set

hong0980, 2022-04-19:
运行下 cat  /sys/devices/system/cpu/cpufreq/policy0/scaling_available_governors 看看有几个

coolsnowwolf, 2026-09-05:
cpufreq 可用 governor 取决于 K3 平台 CPU 驱动和内核配置；日志显示仅编译了 performance governor，缺少可用频率表，LuCI 无法凭空提供调频，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf