coolsnowwolf/lede #8624: 可以帮忙看下，编译错误，是什么插件冲突了？还是哪里不对

sd46235599111, 2022-01-06:
可以帮忙看下，编译错误，是什么插件冲突了？还是哪里不对

Collected errors:
 * check_data_file_clashes: Package libgd wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so
        But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so.3
        But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so.3.0.10
        But that file is already provided by package  * libgd-full
 * opkg_install_cmd: Cannot install package luci-app-vnstat.
 * check_data_file_clashes: Package libgd wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so
        But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so.3
        But that file is already provided by package  * libgd-full
 * check_data_file_clashes: Package libgd wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so.3.0.10
        But that file is already provided by package  * libgd-full
 * opkg_install_cmd: Cannot install package libgd.
package/Makefile:66: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/vagrant/lede'
package/Makefile:111: recipe for target '/home/vagrant/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install' failed
make[1]: *** [/home/vagrant/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/vagrant/lede'
/home/vagrant/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

Dimitri2020007, 2022-01-06:
vnstat与kodiexpoler冲突了

前者用libgd后者用libgd-full

只能2选1

sd46235599111, 2022-01-06:
我把vnstat与kodiexpoler都删了。不编译，又报错

Collected errors:
 * check_data_file_clashes: Package libgd-full wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so
        But that file is already provided by package  * libgd
 * check_data_file_clashes: Package libgd-full wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so.3
        But that file is already provided by package  * libgd
 * check_data_file_clashes: Package libgd-full wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so.3.0.10
        But that file is already provided by package  * libgd
 * opkg_install_cmd: Cannot install package libgd-full.
 * check_data_file_clashes: Package libgd-full wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so
        But that file is already provided by package  * libgd
 * check_data_file_clashes: Package libgd-full wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so.3
        But that file is already provided by package  * libgd
 * check_data_file_clashes: Package libgd-full wants to install file /home/vagrant/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/libgd.so.3.0.10
        But that file is already provided by package  * libgd
 * opkg_install_cmd: Cannot install package php8-mod-gd.
package/Makefile:66: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/vagrant/lede'
package/Makefile:111: recipe for target '/home/vagrant/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install' failed
make[1]: *** [/home/vagrant/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/vagrant/lede'
/home/vagrant/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

Dimitri2020007, 2022-01-06:
重新手搓份.config，在选luci-app的时候小心libgd和libgd-full有冲突的项目

closed 2022-01-06 by sd46235599111