coolsnowwolf/lede #8627: 路由器获取ipv6有问题，不知道是什么原因。

sd46235599111, 2022-01-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
ipv6不知道是什么问题，这个rom的路由器本身可以获取到ipv6，
但是其它设备连接到这个路由器，都不会分配ipv6。
rom是我自己编译的，有加了ipv6
已经选了这个Extra packages  --->  ipv6helper  

刷机后，什么都不设置，路由器自动获取ipv6   其它设备连接都不会获取。

该设置，整个路由器的网络瘫痪，上不了网，但是有ipv6，
改设置是下面这样
需要IPv6请在路由器上--网络--接口->LAN->DHCP服务器->IPv6设置:
	   路由器通告模式->改为服务器模式.
	   DHCPv6服务->改为服务器模式
	   NDP代理->禁用
	   总是通告默认路由->勾选
路由器上--网络--接口--全局网络选项--IPV6 ULA前缀里面的都删了，留空。
路由器上--网络--接口--WAN--高级设置--使用内置的ipv6管理打钩
路由器上--网络--防火墙--基本设置--防火墙区域设置--转发选“接受” （里面有拒绝的都改成接受）
路由器上--网络——DHCP/DNS——高级设置中勾选了“禁止解析 IPv6 DNS 记录”打钩
路由器上--网络——负载均衡——策略中备用成员要改成“默认（使用主路由表）”
路由器上--网络--接口--WAN6 可以不用设置，默认就行，或者删了
重启路由器

### (2) 路由器型号和固件版本

A：竟斗云ipq40xx-p2w_r619ac-128m

### (3) 详细日志

A：

sd46235599111, 2022-01-06:
[image]

AdorableJiang, 2022-01-07:
该设备上级有其他路由器吗？还是说直接连接到了入户解调器（通常是光猫）？

sd46235599111, 2022-01-07:
> 该设备上级有其他外接吗？还是说直接连接到了入户解调器（通常是光猫）？

是啊，上级就是光猫了，路由器拨号。

AdorableJiang, 2022-01-07:
方便的话把 网络→接口 中 接口总览完整地贴出来

> [image]

sd46235599111, 2022-01-07:
> 该设备上级有其他路由器吗？还是说直接连接到了入户解调器（通常是光猫）？

[image]

AdorableJiang, 2022-01-07:
WAN接口的 修改→物理设置
看一下绑定的物理端口

sd46235599111, 2022-01-07:
> WAN接口的 修改→物理设置 看一下绑定的物理端口

[image]

AdorableJiang, 2022-01-07:
> > WAN接口的 修改→物理设置 看一下绑定的物理端口
> 
> [image]

请问你的VLAN是如何划分的呢？图上的接口看起来有些 非同寻常
下面是我设备上的接口
[image]

EnnawYang, 2022-01-07:
中继模式试试？

sd46235599111, 2022-01-07:
物理设置，刷机完，自动生成的

AdorableJiang, 2022-01-07:
> 中继模式试试？

支持 @EnnawYang 的建议。
但蹊跷的是，根据此前的经验（南方某地电信），光猫桥接+路由器拨号 的话，拨号的路由器理应可以获得PD以进行IPv6地址下发的。根据你所述的症状，参考[这篇文章](https://zhuanlan.zhihu.com/p/362151770)

> 不过话又说回来，无论划分多少个子网，前面说了所有单播地址（全球公网地址）中子网必须是64bit，就是哪怕只是划分一个 /60 的大子网，路由的通告分配地址的时候，也只是获取 /60 大子网的其中一个 /64 的子网来使用，也就是默认情况下，家用很多路由，比如华硕，华为，TPlink，一类，拨号获得了一个 240e:3c1:5665:1cd0::/60 前缀，然后抽取了 240e:3c1:5665:1cd0::/64 这个子网使用，而其他子网就扔掉了。也因为这个原因，导致路由下再接一个二级路由就没有 IPV6（注1） 。
>
>注1：也是因为这些路由并没有加入DHCP-PD下发服务导致。

顺便问一下，你的路由器此前使用它原本的固件时有这个问题吗？

sd46235599111, 2022-01-07:
> 支持 @EnnawYang 的建议。 但蹊跷的是，根据此前的经验（南方某地电信），光猫桥接+路由器拨号 的话，拨号的路由器理应可以获得PD以进行IPv6地址下发的。根据你所述的症状，参考[这篇文章](https://zhuanlan.zhihu.com/p/362151770)
>
>
> 顺便问一下，你的路由器此前使用它原本的固件时有这个问题吗？

好复杂没看懂，你意思是路由器，固件没有dhcp-pd？

pangfeng161, 2022-01-07:
一句话都是运营商的问题,我这移动20年光猫拨号或者路由器拨号 下级设备根本不能用ipv6(有ipv6地址但无法通过测试)21年完全ok

sd46235599111, 2022-01-07:
> 好复杂没看懂，你意思是路由器，固件没有dhcp-pd？

我不用竟斗云路由器，我用光猫+斐讯k2拨号（刷的老毛子，能获得ipv6，手机wifi电脑接k2都有ipv6。），二级路由红米ax6（红米自己的系统能获得ipv6，手机wifi电脑有线接红米ax6都有ipv6），这个说明，运营商ipv6应该没问题。
这样看问题就出在竟斗云路由器上。  刚编译的固件，跟以前编译的固件，都是上面发帖这个问题。

sd46235599111, 2022-01-07:
> 支持 @EnnawYang 的建议。 但蹊跷的是，根据此前的经验（南方某地电信），光猫桥接+路由器拨号 的话，拨号的路由器理应可以获得PD以进行IPv6地址下发的。根据你所述的症状，参考[这篇文章](https://zhuanlan.zhihu.com/p/362151770)
>
>
> 顺便问一下，你的路由器此前使用它原本的固件时有这个问题吗？

以前的固件可以获取到ipv6  后来宽带账号出问题，停了几天，后面我编译新固件后，就不支持ipv6了，旧的固件也不支持了

sd46235599111, 2022-01-07:
> 中继模式试试？

[image]

你说的是这个中继吗

EnnawYang, 2022-01-07:
> 
> 你说的是这个中继吗

是的，下面两个也改成中继

sd46235599111, 2022-01-07:
> > 你说的是这个中继吗
> 
> 是的，下面两个也改成中继

还是一样的问题

AdorableJiang, 2022-01-07:
用ssh连接到路由器，贴一下
```
cat /etc/config/dhcp
```
和
```
ifconfig
```
的输出

sd46235599111, 2022-01-08:
> cat /etc/config/dhcp

root@OpenWrt:~# cat /etc/config/dhcp

config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'
        option filter_aaaa '1'
        option confdir '/tmp/dnsmasq.d'
        option noresolv '0'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv6 'server'
        option ra 'server'
        option ra_slaac '1'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'

config srvhost
        option srv '_vlmcs._tcp'
        option target 'OpenWrt'
        option port '1688'
        option class '0'
        option weight '100'

root@OpenWrt:~# 
root@OpenWrt:~# 
root@OpenWrt:~# ifconfig
br-lan    Link encap:Ethernet  HWaddr 1C:40:E8:11:16:31  
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fd10:e4cd:95f6::1/60 Scope:Global
          inet6 addr: fe80::1e40:e8ff:fe11:1631/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5177 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7134 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:408547 (398.9 KiB)  TX bytes:2497359 (2.3 MiB)

docker0   Link encap:Ethernet  HWaddr 02:42:2C:75:8A:3B  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 1C:40:E8:11:16:31  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5680 errors:0 dropped:249 overruns:0 frame:0
          TX packets:7150 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:552194 (539.2 KiB)  TX bytes:2500438 (2.3 MiB)

eth1      Link encap:Ethernet  HWaddr 1C:40:E8:11:16:30  
          inet addr:192.168.9.24  Bcast:192.168.9.255  Mask:255.255.255.0
          inet6 addr: 2408:8248:420:8ac0:1e40:e8ff:fe11:1630/64 Scope:Global
          inet6 addr: fe80::1e40:e8ff:fe11:1630/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1141 errors:0 dropped:48 overruns:0 frame:0
          TX packets:870 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:437054 (426.8 KiB)  TX bytes:104251 (101.8 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:191 errors:0 dropped:0 overruns:0 frame:0
          TX packets:191 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:16584 (16.1 KiB)  TX bytes:16584 (16.1 KiB)

wlan0     Link encap:Ethernet  HWaddr 1C:40:E8:11:16:32  
          inet6 addr: fe80::1e40:e8ff:fe11:1632/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:622 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:77802 (75.9 KiB)

wlan1     Link encap:Ethernet  HWaddr 1C:40:E8:11:16:33  
          inet6 addr: fe80::1e40:e8ff:fe11:1633/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:622 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:77802 (75.9 KiB)

root@OpenWrt:~#

sd46235599111, 2022-01-08:
> 用ssh连接到路由器，贴一下
> 
> ```
> cat /etc/config/dhcp
> ```
> 
> 和
> 
> ```
> ifconfig
> ```
> 
> 的输出

root@OpenWrt:~# cat /etc/config/dhcp

config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'
        option filter_aaaa '1'
        option confdir '/tmp/dnsmasq.d'
        option noresolv '0'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv6 'server'
        option ra 'server'
        option ra_slaac '1'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'

config srvhost
        option srv '_vlmcs._tcp'
        option target 'OpenWrt'
        option port '1688'
        option class '0'
        option weight '100'

root@OpenWrt:~# 
root@OpenWrt:~# 
root@OpenWrt:~# ifconfig
br-lan    Link encap:Ethernet  HWaddr 1C:40:E8:11:16:31  
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fd10:e4cd:95f6::1/60 Scope:Global
          inet6 addr: fe80::1e40:e8ff:fe11:1631/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5177 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7134 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:408547 (398.9 KiB)  TX bytes:2497359 (2.3 MiB)

docker0   Link encap:Ethernet  HWaddr 02:42:2C:75:8A:3B  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 1C:40:E8:11:16:31  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5680 errors:0 dropped:249 overruns:0 frame:0
          TX packets:7150 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:552194 (539.2 KiB)  TX bytes:2500438 (2.3 MiB)

eth1      Link encap:Ethernet  HWaddr 1C:40:E8:11:16:30  
          inet addr:192.168.9.24  Bcast:192.168.9.255  Mask:255.255.255.0
          inet6 addr: 2408:8248:420:8ac0:1e40:e8ff:fe11:1630/64 Scope:Global
          inet6 addr: fe80::1e40:e8ff:fe11:1630/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1141 errors:0 dropped:48 overruns:0 frame:0
          TX packets:870 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:437054 (426.8 KiB)  TX bytes:104251 (101.8 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:191 errors:0 dropped:0 overruns:0 frame:0
          TX packets:191 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:16584 (16.1 KiB)  TX bytes:16584 (16.1 KiB)

wlan0     Link encap:Ethernet  HWaddr 1C:40:E8:11:16:32  
          inet6 addr: fe80::1e40:e8ff:fe11:1632/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:622 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:77802 (75.9 KiB)

wlan1     Link encap:Ethernet  HWaddr 1C:40:E8:11:16:33  
          inet6 addr: fe80::1e40:e8ff:fe11:1633/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:622 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:77802 (75.9 KiB)

root@OpenWrt:~#

AdorableJiang, 2022-01-08:
> > 该设备上级有其他路由器吗？还是说直接连接到了入户解调器（通常是光猫）？
> 
> [image]
> 
> 这个是我编译好，什么都没改动设置的图，路由器本身是有ipv6 ，但是其它设备连接获取不到ipv6

LAN→右侧“修改”，底部IPv6设置的 路由通告服务、DHCPv6 服务和NDP 代理全部选为中继。保存&应用
ssh连接路由器，备份dhcp文件
```
cp /etc/config/dhcp /etc/config/dhcp.backup
```
然后编辑dhcp文件
```
vim /etc/config/dhcp
```
找到`wan`接口
```
config dhcp 'wan'
        option interface 'wan'
        option ignore '1'
```
在下面加上`wan`口的中继选项，并指定`wan`口为中继模式中的`master`角色
```
config dhcp 'wan'
        option interface 'wan'
        option ignore '1'
        option ra 'relay'
        option dhcpv6 'relay'
        option ndp 'relay'
        option master '1'
```
保存并退出。重启odhcpd服务
```
./etc/init.d/odhcpd restart
```
或者重启路由器。此时，看子设备是否可以分配到IPv6地址。

如果出现更严重的问题，登入ssh还原dhcp文件
```
cp /etc/config/dhcp.backup /etc/config/dhcp
```

sd46235599111, 2022-01-08:
> LAN→右侧“修改”，底部IPv6设置的 路由通告服务、DHCPv6 服务和NDP 代理全部选为中继。保存&应用 ssh连接路由器，备份dhcp文件
>
> ```
> cp /etc/config/dhcp /etc/config/dhcp.backup
> ```
>
> 然后编辑dhcp文件
>
> ```
> vim /etc/config/dhcp
> ```
>
> 找到`wan`接口
>
> ```
> config dhcp [...]

[image]

[image]

因为家里人要用网络，所以不能直接光猫+竟斗云拨号测试，只能二级路由测试，平时我也是要用二级路由。

改好后我现在是光猫桥接+斐讯k2拨号，二级路由器是竟斗云，手机连wifi、电脑网线连竟斗云有ipv6
我用手机网络（不是wifi）去ping ipv6 有延迟没掉包，可以ping的通。

但是用https://www.test-ipv6.com/ 测试，没有ipv6。

sd46235599111, 2022-01-08:
弄了个域名解析ipv6   ping域名，不行，  域名我以前弄过，不是操作问题， 貌似还是ipv6有点怪

sd46235599111, 2022-01-09:
> LAN→右侧“修改”，底部IPv6设置的 路由通告服务、DHCPv6 服务和NDP 代理全部选为中继。保存&应用 ssh连接路由器，备份dhcp文件
>
> ```
> cp /etc/config/dhcp /etc/config/dhcp.backup
> ```
>
> 然后编辑dhcp文件
>
> ```
> vim /etc/config/dhcp
> ```
>
> 找到`wan`接口
>
> ```
> config dhcp [...]
[image]

电脑连接看是有ipv6  但是用https://www.test-ipv6.com/ 测试，没有ipv6。

电脑防火墙已经关了

AdorableJiang, 2022-01-09:
> [image]
> 
> 电脑连接看是有ipv6 但是用https://www.test-ipv6.com/ 测试，没有ipv6。
> 
> 电脑防火墙已经关了

这样了都不行那是真的麻了呀

feheadls, 2022-01-18:
> [image]
>
> [image]
>
> 因为家里人要用网络，所以不能直接光猫+竟斗云拨号测试，只能二级路由测试，平时我也是要用二级路由。
>
> 改好后我现在是光猫桥接+斐讯k2拨号，二级路由器是竟斗云，手机连wifi、电脑网线连竟斗云有ipv6 我用手机网络（不是wifi）去ping ipv6 有延迟没掉包，可以ping的通。
>
> 但是用https://www.test-ipv6.com/ [...]

从你电脑网卡信息看，已经获取到IPV6地址了，但没获取到DNS，你可以手动指定一个公共DNS地址看看能否正常IPV6的网络访问，再想办法看是哪个环节导致DHCPV6没有把DNS给分配到电脑。

Klosw, 2022-04-05:
我也遇到了 正常的 ipv6 有个分发前缀 ,  对比了正常的IPV6有个 分发前缀  无法下发 ipv6 我根据上面操作 电脑获取到了 ipv6 通过别的网络无法ping 

不正常的ipv6
[image]
正常的ipv6
[image]

microcai, 2023-05-09:
遇到同样的问题了。 使用 ubnt 的路由器， 可以正确获取到 ipv6-pd.
但是使用 wrt 拨号，就只能让 wan6 拿到个 ipv6 地址， 而拿不到 ipv6-pd， 导致 LAN 没有ipv6.
从 pppoe 设备抓包后，可以看到 运营商回的包正确的返回了 ipv6-pd 前缀。不知为何 odhcp 选择忽略运营商的应答。

rtyyrfe, 2023-05-23:
我也遇到同样问题

wxdtt, 2023-06-09:
> 遇到同样的问题了。 使用 ubnt 的路由器， 可以正确获取到 ipv6-pd. 但是使用 wrt 拨号，就只能让 wan6 拿到个 ipv6 地址， 而拿不到 ipv6-pd， 导致 LAN 没有ipv6. 从 pppoe 设备抓包后，可以看到 运营商回的包正确的返回了 ipv6-pd 前缀。不知为何 odhcp 选择忽略运营商的应答。

同样的情况，手动设置PD就通了，但是每次重新拨号PD都会变，很头疼

Klosw, 2023-06-12:
> 

怎么设置?

Klosw, 2023-06-12:
> > 遇到同样的问题了。使用ubnt的路由器，可以正确获取到ipv6-pd。但是使用wrt拨号，只能让wan6拿到一个ipv6地址，而拿不到ipv6-pd，导致局域网没有ipv6。从pppoe设备抓包后，可以看到运营商回的包正确的返回了ipv6-pd之前。不知道为什么odhcp选择忽略运商的回答。
> 
> 同样的情况，手动设置PD就通了，但是每次重新拔号PD都会变，很头疼

怎么设置?

closed 2026-09-05 by coolsnowwolf