coolsnowwolf/lede #8632: 有线网口频繁上下线

WhiteHDD, 2022-01-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
这个问题可能已经有1/2个月了，开始一直以为是个别终端的问题，这几天发现是软路由的问题，每个有线网口只要连接着设备，每天或多或少都会出现自动上下线,。

### (1) 具体问题

A：有线网口频繁上下线

### (2) 路由器型号和固件版本

A：X86 R22.1.1

### (3) 详细日志
Thu Jan  6 20:03:52 2022 kern.info kernel: [ 2079.771681] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Down
Thu Jan  6 20:03:52 2022 kern.info kernel: [ 2079.777943] br-lan: port 3(eth2) entered disabled state
Thu Jan  6 20:03:52 2022 daemon.notice netifd: Network device 'eth2' link is down
Thu Jan  6 20:03:54 2022 kern.info kernel: [ 2081.632067] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Up 10 Mbps Full Duplex, Flow Control: RX/TX
Thu Jan  6 20:03:54 2022 kern.info kernel: [ 2081.641747] br-lan: port 3(eth2) entered blocking state
Thu Jan  6 20:03:54 2022 kern.info kernel: [ 2081.647185] br-lan: port 3(eth2) entered forwarding state
Thu Jan  6 20:03:54 2022 daemon.notice netifd: Network device 'eth2' link is up
Thu Jan  6 20:05:42 2022 kern.info kernel: [ 2189.942994] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Down
Thu Jan  6 20:05:42 2022 kern.info kernel: [ 2189.949194] br-lan: port 3(eth2) entered disabled state
Thu Jan  6 20:05:42 2022 daemon.notice netifd: Network device 'eth2' link is down
Thu Jan  6 20:05:45 2022 kern.info kernel: [ 2192.843335] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Thu Jan  6 20:05:45 2022 kern.info kernel: [ 2192.853533] br-lan: port 3(eth2) entered blocking state
Thu Jan  6 20:05:45 2022 kern.info kernel: [ 2192.858957] br-lan: port 3(eth2) entered forwarding state
Thu Jan  6 20:05:45 2022 daemon.notice netifd: Network device 'eth2' link is up
Thu Jan  6 20:35:15 2022 kern.info kernel: [ 3963.265845] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Down
Thu Jan  6 20:35:15 2022 kern.info kernel: [ 3963.272100] br-lan: port 1(eth0) entered disabled state
Thu Jan  6 20:35:15 2022 daemon.notice netifd: Network device 'eth0' link is down
Thu Jan  6 20:35:17 2022 kern.info kernel: [ 3964.946258] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 10 Mbps Full Duplex, Flow Control: RX/TX
Thu Jan  6 20:35:17 2022 kern.info kernel: [ 3964.955930] br-lan: port 1(eth0) entered blocking state
Thu Jan  6 20:35:17 2022 kern.info kernel: [ 3964.961334] br-lan: port 1(eth0) entered forwarding state
Thu Jan  6 20:35:17 2022 daemon.notice netifd: Network device 'eth0' link is up
Thu Jan  6 20:35:25 2022 kern.info kernel: [ 3973.085958] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Down
Thu Jan  6 20:35:25 2022 kern.info kernel: [ 3973.092149] br-lan: port 1(eth0) entered disabled state
Thu Jan  6 20:35:25 2022 daemon.notice netifd: Network device 'eth0' link is down
Thu Jan  6 23:00:23 2022 kern.info kernel: [12670.591294] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Down
Thu Jan  6 23:00:23 2022 kern.info kernel: [12670.597545] br-lan: port 3(eth2) entered disabled state
Thu Jan  6 23:00:23 2022 daemon.notice netifd: Network device 'eth2' link is down
Thu Jan  6 23:00:24 2022 kern.info kernel: [12672.261713] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Up 10 Mbps Full Duplex, Flow Control: RX/TX
Thu Jan  6 23:00:24 2022 kern.info kernel: [12672.271398] br-lan: port 3(eth2) entered blocking state
Thu Jan  6 23:00:24 2022 kern.info kernel: [12672.276813] br-lan: port 3(eth2) entered forwarding state
Thu Jan  6 23:00:24 2022 daemon.notice netifd: Network device 'eth2' link is up
Fri Jan  7 06:27:35 2022 kern.info kernel: [39503.900061] igb 0000:05:00.0 eth4: igb: eth4 NIC Link is Down
Fri Jan  7 06:27:35 2022 kern.info kernel: [39503.906314] br-lan: port 5(eth4) entered disabled state
Fri Jan  7 06:27:35 2022 daemon.notice netifd: Network device 'eth4' link is down
Fri Jan  7 06:27:38 2022 kern.info kernel: [39506.680487] igb 0000:05:00.0 eth4: igb: eth4 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Fri Jan  7 06:27:38 2022 kern.info kernel: [39506.930211] br-lan: port 5(eth4) entered blocking state
Fri Jan  7 06:27:38 2022 kern.info kernel: [39506.935624] br-lan: port 5(eth4) entered forwarding state
Fri Jan  7 06:27:38 2022 daemon.notice netifd: Network device 'eth4' link is up
Fri Jan  7 06:27:43 2022 kern.info kernel: [39511.590237] igb 0000:05:00.0 eth4: igb: eth4 NIC Link is Down
Fri Jan  7 06:27:43 2022 kern.info kernel: [39511.596394] br-lan: port 5(eth4) entered disabled state
Fri Jan  7 06:27:43 2022 daemon.notice netifd: Network device 'eth4' link is down
Fri Jan  7 06:27:46 2022 kern.info kernel: [39514.310663] igb 0000:05:00.0 eth4: igb: eth4 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Fri Jan  7 06:27:46 2022 kern.info kernel: [39514.560382] br-lan: port 5(eth4) entered blocking state
Fri Jan  7 06:27:46 2022 kern.info kernel: [39514.565805] br-lan: port 5(eth4) entered forwarding state
Fri Jan  7 06:27:46 2022 daemon.notice netifd: Network device 'eth4' link is up
Fri Jan  7 08:00:36 2022 kern.info kernel: [45085.282645] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Down
Fri Jan  7 08:00:36 2022 kern.info kernel: [45085.288852] br-lan: port 3(eth2) entered disabled state
Fri Jan  7 08:00:36 2022 daemon.notice netifd: Network device 'eth2' link is down
Fri Jan  7 08:00:39 2022 kern.info kernel: [45088.193069] igb 0000:03:00.0 eth2: igb: eth2 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Fri Jan  7 08:00:39 2022 kern.info kernel: [45088.322792] br-lan: port 3(eth2) entered blocking state
Fri Jan  7 08:00:39 2022 kern.info kernel: [45088.328181] br-lan: port 3(eth2) entered forwarding state
[... 18 lines cut ...]
Fri Jan  7 10:14:06 2022 kern.info kernel: [53094.821117] igb 0000:05:00.0 eth4: igb: eth4 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Fri Jan  7 10:14:06 2022 kern.info kernel: [53095.070837] br-lan: port 5(eth4) entered blocking state
Fri Jan  7 10:14:06 2022 kern.info kernel: [53095.076262] br-lan: port 5(eth4) entered forwarding state
Fri Jan  7 10:14:06 2022 daemon.notice netifd: Network device 'eth4' link is up
Fri Jan  7 10:14:09 2022 kern.info kernel: [53097.600811] igb 0000:05:00.0 eth4: igb: eth4 NIC Link is Down
Fri Jan  7 10:14:09 2022 kern.info kernel: [53097.607052] br-lan: port 5(eth4) entered disabled state
Fri Jan  7 10:14:09 2022 daemon.notice netifd: Network device 'eth4' link is down
Fri Jan  7 10:14:12 2022 kern.info kernel: [53100.541234] igb 0000:05:00.0 eth4: igb: eth4 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Fri Jan  7 10:14:12 2022 kern.info kernel: [53100.790965] br-lan: port 5(eth4) entered blocking state
Fri Jan  7 10:14:12 2022 kern.info kernel: [53100.796385] br-lan: port 5(eth4) entered forwarding state
Fri Jan  7 10:14:12 2022 daemon.notice netifd: Network device 'eth4' link is up
Fri Jan  7 10:14:21 2022 kern.info kernel: [53109.891057] igb 0000:05:00.0 eth4: igb: eth4 NIC Link is Down
Fri Jan  7 10:14:21 2022 kern.info kernel: [53109.897264] br-lan: port 5(eth4) entered disabled state
Fri Jan  7 10:14:21 2022 daemon.notice netifd: Network device 'eth4' link is down
Fri Jan  7 10:14:24 2022 kern.info kernel: [53112.751479] igb 0000:05:00.0 eth4: igb: eth4 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Fri Jan  7 10:14:24 2022 kern.info kernel: [53113.001203] br-lan: port 5(eth4) entered blocking state
Fri Jan  7 10:14:24 2022 kern.info kernel: [53113.006599] br-lan: port 5(eth4) entered forwarding state
Fri Jan  7 10:14:24 2022 daemon.notice netifd: Network device 'eth4' link is up
Fri Jan  7 10:14:25 2022 kern.info kernel: [53114.151144] igb 0000:05:00.0 eth4: igb: eth4 NIC Link is Down
Fri Jan  7 10:14:25 2022 kern.info kernel: [53114.157322] br-lan: port 5(eth4) entered disabled state
Fri Jan  7 10:14:25 2022 daemon.notice netifd: Network device 'eth4' link is down
Fri Jan  7 10:14:28 2022 kern.info kernel: [53117.061570] igb 0000:05:00.0 eth4: igb: eth4 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Fri Jan  7 10:14:28 2022 kern.info kernel: [53117.311289] br-lan: port 5(eth4) entered blocking state
Fri Jan  7 10:14:28 2022 kern.info kernel: [53117.316716] br-lan: port 5(eth4) entered forwarding state
Fri Jan  7 10:14:28 2022 daemon.notice netifd: Network device 'eth4' link is up

Her0R0cket, 2022-01-08:
新内核编译好像要删.config,重新配置再编译,我没清配置前编译出来的也是这样断断续续的.而且我的网卡还是rtl8111的.

iwrt, 2022-01-12:
同样问题，有线网口频繁上下线。

Jinnrry, 2022-01-13:
同样问题，我还以为我路由器坏了

WhiteHDD, 2022-01-15:
三天前重新编译了，make clean， 去掉了samba4用回了cifsd（不确定与网口断线是否有关，samba4共享总是崩），没有修改.config，用到现在还没发现有网口上下线的情况，供大家参考

XinSSS, 2022-01-27:
同样问题，有线网口频繁上下线。+1

XinSSS, 2022-01-27:
> 新内核编译好像要删.config,重新配置再编译,我没清配置前编译出来的也是这样断断续续的.而且我的网卡还是rtl8111的.

你解决了吗？ 我这边新弄的还是不行哦

Jinnrry, 2022-01-27:
@XinSSS 我重新编译的时候把除了科学上网以外的插件全去掉了，现在没问题了，我猜可能是某个插件引起的

WhiteHDD, 2022-01-28:
12号左右重新编译了一个，有一个连着无线路由（桥接模式，当AP用）的端口还是会断连，1-2天一次，其他有线端口还没有发现自动断连，其他设备每天都会开机关机，只有这个ap是不关的，可能确实和某个网络有关的组件有关，过段时间再重新编译个更干净的试试。

wjz304, 2023-10-07:
虚拟机配置文件添加
pciPassthru0.msiEnabled = “FALSE“

closed 2026-09-05 by coolsnowwolf