coolsnowwolf/lede #8633: ksmbd 3.3.9: compiling error with kernel 5.15

tmptr, 2022-01-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
ksmbd related compile error.
Sorry to interupt when compiling with the latest kernel 5.15 ,there is encouter some issue as desc below .
I selected the cifs-mount cifsd , smb4 luci app. 
https://github.com/namjaejeon/ksmbd/commit/eaecf782d173dba4df0093f5eaf80eac33b67502
Above link seems fix this problem. Could you the maintainer help to update lib of ksmbd. Thank you.
### (2) 路由器型号和固件版本

A：
X86 64 Kernel 5.15

### (3) 详细日志

A：
make[4]: Entering directory '/home/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
  CC [M]  /home/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/vfs.o
/home/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/vfs.c: In function 'ksmbd_vfs_setattr':
/home/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/vfs.c:674:9: error: implicit declaration of function 'locks_verify_truncate'; did you mean 'ksmbd_vfs_truncate'? [-Werror=implicit-function-declaration]
   err = locks_verify_truncate(inode, NULL, attrs->ia_size);
         ^~~~~~~~~~~~~~~~~~~~~
         ksmbd_vfs_truncate
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:277: /home/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/vfs.o] Error 1
make[4]: *** [Makefile:1868: /home/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9] Error 2
make[4]: Leaving directory '/home/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
make[3]: *** [Makefile:74: /home/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/.built] Error 2
make[3]: Leaving directory '/home/lede/feeds/packages/kernel/ksmbd'
time: package/feeds/packages/ksmbd/compile#2.31#0.46#2.74
    ERROR: package/feeds/packages/ksmbd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/ksmbd/compile] Error 1
make[2]: Leaving directory '/home/lede'
make[1]: *** [package/Makefile:110: /home/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wrs/lede'
make: *** [/home/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-01-07:
更新ksmbd我更新了没出错

tmptr, 2022-01-07:
Thanks @WYC-2020 , i try to hard code and have it updated first.

GreatMichaelLee, 2022-01-10:
看我的#8642 @tmptr

closed 2022-02-14 by tmptr