coolsnowwolf/lede #8634: er-x 2022/1/7默认配置编译固件cpu负载稳定2，未开插件

avsba001, 2022-01-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：路由器空载  cpu负载持续在2，
[image]

### (2) 路由器型号和固件版本

A：路由器：ubnt er-x   ,
OpenWrt R22.1.1 / LuCI Master (git-21.335.48743-5f363d9)

### (3) 详细日志

A：Fri Jan  7 12:13:43 2022 kern.notice kernel: [    0.000000] Linux version 5.4.162 (avsba001@avsba001) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4071-44c6f8502)) #0 SMP Thu Jan 6 08:31:04 2022
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] MIPS: machine is Ubiquiti EdgeRouter X
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Fri Jan  7 12:13:43 2022 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Jan  7 12:13:43 2022 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Jan  7 12:13:43 2022 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] Zone ranges:
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000]   HighMem  empty
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] Early memory node ranges
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Fri Jan  7 12:13:43 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Fri Jan  7 12:13:43 2022 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Fri Jan  7 12:13:43 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Fri Jan  7 12:13:43 2022 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
Fri Jan  7 12:13:43 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
Fri Jan  7 12:13:43 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Fri Jan  7 12:13:43 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=0006680e
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=0006680e
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] Memory: 250168K/262144K available (6131K kernel code, 250K rwdata, 1288K rodata, 1276K init, 238K bss, 11976K reserved, 0K cma-reserved, 0K highmem)
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Fri Jan  7 12:13:43 2022 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.015492] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.087818] pid_max: default: 32768 minimum: 301
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.097154] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.111558] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.129672] rcu: Hierarchical SRCU implementation.
Fri Jan  7 12:13:43 2022 kern.info kernel: [    0.139794] smp: Bringing up secondary CPUs ...
Fri Jan  7 12:13:43 2022 kern.warn kernel: [    6.732916] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Fri Jan  7 12:13:43 2022 kern.warn kernel: [    6.732926] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[... 298 lines cut ...]
Fri Jan  7 13:42:50 2022 daemon.err uwsgi[2118]: Fri Jan  7 13:42:50 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Fri Jan  7 13:43:05 2022 daemon.info uwsgi-luci: uWSGI worker 2 cheaped.
Fri Jan  7 13:43:05 2022 daemon.err uwsgi[2118]: Fri Jan  7 13:43:05 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Fri Jan  7 13:44:24 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Jan  7 13:44:24 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Jan  7 13:45:38 2022 daemon.info uwsgi-luci: Fri Jan  7 13:45:38 2022 - SIGPIPE: writing to a closed pipe/socket/fd (probably the client disconnected) on request /cgi-bin/luci/admin/network/turboacc/status?_=0.8482131421748051 (ip 192.168.3.1) !!!
Fri Jan  7 13:45:38 2022 daemon.info uwsgi-luci: Fri Jan  7 13:45:38 2022 - uwsgi_response_writev_headers_and_body_do(): Broken pipe [core/writer.c line 306] during GET /cgi-bin/luci/admin/network/turboacc/status?_=0.8482131421748051 (192.168.3.1)
Fri Jan  7 13:45:44 2022 daemon.info uwsgi-luci: Respawned uWSGI worker 2 (new pid: 8768)
Fri Jan  7 13:46:04 2022 daemon.info uwsgi-luci: uWSGI worker 1 cheaped.
Fri Jan  7 13:46:04 2022 daemon.err uwsgi[2118]: Fri Jan  7 13:46:04 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Fri Jan  7 13:51:00 2022 daemon.info uwsgi-luci: Fri Jan  7 13:51:00 2022 - SIGPIPE: writing to a closed pipe/socket/fd (probably the client disconnected) on request /cgi-bin/luci/admin/services/nps/status?_=0.4215429125296173 (ip 192.168.3.1) !!!
Fri Jan  7 13:51:00 2022 daemon.info uwsgi-luci: Fri Jan  7 13:51:00 2022 - uwsgi_response_writev_headers_and_body_do(): Broken pipe [core/writer.c line 306] during GET /cgi-bin/luci/admin/services/nps/status?_=0.4215429125296173 (192.168.3.1)
Fri Jan  7 13:51:54 2022 user.notice FRPC: Starting Nps service
Fri Jan  7 13:59:34 2022 daemon.err nginx[2666]: 2022/01/07 13:59:34 [crit] 2992#0: *607 SSL_do_handshake() failed (SSL: error:02015016:system library:ioctl:Invalid argument error:2406B076:random number generator:RAND_DRBG_generate:reseed error error:2406E06E:random number generator:RAND_DRBG_reseed:error retrieving entropy error:2406B076:random number generator:RAND_DRBG_generate:reseed error error:14209044:SSL routines:tls_early_post_process_client_hello:internal error) while SSL handshaking, client: 192.168.3.1, server: 0.0.0.0:443
Fri Jan  7 14:05:47 2022 daemon.err uwsgi[2118]: Fri Jan  7 14:05:47 2022 - [emperor] bringing back instance /etc/uwsgi/vassals/luci-webui.ini to on-demand mode
Fri Jan  7 14:05:47 2022 daemon.info uwsgi-luci: Fri Jan  7 14:05:47 2022 - received message 0 from emperor
[... cut ...]
Fri Jan  7 14:29:28 2022 daemon.err uwsgi[2118]: Fri Jan  7 14:29:28 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini has been spawned
Fri Jan  7 14:29:28 2022 daemon.err uwsgi[2118]: Fri Jan  7 14:29:28 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Fri Jan  7 14:29:28 2022 daemon.info uwsgi-luci: spawned uWSGI master process (pid: 12204)
Fri Jan  7 14:29:28 2022 daemon.err uwsgi[2118]: Fri Jan  7 14:29:28 2022 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Fri Jan  7 14:43:21 2022 auth.err passwd: password for root changed by root

avsba001, 2022-01-07:
查明白了，nps反代开起来就有这debuff

closed 2022-01-07 by avsba001

maxmib, 2022-08-24:
我的erx刷了最新的固件后eth0开机后就没反应了，用的pppoe拨号，插上网线不亮灯，只有开机的时候亮灯，请问你的有没有这种情况

avsba001, 2022-09-10:
> 我的erx刷了最新的固件后eth0开机后就没反应了，用的pppoe拨号，插上网线不亮灯，只有开机的时候亮灯，请问你的有没有这种情况

没插lan口，有的固件不插lan就丢网口