coolsnowwolf/lede #8635: 关于ssh爆破问题的咨询

Guxing0321, 2022-01-07:
Hi team，

最近一段时间，总是在后台日志中看到ssh爆破的信息
[image]

我查了下，网上有些脚本来自动添加防火墙策略的

我想问下，有没有想关的插件或者工具能帮助更加优雅高效的解决这个问题

BR
Robin

EnnawYang, 2022-01-07:
[image]
[image]

avsba001, 2022-01-07:
fail2ban

Guxing0321, 2022-01-07:
Hi. EnnawYang,

我这边跟你配置是一样的
[image]

[image]

我用了ddns，加了一个端口转发，我现在不确定是那边爆破的我22号端口，还是10022端口

[image]

BR
Robin

Guxing0321, 2022-01-07:
Hi avsba001

好的，我来研究下 fail2ban相关，谢谢哈

BR
Robin

EnnawYang, 2022-01-07:
> 
> 我用了ddns，加了一个端口转发，我现在不确定是那边爆破的我22号端口，还是10022端口
> 
> BR Robin

从wan10022访问内网机器的22端口，反正是暴露在公网下的，你可以试下楼上推荐的工具，但我没用过，不过不建议把重要端口暴露出去

Guxing0321, 2022-01-07:
> > 我用了ddns，加了一个端口转发，我现在不确定是那边爆破的我22号端口，还是10022端口
> > BR Robin
> 
> 从wan10022访问内网机器的22端口，反正是暴露在公网下的，你可以试下楼上推荐的工具，但我没用过，不过不建议把重要端口暴露出去

好的，谢谢哈，我来研究下上面的工具，不行我就给ssh端口禁用了

BR
Robin

closed 2022-01-07 by Guxing0321