coolsnowwolf/lede #8636: PHP8启动错误

shenyz, 2022-01-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：可道云的PHP不能启动

### (2) 路由器型号和固件版本

A：x86  2022.1.1

### (3) 详细日志

A：

Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'curl.so' (tried: /usr/lib/php/curl.so (Error loading shared library /usr/lib/php/curl.so: No such file or directory), /usr/lib/php/curl.so.so (Error loading shared library /usr/lib/php/curl.so.so: No such file or directory)) in Unknown on line 0
Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'dom.so' (tried: /usr/lib/php/dom.so (Error loading shared library /usr/lib/php/dom.so: No such file or directory), /usr/lib/php/dom.so.so (Error loading shared library /usr/lib/php/dom.so.so: No such file or directory)) in Unknown on line 0
Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'gd.so' (tried: /usr/lib/php/gd.so (Error loading shared library /usr/lib/php/gd.so: No such file or directory), /usr/lib/php/gd.so.so (Error loading shared library /usr/lib/php/gd.so.so: No such file or directory)) in Unknown on line 0
Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'iconv.so' (tried: /usr/lib/php/iconv.so (Error loading shared library /usr/lib/php/iconv.so: No such file or directory), /usr/lib/php/iconv.so.so (Error loading shared library /usr/lib/php/iconv.so.so: No such file or directory)) in Unknown on line 0
Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'mbstring.so' (tried: /usr/lib/php/mbstring.so (Error loading shared library /usr/lib/php/mbstring.so: No such file or directory), /usr/lib/php/mbstring.so.so (Error loading shared library /usr/lib/php/mbstring.so.so: No such file or directory)) in Unknown on line 0
Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'mysqlnd.so' (tried: /usr/lib/php/mysqlnd.so (Error loading shared library /usr/lib/php/mysqlnd.so: No such file or directory), /usr/lib/php/mysqlnd.so.so (Error loading shared library /usr/lib/php/mysqlnd.so.so: No such file or directory)) in Unknown on line 0
Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'pdo.so' (tried: /usr/lib/php/pdo.so (Error loading shared library /usr/lib/php/pdo.so: No such file or directory), /usr/lib/php/pdo.so.so (Error loading shared library /usr/lib/php/pdo.so.so: No such file or directory)) in Unknown on line 0
Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'pdo_mysql.so' (tried: /usr/lib/php/pdo_mysql.so (Error loading shared library /usr/lib/php/pdo_mysql.so: No such file or directory), /usr/lib/php/pdo_mysql.so.so (Error loading shared library /usr/lib/php/pdo_mysql.so.so: No such file or directory)) in Unknown on line 0
Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'pdo_sqlite.so' (tried: /usr/lib/php/pdo_sqlite.so (Error loading shared library /usr/lib/php/pdo_sqlite.so: No such file or directory), /usr/lib/php/pdo_sqlite.so.so (Error loading shared library /usr/lib/php/pdo_sqlite.so.so: No such file or directory)) in Unknown on line 0
Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'session.so' (tried: /usr/lib/php/session.so (Error loading shared library /usr/lib/php/session.so: No such file or directory), /usr/lib/php/session.so.so (Error loading shared library /usr/lib/php/session.so.so: No such file or directory)) in Unknown on line 0
Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'sqlite3.so' (tried: /usr/lib/php/sqlite3.so (Error loading shared library /usr/lib/php/sqlite3.so: No such file or directory), /usr/lib/php/sqlite3.so.so (Error loading shared library /usr/lib/php/sqlite3.so.so: No such file or directory)) in Unknown on line 0
Fri Jan  7 15:38:08 2022 daemon.err uhttpd[3920]: [07-Jan-2022 15:38:08] NOTICE: PHP message: PHP Warning:  PHP Startup: Unable to load dynamic library 'zip.so' (tried: /usr/lib/php/zip.so (Error loading shared library /usr/lib/php/zip.so: No such file or directory), /usr/lib/php/zip.so.so (Error loading shared library /usr/lib/php/zip.so.so: No such file or directory)) in Unknown on line 0

Beginner-Go, 2022-01-07:
PHP7已全面移除，应该是可道云不兼容PHP8

WYC-2020, 2022-01-07:
这个有时间看下，看这个应该是/usr/lib/php这个目录不存在 加载动态库失败

WYC-2020, 2022-01-07:
[image]
 吧这里改下试下 @shenyz

lxwq, 2022-01-08:
PHP7在官方还在维护,不知道为什么会被放弃，
我是手动添加了PHP7进去使用的,

WYC-2020, 2022-01-08:
改下我说的哪里，应该就可以用php8了

xiaoqi2020, 2022-01-08:
> 改下我说的哪里，应该就可以用php8了

请问是改成这样子吗？
[image]
我改成这样子还是启动不了

shenyz, 2022-01-08:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [ X] 没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：可道云的PHP不能启动
>
> ### (2) 路由器型号和固件版本
> A：x86 2022.1.1
>
> ### (3) 详细日志
> A：
>
> Fri Jan 7 15:38:08 2022 [...]

> > 改下我说的哪里，应该就可以用php8了
> 
> 请问是改成这样子吗？ [image] 我改成这样子还是启动不了

我今天认真看下了是可道云的启动文件两处写错了
编辑 /etc/init.d/kodexplorer 文件
在113行处 extension_dir = "/usr/lib/php"   改成 extension_dir = "/usr/lib/php8"
在150行处   /usr/bin/php-fpm改成 /usr/bin/php8-fpm

WYC-2020, 2022-01-08:
那玩意是更新php后kodexplorer修改不完全导致的

xiaoqi2020, 2022-01-08:
> > > 改下我说的哪里，应该就可以用php8了
> > 
> > 
> > 请问是改成这样子吗？ [image] 我改成这样子还是启动不了
> 
> 我今天认真看下了是可道云的启动文件两处写错了 编辑 /etc/init.d/kodexplorer 文件 在113行处 extension_dir = "/usr/lib/php" 改成 extension_dir = "/usr/lib/php8" 在150行处 /usr/bin/php-fpm改成 /usr/bin/php8-fpm

可以了，谢谢大佬！

WYC-2020, 2022-01-08:
应该说的都还没搞完，算了我编译一个 我看下

shenyz, 2022-01-08:
> 应该说的都还没搞完，算了我编译一个 我看下

辛苦看下了，感谢

WYC-2020, 2022-01-08:
https://github.com/WYC-2020/lede/commit/0eac96ac86f6fbbfa54d13088fb3b05e7809fe16
看这个

catchfishday, 2022-01-21:
> > > 改下我说的哪里，应该就可以用php8了
> > 
> > 
> > 请问是改成这样子吗？ [image] 我改成这样子还是启动不了
> 
> 我今天认真看下了是可道云的启动文件两行可能在修改的时候遗忘了 编辑 /etc/init.d/kodexplorer 文件 在113行处 extension_dir = "/usr/lib/php" 改成 extension_dir = "/usr/lib/php8" 在150行处 /usr/bin/php-fpm改成 /usr/bin/php8-fpm

这么改一下，kodbox v1.26版本可以正常启动，个人版本的kodexplorer v4.47还是不太行，起不来。 关闭服务以后，Nginx 服务可以正常关闭，php8-fpm程序依旧关不了。再等等作者适配吧。

WYC-2020, 2022-01-21:
@catchfishday 有时间我再看下，我记得我试了能关闭，有时间我看下是不是box升级导致的，我看很多用kill关闭不行，需要加另外参数

WYC-2020, 2022-01-23:
我这边可道云 php8-fpm程序能关闭 个人版本我还不知道怎么操作，kodbox版本目前我没发现问题 @catchfishday

WYC-2020, 2022-01-23:
我这边测试吧那句话注释掉就可以了，然后就可用个人版，但是功能不知道有什么影响，我这边测试了下感觉没啥问题，最好还是不要用个人版本了，目前默认都是被kodbox替换了
[image]
[image]
@catchfishday

WYC-2020, 2022-01-23:
我这边已经修复了，考虑到官方的不能修复，所以我自己架设了一套，
[image]

新增更新源选项，重置功能，以及个人和企业版选择，目前个人版修复只能用github源作为更新，拿去慢慢用吧
https://github.com/WYC-2020/lede/commit/b9b444921fe47160a9b3196b5a66fca2ff98c0e1 @shenyz @catchfishday @xiaoqi2020 @lxwq

xiaoqi2020, 2022-01-23:
> 我这边已经修复了，考虑到官方的不能修复，所以我自己架设了一套， [image]
> 
> 新增更新源选项，重置功能，以及个人和企业版选择，目前个人版修复只能用github源作为更新，拿去慢慢用吧 [WYC-2020/lede@b9b4449](https://github.com/WYC-2020/lede/commit/b9b444921fe47160a9b3196b5a66fca2ff98c0e1) @shenyz @catchfishday @xiaoqi2020 @lxwq

感谢大佬的辛苦付出！

WYC-2020, 2022-01-24:
@zhangguanzhang 我强迫你用了，傻不拉叽，不懂就不要瞎几把乱说

zhangguanzhang, 2022-01-24:
> @zhangguanzhang 我强迫你用了，傻不拉叽，不懂就不要瞎几把乱说

sb

xiaoqi2020, 2022-01-25:
> > @zhangguanzhang 我强迫你用了，傻不拉叽，不懂就不要瞎几把乱说
> 
> sb

哈皮一个！

zhangguanzhang, 2022-01-25:
> > > @zhangguanzhang 我强迫你用了，傻不拉叽，不懂就不要瞎几把乱说
> > 
> > 
> > sb
> 
> 哈皮一个！

WYC养的狗真听话

xiaoqi2020, 2022-01-25:
> > > > @zhangguanzhang 我强迫你用了，傻不拉叽，不懂就不要瞎几把乱说
> > > 
> > > 
> > > sb
> > 
> > 
> > 哈皮一个！
> 
> WYC养的狗真听话

你就是个智障

zhangguanzhang, 2022-01-25:
> > > > > @zhangguanzhang 我强迫你用了，傻不拉叽，不懂就不要瞎几把乱说
> > > > 
> > > > 
> > > > sb
> > > 
> > > 
> > > 哈皮一个！
> > 
> > 
> > WYC养的狗真听话
> 
> 你就是个智障

狗儿子继续叫

zhangguanzhang, 2022-01-25:
@xiaoqi2020 火葬场说你������️粘锅了让你去看看

xiaoqi2020, 2022-01-25:
> @xiaoqi2020 火葬场说你������️粘锅了让你去看看

你翔吃多了是吧？到处喷！

zhangguanzhang, 2022-01-25:
@xiaoqi2020 你长的怪像人殊不知是只人面犬

zhangguanzhang, 2022-01-25:
@xiaoqi2020 你的脸有碍观瞻

zhangguanzhang, 2022-01-25:
@xiaoqi2020 当初惊艳,完完全全,只为世面见得少。

zhangguanzhang, 2022-01-25:
@xiaoqi2020 嫉妒是一种不好的情感，我理解你嫉妒我有家人，可你这么说有点过分了

zhangguanzhang, 2022-01-25:
@xiaoqi2020 你自拍一张就是你的全家福

zhangguanzhang, 2022-01-25:
@xiaoqi2020 远看是美景，近看想报警

zhangguanzhang, 2022-01-25:
@xiaoqi2020 你在无中生有 你在暗度陈仓 你在凭空想象 你在凭空捏造 你在无言无语 你在无可救药 你是逝者安息 你是一路走好 你是傻子巴拉 你是永无止境 你是没钱买药 你是头脑有病 你是眼里有泡 你是嘴里刘能 你是污言秽语 你是咎由自取 你是殃及无辜 你是祸害众生 你是仓皇失措 你是暗度陈仓 你是无可救药

zhangguanzhang, 2022-01-25:
@xiaoqi2020 胖姑娘袭花衫，花都胖起来。

zhangguanzhang, 2022-01-25:
@xiaoqi2020 即使是做咸鱼，也要做最咸的那一条。

zhangguanzhang, 2022-01-25:
@xiaoqi2020 就是嫉妒我户口本可以翻页（说他是孤儿）

zhangguanzhang, 2022-01-25:
@xiaoqi2020 祝您寿比昙花

zhangguanzhang, 2022-01-25:
@xiaoqi2020 尔等狗彘鼠虫之辈,焉与我并列之？

zhangguanzhang, 2022-01-25:
@xiaoqi2020 你傻了吧唧好像我去年路过怡红院的那个老鸨的狗咬死的一只鸡

zhangguanzhang, 2022-01-25:
@xiaoqi2020 掀起你的头盖骨，让我来看看里面的水

zhangguanzhang, 2022-01-25:
@xiaoqi2020 去拼多多拼个妈再来跟我说话吧

zhangguanzhang, 2022-01-25:
@xiaoqi2020 你是一坨屎，人渣

zhangguanzhang, 2022-01-25:
@xiaoqi2020 他们怎么就能因为你长的像就说你是猪呢？

zhangguanzhang, 2022-01-25:
@xiaoqi2020 每个人都有妈妈，而你就不一样了，你不是人

zhangguanzhang, 2022-01-25:
@xiaoqi2020 |    | |    / |  /~~ |  _x000D_

zhangguanzhang, 2022-01-25:
@xiaoqi2020 |    | |   /  |   `--.    |  _x000D_

zhangguanzhang, 2022-01-25:
@xiaoqi2020 人不犯我我不犯人 人若犯我 我挖你祖坟

zhangguanzhang, 2022-01-25:
@xiaoqi2020 你好像美羊羊，三千多集没有妈

zhangguanzhang, 2022-01-25:
@xiaoqi2020 我确实弱，你确实猛，猛得像威猛先生。

zhangguanzhang, 2022-01-25:
@xiaoqi2020 你简直浪费你爸妈那仅有两分钟的一晚

zhangguanzhang, 2022-01-25:
@EnnawYang 你也要参战是吗

zhangguanzhang, 2022-01-25:
@xiaoqi2020 你家开的火葬场是用来自娱自乐的吗？

zhangguanzhang, 2022-01-25:
@xiaoqi2020 @EnnawYang 1、说话都说不清楚，老跟嘴里含着粪似的。

zhangguanzhang, 2022-01-25:
@xiaoqi2020 @EnnawYang 2、你的长相很提神的说！

zhangguanzhang, 2022-01-25:
@xiaoqi2020 @EnnawYang 3、夸你两句就飘飘然了，其实我是不忍心说你是先天性傻逼所以才夸你，哪知道你竟这么不懂得分寸。

zhangguanzhang, 2022-01-25:
@xiaoqi2020 @EnnawYang 4、我拜托你，不好再出来吓人了，要不然这世界就要被灭亡了。

zhangguanzhang, 2022-01-25:
@xiaoqi2020 @EnnawYang 5、你全家非主流，你妈黒袜子，你爸锡纸头。

xiaoqi2020, 2022-01-25:
> @xiaoqi2020 @EnnawYang 4、我拜托你，不好再出来吓人了，要不然这世界就要被灭亡了。

这世界上怎么会有你这种SB的？网上牛逼哄哄，现实中一条狗！

zhangguanzhang, 2022-01-25:
@xiaoqi2020 @EnnawYang 6、没事求你不好腆着个逼脸出来蹦达，实在是太恶心人了。

zhangguanzhang, 2022-01-25:
@xiaoqi2020 @EnnawYang 7、人生若只如初见，劳资当初就就应在你妈B里插把剑，一刀刺死你个小杂种。

[10 more replies]

closed 2026-09-06 by coolsnowwolf