coolsnowwolf/lede #8637: 编译https-dns-proxy时报错，多种尝试无效。

bollus, 2022-01-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
编译https-dns-proxy时报错
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题

A：尝试过vm虚拟机，云服务器，actions，WLS。
有清理编译缓存，tmp，但始终编译https-dns-proxy报错。

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：make[3]: Entering directory '/home/caomei/lede/feeds/packages/net/https-dns-proxy'
mkdir -p /home/caomei/lede/dl
SHELL= flock /home/caomei/lede/tmp/.https-dns-proxy-2021-06-03.tar.xz.flock -c '  	/home/caomei/lede/scripts/download.pl "/home/caomei/lede/dl" "https-dns-proxy-2021-06-03.tar.xz" "b65161936269aa3117debad0fcfce157024726b78d7e7da77c226f7aa8da5b4d" "" || (  /home/caomei/lede/scripts/dl_github_archive.py --dl-dir="/home/caomei/lede/dl" --url="https://github.com/aarond10/https_dns_proxy" --version="5651b984f770a8bcecb14aeffc224703f8f82586" --subdir="https-dns-proxy-2021-06-03" --source="https-dns-proxy-2021-06-03.tar.xz" --hash="b65161936269aa3117debad0fcfce157024726b78d7e7da77c226f7aa8da5b4d" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/caomei/lede/tmp/dl && cd /home/caomei/lede/tmp/dl && rm -rf https-dns-proxy-2021-06-03 && [ \! -d https-dns-proxy-2021-06-03 ] && git clone  https://github.com/aarond10/https_dns_proxy https-dns-proxy-2021-06-03 && (cd https-dns-proxy-2021-06-03 && git checkout 5651b984f770a8bcecb14aeffc224703f8f82586 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd https-dns-proxy-2021-06-03 && git log -1 --format='\''@%ct'\''` && rm -rf https-dns-proxy-2021-06-03/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c https-dns-proxy-2021-06-03 | 	xz -zc -7e > /home/caomei/lede/tmp/dl/https-dns-proxy-2021-06-03.tar.xz && mv /home/caomei/lede/tmp/dl/https-dns-proxy-2021-06-03.tar.xz /home/caomei/lede/dl/ && rm -rf https-dns-proxy-2021-06-03; );  )    '
touch /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03/.prepared_3d0fa673e0211f5f1781a5c181bbb531_6664517399ebbbc92a37c5bb081b5c53_check
. /home/caomei/lede/include/shell.sh; xzcat /home/caomei/lede/dl/https-dns-proxy-2021-06-03.tar.xz | tar -C /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03
touch /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03/.prepared_3d0fa673e0211f5f1781a5c181bbb531_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03/.configured_*
rm -f /home/caomei/lede/staging_dir/target-x86_64_musl/stamp/.https-dns-proxy_installed
mkdir -p /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03
(cd /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03; CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03=https-dns-proxy-2021-06-03 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03=https-dns-proxy-2021-06-03 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " LDFLAGS="-L/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=x86_64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/caomei/lede/staging_dir/target-x86_64_musl/usr;/home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/caomei/lede/staging_dir/target-x86_64_musl -DCMAKE_PREFIX_PATH=/home/caomei/lede/staging_dir/target-x86_64_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE -DCMAKE_GENERATOR="Ninja" -DCLANG_TIDY_EXE= /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03 )
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/caomei/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found Git: /home/caomei/lede/staging_dir/host/bin/git (found version "2.25.1") 
-- Version: 2022.01.04-44c6f8502
-- clang-tidy not found.
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBCURL_INCLUDE_DIR
   used as include directory in directory /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03
   used as include directory in directory /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03
   used as include directory in directory /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03
   used as include directory in directory /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03
   used as include directory in directory /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03
   used as include directory in directory /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03
   used as include directory in directory /home/caomei/lede/build_dir/target-x86_64_musl/https-dns-proxy-2021-06-03

CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0021 is not set: Fatal error on relative paths in
  INCLUDE_DIRECTORIES target property.  Run "cmake --help-policy CMP0021" for
[... cut ...]
time: package/feeds/packages/https-dns-proxy/compile#0.21#1.23#9.79
    ERROR: package/feeds/packages/https-dns-proxy failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/https-dns-proxy/compile] Error 1
make[2]: Leaving directory '/home/caomei/lede'
make[1]: *** [package/Makefile:110: /home/caomei/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/caomei/lede'
make: *** [/home/caomei/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-01-07:
没编译过有时间看下

bollus, 2022-01-07:
> 没编译过有时间看下
好的，谢谢了！

WYC-2020, 2022-01-08:
[image]
我这边5.15编译 这样修复就可以了
https://github.com/WYC-2020/lede/commit/d080859e8ab5290f3ebe92626d27ebb8840cc470

WYC-2020, 2022-01-08:
@bollus

WYC-2020, 2022-01-08:
不过看你这个错误，不是由于这个原因导致的错误，说说你编译的是多少？5.4 5.10 5.15 @bollus

bollus, 2022-01-08:
> 不过看你这个错误，不是由于这个原因导致的错误，说说你编译的是多少？5.4 5.10 5.15 @bollus

我这里是5.10 @WYC-2020

bollus, 2022-01-08:
我尝试换成5.15 然后用你的方式修复试试

WYC-2020, 2022-01-08:
@bollus 我刚5.10 编译也没问题，你先按我按个修复 然后再试下勒

bollus, 2022-01-08:
@WYC-2020 好的

bollus, 2022-01-08:
@WYC-2020 刚刚又看了下你的日志，你是编译的luci-app-https-dns-proxy，我这里是openwrt原本的https-dns-proxy，应该是个依赖

WYC-2020, 2022-01-08:
你是官方op,那不清楚了我没编译过 @bollus

WYC-2020, 2022-01-08:
我这边编译luci-app-https-dns-proxy 他也会去依赖https-dns-proxy我看了，我这边的版本和你那个是一样的 @bollus

bollus, 2022-01-08:
@WYC-2020 我刚试了按你的方式修复，还是一样的
[image]
😂

WYC-2020, 2022-01-08:
@bollus 刚我去官方op上面看到别人的是这个版本
[image]
你这个是在哪里组合的，你用官方的op用官方的package看下

bollus, 2022-01-09:
@WYC-2020 嗯嗯，我也准备用官方的op试下了

closed 2024-07-04 by bollus