coolsnowwolf/lede #8642: ERROR: package/feeds/packages/ksmbd failed to build.

GreatMichaelLee, 2022-01-08:
5.15 编译出错， 是make clean && rm -rf ./tmp 后重新编译的，哪位大佬看看什么原因？

SHELL= flock /home/builder/lede_x86/tmp/.keyutils-1.6.3.tar.gz.flock -c '       /home/builder/lede_x86/scripts/download.pl "/home/builder/lede_x86/dl" "keyutils-1.6.3.tar.gz" "a61d5706136ae4c05bd48f86186bcfdbd88dd8bd5107e3e195c924cfc1b39bb4" "" "https://git.kernel.org/pub/scm/linux/kernel/git/dhowells/keyutils.git/snapshot/"    '
mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp
echo '1' | cmp -s - /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/libkeyutils.version || { echo '1' > /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/libkeyutils.version;  }
SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/keyutils-1.6.3/.pkgdir/libkeyutils/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.libkeyutils_installed
echo "libkeyutils" >> /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/keyutils.default.install
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/libs/keyutils'
time: package/feeds/packages/keyutils/compile#0.19#0.12#0.29
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/kernel/ksmbd'
mkdir -p /home/builder/lede_x86/dl
SHELL= flock /home/builder/lede_x86/tmp/.ksmbd-3.3.9.tar.gz.flock -c '          /home/builder/lede_x86/scripts/download.pl "/home/builder/lede_x86/dl" "ksmbd-3.3.9.tar.gz" "c196d1773b9f89221133780fd189b550acbc56ac93c2e79260a70eab9853b3e1" "" "https://codeload.github.com/cifsd-team/cifsd/tar.gz/3.3.9?"    '
rm -f /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/.built
touch /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/.built_check
make -C /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12 KCFLAGS="-fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/builder/lede_x86/staging_dir/host/include -I/home/builder/lede_x86/staging_dir/hostpkg/include -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Jan  7 03:21:27 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/builder/lede_x86/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/builder/lede_x86/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.12 M="/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9" EXTRA_CFLAGS="-DCONFIG_SMB_INSECURE_SERVER=1" CONFIG_SMB_INSECURE_SERVER=y CONFIG_SMB_SERVER=m modules
make[4]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
  CC [M]  /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/vfs.o
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/vfs.c: In function 'ksmbd_vfs_setattr':
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/vfs.c:674:9: error: implicit declaration of function 'locks_verify_truncate'; did you mean 'ksmbd_vfs_truncate'? [-Werror=implicit-function-declaration]
   err = locks_verify_truncate(inode, NULL, attrs->ia_size);
         ^~~~~~~~~~~~~~~~~~~~~
         ksmbd_vfs_truncate
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:277: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/vfs.o] Error 1
make[4]: *** [Makefile:1868: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
make[3]: *** [Makefile:74: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.3.9/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/kernel/ksmbd'
time: package/feeds/packages/ksmbd/compile#1.45#0.57#2.42
    ERROR: package/feeds/packages/ksmbd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/ksmbd/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$

WYC-2020, 2022-01-08:
升级看下，我这边没问题，我升级了的

GreatMichaelLee, 2022-01-08:
> 升级看下，我这边没问题，我升级了的

升级是指？ git pull .....还是apt upgrade ?-_-!

GreatMichaelLee, 2022-01-08:
> > 升级看下，我这边没问题，我升级了的
> 
> 升级是指？ git pull .....还是apt upgrade ?-_-!

二者都试了，没用，还是报这个

WYC-2020, 2022-01-08:
额更新ksmbd啊

GreatMichaelLee, 2022-01-08:
> 额更新ksmbd啊

how to?

GreatMichaelLee, 2022-01-08:
我看这个是在scripts/download.pl里自动下载的3.3.9版本啊？好像最新支持ntfs3的是3.4.3版本？那我没法改啊

SHELL= flock /home/builder/lede_x86/tmp/.ksmbd-3.3.9.tar.gz.flock -c ' /home/builder/lede_x86/scripts/download.pl "/home/builder/lede_x86/dl" "ksmbd-3.3.9.tar.gz" "c196d1773b9f89221133780fd189b550acbc56ac93c2e79260a70eab9853b3e1" "" "https://codeload.github.com/cifsd-team/cifsd/tar.gz/3.3.9?" '

WYC-2020, 2022-01-08:
自己fork改或者用我那个仓库，随便都可以，或者你feed update后替换那个ksmbd文件夹不就可以了，方法太多

GreatMichaelLee, 2022-01-08:
> 自己fork改或者用我那个仓库，随便都可以，或者你feed update后替换那个ksmbd文件夹不就可以了，方法太多

可是可以，那到时候origin分支改了，我本地又要check out，麻烦…只想本地跟上游sync,除了极个别的插件外，能不能PR改下？而且这个理应要改吧？

GreatMichaelLee, 2022-01-08:
> 自己fork改或者用我那个仓库，随便都可以，或者你feed update后替换那个ksmbd文件夹不就可以了，方法太多

没用，搞不定了，我改了ksmod的makefile, `feeds/packages/kernel/ksmbd/Makefile`

更新了包版本3.3.9到3.4.3，也改了HASH值，但重新编译又说有新的ko依赖，疯了

[image]

新的ko依赖错误
[image]

WYC-2020, 2022-01-08:
自己去看我的

GreatMichaelLee, 2022-01-08:
> 自己去看我的

看不懂，代码目录结构都不熟，管中窥豹......-_-!算了，不搞了....小白就图个乐呵。

WYC-2020, 2022-01-08:
吧相同目录 覆盖你相同目录 就可以 这都不会 哈哈

GreatMichaelLee, 2022-01-08:
> 吧相同目录 覆盖你相同目录 就可以 这都不会 哈哈

不知道你说的哪个目录？

> 吧相同目录 覆盖你相同目录 就可以 这都不会 哈哈

It doesn't work ,man ......

GreatMichaelLee, 2022-01-09:
这个问题我另外专门找了一台干净的oracle云服务器，重新拉的代码，直接一上来就选的testing kernel, 然后手工重新make menuconfig 把包一个个个选上，杜绝了拷贝.config不干净的问题，重新编译，还是不行。把ksmbd手工换成3.4.3还是报缺库，源代码是3.3.9，麻烦各位大佬再看看啊。

[image]

WYC-2020, 2022-01-09:
你还是不要折腾了哈哈

GreatMichaelLee, 2022-01-09:
> 你还是不要折腾了哈哈

哎，无语。No true help.

WYC-2020, 2022-01-09:
@GreatMichaelLee  那么简单 都给你说了替换ksmbd的Make成我的，都太简单了 所以我都不想说，
[image]

GreatMichaelLee, 2022-01-10:
> @GreatMichaelLee 那么简单 都给你说了替换ksmbd的Make成我的，都太简单了 所以我都不想说， [image]

谦虚是种美德，人可以有傲骨，但不能有傲气。如果人人都是对代码了如指掌，都只需要语焉不详也能心领神会，那天下无bug.

GreatMichaelLee, 2022-01-10:
> @GreatMichaelLee 那么简单 都给你说了替换ksmbd的Make成我的，都太简单了 所以我都不想说， [image]

按你说的改了，没卵用，依旧报这个，而且你的这份对比lede源mk少一个“ #kmod-crypto-arc4 \”，导致编译还会多一个libdes.o的报错

[image]

WYC-2020, 2022-01-10:
@GreatMichaelLee  那我就不知道了，反正我自己的没问题，这个你等他们修复吧，本身我的仓库和l大就有差距哈哈

wozualong, 2022-01-12:
> > @GreatMichaelLee 那么简单 都给你说了替换ksmbd的Make成我的，都太简单了 所以我都不想说， [image]
> 
> 按你说的改了，没卵用，依旧报这个，而且你的这份对比lede源mk少一个“ kmod-crypto-arc4 \”，导致编译还会多一个libdes.o的报错
> 
> [image]

少依赖了，@WYC-2020 大佬当时给我帮我改的makefile是有效的，我当时自己编译报错，大佬加上了+kmod-asn1-decoder依赖，编译完美通过，你加上试试

DEPENDS:= \
		+kmod-nls-base \
		+kmod-nls-utf8 \
		+kmod-crypto-md4 \
		+kmod-crypto-md5 \
		+kmod-crypto-hmac \
		+kmod-crypto-arc4 \
		+kmod-crypto-ecb \
		+kmod-crypto-des \
		+kmod-crypto-sha256 \
		+kmod-crypto-cmac \
		+kmod-crypto-sha512 \
		+kmod-crypto-aead \
		+kmod-crypto-ccm \
		+kmod-crypto-gcm \
		+kmod-lib-crc32c \
		+kmod-asn1-decoder

WYC-2020, 2022-01-12:
@wozualong 是的，你这个是一开始我给你的，还有一个补丁，后面我优化了一下不用补丁，我这边编译完全没问题，你可是试下我的最新的

wozualong, 2022-01-12:
> @wozualong 是的，你这个是一开始我给你的，还有一个补丁，后面我优化了一下不用补丁，我这边编译完全没问题，你可是试下我的最新的

嗯，我知道，我回复他的，等空了更新测试下

wozualong, 2022-01-24:
> @wozualong 是的，你这个是一开始我给你的，还有一个补丁，后面我优化了一下不用补丁，我这边编译完全没问题，你可是试下我的最新的

@WYC-2020 大佬，我去掉你的补丁之后，还是会编译报错
ERROR: "sprint_oid" [/home/wozualong/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ksmbd-3.4.3/ksmbd.ko] undefined!
make[5]: *** [scripts/Makefile.modpost:94: __modpost] Error 1

加上你之前给我的补丁才能正常编译通过

WYC-2020, 2022-01-24:
@wozualong 还要更新一个模块依赖才行

GreatMichaelLee, 2022-01-24:
> @wozualong 还要更新一个模块依赖才行

还要一个+kmod-oid-registry

WYC-2020, 2022-01-24:
@GreatMichaelLee 那是你们有的模块依赖和我的不同，我的已经有模块间接依赖了，所以不需要

j0bb0, 2022-03-01:
> 

哈哈哈，扎心了

closed 2022-07-25 by aiamadeus