coolsnowwolf/lede #8645: 主机型号显示问题

harleywong, 2022-01-08:
今天编译了5.15的固件，刷入后主机型号显示有点乱，前几天编译的没这情况。
[image]
[image]

WYC-2020, 2022-01-08:
明显下面那个是谁自己diy的前面一段

Beginner-Go, 2022-01-08:
[应该是这个commit导致的](https://github.com/coolsnowwolf/lede/commit/dced95e70556960363c366b8783e5695290186f2)

binge8, 2022-01-08:
> [应该是这个commit导致的](https://github.com/coolsnowwolf/lede/commit/dced95e70556960363c366b8783e5695290186f2)

一样，前面加了一大堆，怎么取消啊

binge8, 2022-01-08:
[不是这个？](https://github.com/coolsnowwolf/lede/commit/b143589cf2cba27afb46789146f97ffd765a530f)
怎么取消呢？太累赘

wangqn, 2022-01-08:
我也要同样的问题，使用最新版编译后型号那边会出现Default string Default string/Default string - 
[image]

wkq2004, 2022-01-10:
同样的问题。出现Default string Default string/Default string -

jasonqdu, 2022-01-10:
我最近编译的前面也是有一串：To be filled by O.E.M. To be filled by O.E.M./To be filled by O.E.M. -

musclemm, 2022-01-10:
同样的问题。出现Default string Default string
[image]

wangqn, 2022-01-10:
经过实验此方法可以去掉 Default string Default string/Default string -
\package\lean\autocore\files\x86\autocore
`h=${g}' - '${a}${b}${c}${d}${e}${f}` 改成 `h=${a}${b}${c}${d}${e}${f}`

wkq2004, 2022-01-10:
> 经过实验此方法可以去掉 Default string Default string/Default string - \package\lean\autocore\files\x86\autocore `h=${g}' - '${a}${b}${c}${d}${e}${f}` 改成 `h=${a}${b}${c}${d}${e}${f}`

解决了，谢谢。

closed 2022-02-12 by Beginner-Go