coolsnowwolf/lede #8653: lede系统路由器无法从源更新，是证书问题，求解决方法。

stingzou, 2022-01-08:
`root@LEDE:/tmp# opkg update
Downloading https://openwrt.proxy.ustclug.org/snapshots/targets/mvebu/cortexa9/packages/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/targets/mvebu/cortexa9/packages/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/base/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/base/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/luci/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/luci/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/Packages.gz

root@LEDE:/tmp# opkg update
Downloading https://openwrt.proxy.ustclug.org/snapshots/targets/mvebu/cortexa9/packages/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/targets/mvebu/cortexa9/packages/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/base/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/base/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/luci/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/luci/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/routing/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/routing/Packages.gz

Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/telephony/Packages.gz
*** Failed to download the package list from https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/telephony/Packages.gz

Collected errors:

opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/targets/mvebu/cortexa9/packages/Packages.gz, wget returned 5.
opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/base/Packages.gz, wget returned 5.
opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/luci/Packages.gz, wget returned 5.
opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/Packages.gz, wget returned 5.
opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/routing/Packages.gz, wget returned 5.
opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots/packages/arm_cortex-a9_vfpv3/telephony/Packages.gz, wget returned 5.`
然后wget文件时也是报错：

root@LEDE:/tmp# wget https://openwrt.proxy.ustclug.org/snapshots/targets/mvebu/cortexa9/packages/Packages.gz --2022-01-08 21:38:29-- https://openwrt.proxy.ustclug.org/snapshots/targets/mvebu/cortexa9/packages/Packages.gz Resolving openwrt.proxy.ustclug.org... 2400:ddc0:1000::6417:bae0, 192.109.232.118 Connecting to openwrt.proxy.ustclug.org|2400:ddc0:1000::6417:bae0|:443... connected. ERROR: cannot verify openwrt.proxy.ustclug.org's certificate, issued by 'CN=R3,O=Let\'s Encrypt,C=US': Issued certificate has expired. To connect to openwrt.proxy.ustclug.org insecurely, use --no-check-certificate'.`

求助，这问题咋解决，我现在无法安装任何软件。

WYC-2020, 2022-01-08:
不是给你提示了--no-check-certificate

stingzou, 2022-01-08:
> 不是给你提示了--no-check-certificate

opkg update又不能用--no-check-certificate。

我现在换到downloads.openwrt.org的源，opkg update时遇到有个sig无法下载，但其他的都是好的。
Downloading https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/arm_core
Downloading https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/base/Packages.gz
Updated list of available packages in /var/opkg-lists/arm_base
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/arm_luci
**Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.**
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/arm_packages
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/arm_routing
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/arm_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/telephony/Packages.sig
Signature check passed.

WYC-2020, 2022-01-08:
我都是自己编译自带然后自己在op上搭建源，这样用的哈哈

hxlls, 2022-01-10:
[image]
看图，划线部分前面加上#注销掉，然后更新试试

bu2hao, 2024-11-16:
> [image] 看图，划线部分前面加上#注销掉，然后更新试试

厉害，看到你这里豁然开朗，确实解决了

doublebody, 2024-11-22:
mark

coolsnowwolf, 2026-09-05:
该日志中的证书已于 2022 年过期，且使用的是第三方 `openwrt.proxy.ustclug.org` 镜像；这不是 LEDE 源码或 opkg 的固定错误。设备需确认系统时间正确、安装/更新 `ca-bundle`，并可切换到当前有效的官方/其他镜像。当前 LEDE 仍提供标准 CA 证书包，issue 已过时。

closed 2026-09-05 by coolsnowwolf