coolsnowwolf/lede #8654: 求助，内核i915报错

jaychin2, 2022-01-08:
内核日志：
[   63.748389] i915 0000:00:02.0: [drm] Failed to load DMC firmware i915/tgl_dmc_ver2_12.bin. Disabling runtime power management.
[   63.760960] i915 0000:00:02.0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915

使用一段时间会出现重启，而且无法重启成功

deplives, 2022-01-09:
https://github.com/coolsnowwolf/lede/commit/bcad74fa27a6383651397e91389059818401a0f9
这个commit已经修复

jaychin2, 2022-01-09:
> [bcad74f](https://github.com/coolsnowwolf/lede/commit/bcad74fa27a6383651397e91389059818401a0f9) 这个commit已经修复

啥时候修复的今天吗？

deplives, 2022-01-09:
> bcad74f

你但凡点开那个 commit 看一下也不会问啥时候修复的

minlang112, 2022-01-09:
> > [bcad74f](https://github.com/coolsnowwolf/lede/commit/bcad74fa27a6383651397e91389059818401a0f9) 这个commit已经修复
> 
> 啥时候修复的今天吗？

10 days ago……

jaychin2, 2022-01-09:
问题是我今天编译的出的这个！

jaychin2, 2022-01-09:
几天前就一直有，重新拉库也还是有。明天编译完再看吧。

deplives, 2022-01-10:
> 问题是我今天编译的出的这个！

你 menuconfig 里面选了么？
好家伙你这真的是要手把手的教

jaychin2, 2022-01-10:
> > 问题是我今天编译的出的这个！
> 
> 你 menuconfig 里面选了么？ 好家伙你这真的是要手把手的教

那麻烦大佬告诉我一下位置。我看看选上没有。我是默认的状态

deplives, 2022-01-10:
> 

[image]

jaychin2, 2022-01-10:
> 

谢谢大佬

jaychin2, 2022-01-10:
> > 问题是我今天编译的出的这个！
> 
> 你 menuconfig 里面选了么？ 好家伙你这真的是要手把手的教

大佬，我勾了i915，但是还是有这个问题，是5.15的事吗？

deplives, 2022-01-10:
> > > 问题是我今天编译的出的这个！
> > 
> > 
> > 你 menuconfig 里面选了么？ 好家伙你这真的是要手把手的教
> 
> 大佬，我勾了i915，但是还是有这个问题，是5.15的事吗？

我没试过 5.15 我的 5.4 和 5.10 是没问题

jaychin2, 2022-01-10:
> > > > 问题是我今天编译的出的这个！
> > > 
> > > 
> > > 你 menuconfig 里面选了么？ 好家伙你这真的是要手把手的教
> > 
> > 大佬，我勾了i915，但是还是有这个问题，是5.15的事吗？
> 
> 我没试过 5.15 我的 5.4 和 5.10 是没问题

。。那我去试试5.4吧

closed 2022-01-10 by AmadeusGhost