coolsnowwolf/lede #8658: 使用在线Actions 编译MT7621包时出现错误

seaair-t, 2022-01-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
在使用L大的yml文件时修改了以下内容
 #
          # 在 cat >> .config <<EOF 到 EOF 之间粘贴你的编译配置, 需注意缩进关系
          # 例如:
          cat >> .config <<EOF
          CONFIG_TARGET_ramips=y
          CONFIG_TARGET_ramips_mt7621=y
          CONFIG_TARGET_MULTI_PROFILE=y
          CONFIG_TARGET_IMAGES_GZIP=y
          EOF
          #
          # ===============================================================
          #      
          make defconfig
详见：https://github.com/seaair-t/lede/blob/master/.github/workflows/openwrt-ci.yml
### (1) 具体问题

A：错误说见日志

### (2) 路由器型号和固件版本

A：在线编译MT7621

### (3) 详细日志

A：Signing package index...
bash: line 3: /home/runner/work/lede/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 3: /home/runner/work/lede/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 3: /home/runner/work/lede/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 3: /home/runner/work/lede/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 3: /home/runner/work/lede/lede/staging_dir/host/bin/usign: No such file or directory
bash: line 3: /home/runner/work/lede/lede/staging_dir/host/bin/usign: No such file or directory
make[2]: *** [package/Makefile:88: package/index] Error 127
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [Makefile:120: world] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

详见：https://github.com/seaair-t/lede/runs/4752080313?check_suite_focus=true

closed 2022-01-10 by AmadeusGhost

zhangguanzhang, 2022-10-04:
怎么解决的

mycloa, 2024-08-08:
怎么解决掉啊