coolsnowwolf/lede #8659: Add support for Xiaomi Redmi AX5 AX1800 RA67 (DVB4252CN/DVB4258GL) IPQ6018/AP-CP03-C1

TurboBlaze, 2022-01-09:
@coolsnowwolf 
can we add support for the Xiaomi Redmi AX5 AX1800 RA67 (DVB4252CN/DVB4258GL) IPQ6018/AP-CP03-C1
I have DTS from another device Xiaomi AX1800 - qcom-ipq60xx-cpxx.dts IPQ6018/AP-CP02-C1
https://gist.github.com/macromorgan/e3971e32a6d9e3379e2a8806f0a302c8
and I see we have these DTS https://github.com/coolsnowwolf/lede/tree/master/target/linux/ipq60xx/files/arch/arm64/boot/dts/qcom

Thanks

koloved, 2022-01-09:
all people say that 256 isn’t enough for this firmware.  isn't it?

testermania, 2022-01-09:
Currently, there are many devices that work with 128 MB of memory. Mi C1200 Router 4A Giga etc

Gingernut1978, 2022-01-09:
The problem is that ath11k, the wireless driver/firmware,  uses a lot of memory, not an OWRT restriction .  Even 512mb systems are seeing low memory issues.

testermania, 2022-01-10:
> @Gingernut1978 on the stock firmware via WiFi [image] and with memory all is ok, why?

is that stock wifi db power or  modified?

TurboBlaze, 2022-01-10:
@testermania it's Chinese stock version fw, but you can hack stock global fw and get the same dBm
`wl1 (WiFi 2.4) Tx-Power=28dBm
wl0 (WiFi 5) Tx-Power=26 dBm` 
Look at this video on the Magic iPerf tests https://youtu.be/65O5l9O67l4?t=423

Gingernut1978, 2022-01-10:
> @Gingernut1978 on the stock firmware via WiFi [image] and with memory all is ok, why?

Because Xiaomi have the source code and are a paying customer of Qualcomm. 

They should release their source code according to the GPL, but they don't.

Zurzar, 2022-01-15:
+1
add support for the Xiaomi Redmi AX5
Thanks

TurboBlaze, 2022-01-17:
`/target/linux/ipq60xx/files/arch/arm64/boot/dts/qcom/qcom-ipq6018-gl-ax1800.dts b/target/linux/ipq60xx/files/arch/arm64/boot/dts/qcom/qcom-ipq6018-gl-ax1800.dts`
is have issue with nand partition table.
This is the patch from **Dongming Han**
[PATCH] ipq60xx: fix gl-ax1800 nand partition table
https://forum.gl-inet.com/t/gl-inet-ax1800-openwrt-firmware/19447/6
[0001-ipq60xx-fix-gl-ax1800-nand-partition-table.zip](https://github.com/coolsnowwolf/lede/files/7882421/0001-ipq60xx-fix-gl-ax1800-nand-partition-table.zip)
6

Sora42, 2022-01-18:
> `/target/linux/ipq60xx/files/arch/arm64/boot/dts/qcom/qcom-ipq6018-gl-ax1800.dts b/target/linux/ipq60xx/files/arch/arm64/boot/dts/qcom/qcom-ipq6018-gl-ax1800.dts` is have issue with nand partition table!!!
> 
> This is the patch from **Dongming Han** [PATCH] ipq60xx: fix gl-ax1800 nand partition table [forum.gl-inet.com/t/gl-inet-ax1800-openwrt-firmware/19447/6](https://forum.gl-inet.com/t/gl-inet-ax1800-openwrt-firmware/19447/6) [0001-ipq60xx-fix-gl-ax1800-nand-partition-table.zip](https://github.com/coolsnowwolf/lede/files/7882421/0001-ipq60xx-fix-gl-ax1800-nand-partition-table.zip)

hello, if you need dts, u can use it from the stock kernel, right? or is it not possible in this case?

TurboBlaze, 2022-01-23:
> > 
> 
> hello, if you need dts, u can use it from the stock kernel, right? or is it not possible in this case?
I don't know. How to do this?

varoudis, 2022-01-29:
I would love to see Xiaomi AX1800 supported! Thank you!

sealmi, 2022-02-13:
I want a firmware for ax5 ax1800. Thanks

NishaYume, 2022-06-20:
News?

AbianTang, 2023-03-31:
+1
添加对小米 Redmi AX5 的支持
谢谢

coolsnowwolf, 2026-09-05:
这是 Redmi AX5/RA67 的新增硬件支持请求，但没有完整可启动 DTS、NAND/eMMC 分区、校准数据、固件刷写/恢复路径或硬件验证；不能直接复用另一台 IPQ6018 设备 DTS，当前无法安全合入。

closed 2026-09-05 by coolsnowwolf