coolsnowwolf/lede #8663: qbittorrent添加任务停止运行

xfmly, 2022-01-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
最新源码动态编译qbittorrent，一添加任务就停止运行。

### (1) 具体问题

A：qbittorrent添加任务就停止运行

### (2) 路由器型号和固件版本

A：竞斗云2.0／内核版本:5.4.162／R22.1.1

### (3) 详细日志

A：daemon.info procd: Instance qbittorrent::qbittorrent.main s in a crash loop 6 crashes, 1 seconds since last crash

WYC-2020, 2022-01-10:
我这边也是一样，试下静态库看可行不，静态库是不用编译直接下载现成的

musclemm, 2022-01-10:
> 我这边也是一样，试下静态库看可行不，静态库是不用编译直接下载现成的

我编译的时候选择 Static Build ，但是会报错，选择 Dynamic Build的时候可以编译成功。

check_data_file_clashes: Package qBittorrent-static wants to install file /home/ubuntu/lede/build_dir/target-x86_64_musl/root-x86/usr/bin/qbittorrent-nox
But that file is already provided by package * qbittorrent

opkg_install_cmd: Cannot install package qBittorrent-static.

wkq2004, 2022-01-10:
> 我编译的时候选择 Static Build ，但是会报错，选择 Dynamic Build的时候可以编译成功。
>
> check_data_file_clashes: Package qBittorrent-static wants to install file [...]

一样的情况，现在改成docker了，不管了。

WYC-2020, 2022-01-10:
有时间我gdb看下

Her0R0cket, 2022-01-11:
> 我编译的时候选择 Static Build ，但是会报错，选择 Dynamic Build的时候可以编译成功。
>
> check_data_file_clashes: Package qBittorrent-static wants to install file [...]
你这个要去把默认的qb设成N或M,不然合成固件的时候相同的文件会冲突

WYC-2020, 2022-01-11:
定位出来了，是boost和bt的库有冲突

[Switching to LWP 6725]
get_meta (p=p@entry=0x2da9000000000c <error: Cannot access memory at address 0x2da9000000000c>) at src/malloc/mallocng/meta.h:131
131	src/malloc/mallocng/meta.h: No such file or directory.
(gdb) bt
#0  get_meta (p=p@entry=0x2da9000000000c <error: Cannot access memory at address 0x2da9000000000c>) at src/malloc/mallocng/meta.h:131
#1  0x00007ffff7fad840 in __libc_free (p=0x2da9000000000c) at src/malloc/mallocng/free.c:105
#2  0x00007ffff7fa0585 in free (p=<optimized out>) at src/malloc/free.c:5
#3  0x00007ffff7a164d5 in boost::alignment::aligned_free (ptr=<optimized out>) at /home/wenyanchao/lede/staging_dir/target-x86_64_musl/usr/include/boost/align/detail/aligned_alloc.hpp:45
#4  0x00007ffff7a805df in boost::asio::aligned_delete (ptr=<optimized out>) at /home/wenyanchao/lede/staging_dir/target-x86_64_musl/usr/include/boost/asio/detail/memory.hpp:124
#5  boost::asio::detail::thread_info_base::allocate<boost::asio::detail::thread_info_base::default_tag> (this_thread=<optimized out>, size=size@entry=176, align=align@entry=16)
    at /home/wenyanchao/lede/staging_dir/target-x86_64_musl/usr/include/boost/asio/detail/thread_info_base.hpp:164
#6  0x00007ffff7a827bb in boost::asio::detail::thread_info_base::allocate (align=16, size=176, this_thread=<optimized out>)
    at /home/wenyanchao/lede/staging_dir/target-x86_64_musl/usr/include/boost/asio/detail/thread_info_base.hpp:123
#7  boost::asio::asio_handler_allocate (size=176) at /home/wenyanchao/lede/staging_dir/target-x86_64_musl/usr/include/boost/asio/impl/handler_alloc_hook.ipp:36
#8  boost_asio_handler_alloc_helpers::allocate<std::_Bind<void (libtorrent::lsd::*(std::shared_ptr<libtorrent::lsd>, std::_Placeholder<1>, libtorrent::digest32<160l>, int, int))(boost::system::error_code const&, libtorrent::digest32<160l> const&, int, int)> >(unsigned long, std::_Bind<void (libtorrent::lsd::*(std::shared_ptr<libtorrent::lsd>, std::_Placeholder<1>, libtorrent::digest32<160l>, int, int))(boost::system::error_code const&, libtorrent::digest32<160l> const&, int, int)>&, unsigned long) (align=8, h=..., s=176)
    at /home/wenyanchao/lede/staging_dir/target-x86_64_musl/usr/include/boost/asio/detail/handler_alloc_helpers.hpp:73
#9  boost::asio::detail::hook_allocator<std::_Bind<void (libtorrent::lsd::*(std::shared_ptr<libtorrent::lsd>, std::_Placeholder<1>, libtorrent::digest32<160l>, int, int))(boost::system::error_code const&, libtorrent::digest32<160l> const&, int, int)>, boost::asio::detail::wait_handler<std::_Bind<void (libtorrent::lsd::*(std::shared_ptr<libtorrent::lsd>, std::_Placeholder<1>, libtorrent::digest32<160l>, int, int))(boost::system::error_code const&, libtorrent::digest32<160l> const&, int, int)>, boost::asio::any_io_executor> >::allocate(unsigned long) (this=<synthetic pointer>, this=<synthetic pointer>, n=1)
    at /home/wenyanchao/lede/staging_dir/target-x86_64_musl/usr/include/boost/asio/detail/handler_alloc_helpers.hpp:131
#10 boost::asio::detail::wait_handler<std::_Bind<void (libtorrent::lsd::*(std::shared_ptr<libtorrent::lsd>, std::_Placeholder<1>, libtorrent::digest32<160l>, int, int))(boost::system::error_code const&, libtorrent::digest32<160l> const&, int, int)>, boost::asio::any_io_executor>::ptr::allocate(std::_Bind<void (libtorrent::lsd::*(std::shared_ptr<libtorrent::lsd>, std::_Placeholder<1>, libtorrent::digest32<160l>, int, int))(boost::system::error_code const&, libtorrent::digest32<160l> const&, int, int)>&) (handler=...)
    at /home/wenyanchao/lede/staging_dir/target-x86_64_musl/usr/include/boost/asio/detail/wait_handler.hpp:36
#11 boost::asio::detail::deadline_timer_service<boost::asio::detail::chrono_time_traits<std::chrono::_V2::system_clock, boost::asio::wait_traits<std::chrono::_V2::system_clock> > >::async_wait<std::_Bind<void (libtorrent::lsd::*(std::shared_ptr<libtorrent::lsd>, std::_Placeholder<1>, libtorrent::digest32<160l>, int, int))(boost::system::error_code const&, libtorrent::digest32<160l> const&, int, int)>, boost::asio::any_io_executor>(boost::asio::detail::deadline_timer_service<boost::asio::detail::chrono_time_traits<std::chrono::_V2::system_clock, boost::asio::wait_traits<std::chrono::_V2::system_clock> > >::implementation_type&, std::_Bind<void (libtorrent::lsd::*(std::shared_ptr<libtorrent::lsd>, std::_Placeholder<1>, libtorrent::digest32<160l>, int, int))(boost::system::error_code const&, libtorrent::digest32<160l> const&, int, int)>&, boost::asio::any_io_executor const&) (this=0x7ffff6f563a0, impl=..., handler=..., io_ex=...)
    at /home/wenyanchao/lede/staging_dir/target-x86_64_musl/usr/include/boost/asio/detail/deadline_timer_service.hpp:257
#12 0x00007ffff7a81ffa in boost::asio::basic_waitable_timer<std::chrono::_V2::system_clock, boost::asio::wait_traits<std::chrono::_V2::system_clock>, boost::asio::any_io_executor>::initiate_async_wait::operator()<std::_Bind<void (libtorrent::lsd::*(std::shared_ptr<libtorrent::lsd>, std::_Placeholder<1>, libtorrent::digest32<160l>, int, int))(boost::system::error_code const&, libtorrent::digest32<160l> const&, int, int)> >(std::_Bind<void (libtorrent::lsd::*(std::shared_ptr<libtorrent::lsd>, std::_Placeholder<1>, libtorrent::digest32<160l>, int, int))(boost::system::error_code const&, libtorrent::digest32<160l> const&, int, int)>&&) const (this=<optimized out>, this=<optimized out>, handler=...) at /home/wenyanchao/lede/staging_dir/target-x86_64_musl/usr/include/boost/asio/detail/io_object_impl.hpp:108
[... cut ...]
    at /home/wenyanchao/lede/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/x86_64-openwrt-linux-musl/include/c++/11.2.0/bits/std_thread.h:211
#41 0x00007ffff7122df2 in std::execute_native_thread_routine (__p=<optimized out>) at /home/wenyanchao/lede/build_dir/toolchain-x86_64_gcc-11.2.0_musl/gcc-11.2.0/libstdc++-v3/src/c++11/thread.cc:82
#42 0x00007ffff7fdc658 in start (p=0x7ffff6e29048) at src/thread/pthread_create.c:203
#43 0x00007ffff7fde884 in __clone () at src/thread/x86_64/clone.s:22
Backtrace stopped: frame did not save the PC
初步看内存方面有问题，bt官方也有类似的bug，说的是好像boost1.77以后再迭代器方面有问题

musclemm, 2022-01-11:
> 你这个要去把默认的qb设成N或M,不然合成固件的时候相同的文件会冲突

多谢，晚上我试试

xfmly, 2022-01-11:
> [image] 验证完毕，回滚boost到1.76就不会崩溃了，这个坐等队列修复了再升级boost不迟@xfmly @wkq2004

十分感谢，先回滚。

closed 2022-01-11 by xfmly

david6908, 2022-01-19:
> > [image] 验证完毕，回滚boost到1.76就不会崩溃了，这个坐等队列修复了再升级boost不迟@xfmly @wkq2004
> 
> 十分感谢，先回滚。

也碰到这个问题，请问如何回滚boost 1.76？