coolsnowwolf/lede #8668: 5.15是不是和fullconet不兼容？

xyskwjy, 2022-01-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：   17.052851] ------------[ cut here ]------------
[   17.057595] WARNING: CPU: 2 PID: 2096 at nf_conntrack_register_notifier+0x3b/0x40 [nf_conntrack]
[   17.066488] Modules linked in: xt_FULLCONENAT pppoe ppp_async pppox ppp_mppe ppp_generic ipt_REJECT ax88179_178a zstd xt_time xt_tcpudp xt_state xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_CT via_velocity usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_fsm ts_bm slhc rtl8150 r8168 r8152 r8125 pcnet32 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda mlx5_core mlx4_en mlx4_core macvlan lzo iptable_raw iptable_nat iptable_mangle iptable_filter ip_tables igc iavf i40e forcedeth e1000e crc_ccitt br_netfilter bnx2x bnx2 alx
[   17.066553]  8139too 8139cp configs evdev i2c_dev cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding ixgbe igbvf e1000 amd_xgbe mdio_devres dummy sit mdio tunnel4 ip_tunnel veth vfat fat ntfs3 dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax raid10 raid1 raid0 md_mod nls_utf8 nls_iso8859_1 nls_cp437 ena crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_ssse3 sha1_ssse3 sha1_generic jitterentropy_rng drbg hmac authenc arc4 crypto_acompress uas sdhci_pltfm igb exfat btrfs zstd_decompress zstd_compress xxhash xor raid6_pq lzo_decompress
[   17.154050]  lzo_compress libcrc32c button_hotplug ptp pps_core mii libphy cbc encrypted_keys trusted tpm oid_registry asn1_encoder asn1_decoder
[   17.253640] CPU: 2 PID: 2096 Comm: fw3 Not tainted 5.15.12 #0
[   17.259469] Hardware name: Default string Default string/MW-GMLK-2_5G4L, BIOS 5.13 12/10/2021
[   17.268083] RIP: 0010:nf_conntrack_register_notifier+0x3b/0x40 [nf_conntrack]
[   17.275327] Code: 5a 50 a0 e8 97 b7 4f e1 48 83 bb 10 09 00 00 00 75 18 4c 89 a3 10 09 00 00 48 c7 c7 a0 5a 50 a0 e8 ea b2 4f e1 5b 41 5c 5d c3 <0f> 0b eb e4 90 55 48 89 e5 53 48 89 fb 48 c7 c7 a0 5a 50 a0 e8 5c
[   17.294186] RSP: 0018:ffffc90002ad3af0 EFLAGS: 00010286
[   17.299483] RAX: 0000000000000000 RBX: ffffffff822fe680 RCX: 0000000000000000
[   17.306686] RDX: ffff888100fd6e00 RSI: ffffffffa0b2c450 RDI: ffffffffa0505aa0
[   17.313888] RBP: ffffc90002ad3b00 R08: 0000000000000000 R09: 0000000000000000
[   17.321090] R10: 0000000000000000 R11: 0000000000000007 R12: ffffffffa0b2c450
[   17.328290] R13: ffff88810666a2d8 R14: 0000000000000000 R15: 0000000000000000
[   17.335497] FS:  00007f334364dac8(0000) GS:ffff888277d00000(0000) knlGS:0000000000000000
[   17.343669] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   17.349486] CR2: 00007f334352e000 CR3: 00000001061a2000 CR4: 0000000000350ee0
[   17.356696] Call Trace:
[   17.359218]  
[   17.361394]  0xffffffffa0b2958a
[   17.364614]  xt_check_target+0x86/0x1b0 [x_tables]
[   17.369486]  ? xt_find_match+0x103/0x220 [x_tables]
[   17.374449]  ? xt_find_match+0x213/0x220 [x_tables]
[   17.379404]  ipt_unregister_table_pre_exit+0x2aa/0x420 [ip_tables]
[   17.385663]  ipt_alloc_initial_table+0x44c/0x600 [ip_tables]
[   17.391394]  ipt_register_table+0x9a9/0xdf8 [ip_tables]
[   17.396690]  ? __mod_lruvec_page_state+0x6f/0xa0
[   17.401383]  nf_setsockopt+0x50/0x80
[   17.405040]  ip_setsockopt+0xa0/0x1390
[   17.408862]  raw_setsockopt+0xa5/0xc0
[   17.412597]  sock_common_setsockopt+0x15/0x20
[   17.417025]  __sys_setsockopt+0x9c/0x130
[   17.421020]  __x64_sys_setsockopt+0x1f/0x30
[   17.425276]  do_syscall_64+0x3f/0x90
[   17.428926]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[   17.434055] RIP: 0033:0x7f3343617645
[   17.437706] Code: 83 c4 18 48 89 c7 e9 da 8b fd ff c3 c3 31 c0 c3 49 89 ca 50 48 63 d2 48 63 f6 48 63 ff 45 89 c0 45 31 c9 b8 36 00 00 00 0f 05 <48> 63 f8 e8 b3 8b fd ff 5a c3 48 63 f6 50 48 63 ff 45 31 d2 45 31
[   17.456557] RSP: 002b:00007ffc02310140 EFLAGS: 00000246 ORIG_RAX: 0000000000000036
[   17.464225] RAX: ffffffffffffffda RBX: 00007f334352d080 RCX: 00007f3343617645
[   17.471431] RDX: 0000000000000040 RSI: 0000000000000000 RDI: 000000000000000e
[   17.478634] RBP: 00007f33435489a0 R08: 0000000000002940 R09: 0000000000000000
[   17.485841] R10: 00007f334352d080 R11: 0000000000000246 R12: 00007f3343547810
[   17.493051] R13: 00007f334352f910 R14: 00007f334352f878 R15: 00007f3343535290
[   17.500259]  
[   17.502521] ---[ end trace e085dedb4adcd7f2 ]---

这报错，是不是fullconet的问题？

WYC-2020, 2022-01-12:
单这样看 看不出来 要gdb 调试看下才知道

xyskwjy, 2022-01-16:
> gdb 调试

换回5.10了……。没有这报错

xyskwjy, 2022-01-18:
> 

应该可以确定是fullconet的原因，如果没有勾选打开fullconet，没有那个提示。只有勾选打开fullconet才会有提示。至于GDB调试，我是小白，不太会。不过好像就算是有那提示，貌似也不影响使用。然后好像也是nat1

WYC-2020, 2022-01-19:
嗯看了下确实这样，我这边稍微修复了一下，原来崩溃是因为注册失败了，并且没有返回值可以判断，导致后面异常，我这里修复了返回值，并且增加了回调多次注册，崩溃是没问题了，其他功能自己测试一下，如果要用的话，拒绝吐槽哈
https://github.com/WYC-2020/lede/commit/87863b322720d9e6936cac33f2225585228ad91b @xyskwjy

xyskwjy, 2022-01-20:
> 嗯看了下确实这样，我这边稍微修复了一下，原来崩溃是因为注册失败了，并且没有返回值可以判断，导致后面异常，我这里修复了返回值，并且增加了回调多次注册，崩溃是没问题了，其他功能自己测试一下，如果要用的话，拒绝吐槽哈 [WYC-2020/lede@87863b3](https://github.com/WYC-2020/lede/commit/87863b322720d9e6936cac33f2225585228ad91b) @xyskwjy

谢谢大哥修复。

closed 2022-03-11 by xyskwjy