coolsnowwolf/lede #8669: kmod-phylink is missing dependencies for the following libraries: libphy.ko

vipboddy, 2022-01-10:
make[2]: Leaving directory '/mnt/sdb1/test/lede/package/firmware/prism54-firmware'
time: package/firmware/prism54-firmware/compile#0.21#0.02#0.22
make[2]: Entering directory '/mnt/sdb1/test/lede/package/kernel/linux'
Package kmod-phylink is missing dependencies for the following libraries:
libphy.ko
make[2]: *** [modules/netdevices.mk:123: /mnt/sdb1/test/lede/bin/targets/x86/64/packages/kmod-phylink_5.15.12-1_x86_64.ipk] Error 1
make[2]: Leaving directory '/mnt/sdb1/test/lede/package/kernel/linux'
time: package/kernel/linux/compile#2.41#0.20#2.59
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/mnt/sdb1/test/lede'

helmiau, 2022-01-11:
try to add `CONFIG_PHYLIB=y` to your config

WYC-2020, 2022-01-11:
[image]
kmod-phylink miss depends

WYC-2020, 2022-01-11:
没有问题的，这个libphy默认就是要编译的除非你自己吧他取消了 然后又去选择编译phylink 此时就会出现那个问题，我这边编译都有
[image]
如果要保证phylink编译默认phy也要编译，你在我上面截图哪里加一个依赖就行了，简单 @helmiau

WYC-2020, 2022-01-11:
@vipboddy 
[image]
 加个依赖就可以了

vipboddy, 2022-01-11:
> @vipboddy [image] 加个依赖就可以了

感谢大佬指点

closed 2022-03-29 by AmadeusGhost