coolsnowwolf/lede #8670: linux-ipq40xx 内核编译失败

stonegr, 2022-01-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
[image]

### (2) 路由器型号和固件版本

A：竞斗云2.0

### (3) 详细日志

A：
[image]

谢谢大佬

helmiau, 2022-01-11:
same as mine here

[image]

[image]

i think this error occured on `arm_cortex-a7+neon-vfpv4_musl_eabi` and `arm_arm1176jzf-s+vfp_musl_eabi` architecture (armv7)

stonegr, 2022-01-11:
> 

4.19内核目前编译正常

Cathgao, 2022-01-11:
同样问题
#https://github.com/Cathgao/GDOCK-Cath

```
2022-01-10T21:32:23.9841268Z make[5]: *** [Makefile:1100: vmlinux] Error 1
2022-01-10T21:32:23.9845418Z make[5]: Leaving directory '/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-5.4.162'
2022-01-10T21:32:23.9847571Z make[4]: *** [Makefile:26: /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-5.4.162/.modules] Error 2
2022-01-10T21:32:23.9849194Z make[4]: Leaving directory '/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/target/linux/ipq40xx'
2022-01-10T21:32:23.9858852Z make[3]: *** [Makefile:11: compile] Error 2
2022-01-10T21:32:23.9860191Z make[3]: Leaving directory '/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/target/linux'
2022-01-10T21:32:23.9861189Z time: target/linux/compile#916.39#105.78#538.54
2022-01-10T21:32:23.9862916Z     ERROR: target/linux failed to build.
2022-01-10T21:32:23.9875193Z make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
2022-01-10T21:32:23.9876590Z make[2]: Leaving directory '/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede'
2022-01-10T21:32:23.9879015Z make[1]: *** [target/Makefile:18: /home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_compile] Error 2
2022-01-10T21:32:23.9888826Z make[1]: Leaving directory '/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede'
2022-01-10T21:32:23.9896433Z make: *** [/home/runner/work/GDOCK-Cath/GDOCK-Cath/lede/include/toplevel.mk:230: world] Error 2
2022-01-10T21:32:23.9909275Z ##[error]Process completed with exit code 2.
2022-01-10T21:32:24.0045524Z Post job cleanup.
2022-01-10T21:32:24.3350046Z [command]/usr/bin/git version
2022-01-10T21:32:24.3421277Z git version 2.34.1
2022-01-10T21:32:24.3483582Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2022-01-10T21:32:24.3553923Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2022-01-10T21:32:24.4107382Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2022-01-10T21:32:24.4155480Z http.https://github.com/.extraheader
2022-01-10T21:32:24.4169206Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2022-01-10T21:32:24.4229700Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2022-01-10T21:32:24.4705280Z Cleaning up orphan processes
```

Cathgao, 2022-01-13:
10号以后已不再报错
[image]

stonegr, 2022-01-13:
> 10号以后已不再报错 [image]

我们好像就是10号编的，哈哈

closed 2022-01-17 by AmadeusGhost