coolsnowwolf/lede #8673: 想测试miniupnpd-2.2.3，需要修改那个地方可以进行

moonerick, 2022-01-12:
最近在5.15下minipnp不是很稳定，打算自己来测试各个版本的性能和稳定性，我应该修改那个配置文件可以完成编译文件的修改呢

WYC-2020, 2022-01-12:
改makefile不要说不知道，哈哈

moonerick, 2022-01-12:
> 改makefile不要说不知道，哈哈

这个我还是知道的，我在dl文件夹找到了源码压缩包，在lede\build_dir\target-x86_64_musl找到了解压缩的源码包，但是没有找到更上级的配置定义包，在大雕的库里面搜索，配置出现在coolsnowolf/package下的net文件夹里面，lean里面的没有找到

ZeaKyX, 2022-01-12:
> 配置出现在coolsnowolf/package下的net文件夹里面

这不是你已经帮你自己找到了么😂
https://github.com/coolsnowwolf/packages/blob/master/net/miniupnpd/Makefile

moonerick, 2022-01-13:
找到了，我去测试一下，看直接改版本是否可用，我看有补丁文件夹

  

  
 https://github.com/coolsnowwolf/packages/blob/master/net/miniupnpd/Makefile
 
&mdash;

tossp, 2022-01-13:
等待测试回馈

moonerick, 2022-01-13:
> 等待测试回馈

编译错误，代码如下：
make[3]: Entering directory '/home/sp/lede/feeds/packages/net/miniupnpd'
mkdir -p /home/sp/lede/dl
SHELL= flock /home/sp/lede/tmp/.miniupnpd-2.2.3.tar.gz.flock -c '       /home/sp/lede/scripts/download.pl "/home/sp/lede/dl" "miniupnpd-2.2.3.tar.gz" "f89c310ce9575183af3fec61af65e548f85114133df8caaaa9e204c13b7a9da5" "" "http://miniupnp.free.fr/files"    '
touch /home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3/.prepared_80a7834133f4e613964751aa2c1c6923_6664517399ebbbc92a37c5bb081b5c53_check
. /home/sp/lede/include/shell.sh; gzip -dc /home/sp/lede/dl/miniupnpd-2.2.3.tar.gz | tar -C /home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3
echo "OpenWrt" | tr \(\)\  _ >/home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3/os.openwrt
touch /home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3/.prepared_80a7834133f4e613964751aa2c1c6923_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3/.configured_*
rm -f /home/sp/lede/staging_dir/target-x86_64_musl/stamp/.miniupnpd_installed
(cd /home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3/./; if [ -x ./configure ]; then find /home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3/ -name config.guess | xargs -r chmod u+w; find /home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3/ -name config.guess | xargs -r -n1 cp --remove-destination /home/sp/lede/scripts/config.guess; find /home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3/ -name config.sub | xargs -r chmod u+w; find /home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3/ -name config.sub | xargs -r -n1 cp --remove-destination /home/sp/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3=miniupnpd-2.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3=miniupnpd-2.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3=miniupnpd-2.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/sp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/sp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/sp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/sp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/sp/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
Option not recognized : --target=x86_64-openwrt-linux
use -h option to display help
make[3]: *** [Makefile:84: /home/sp/lede/build_dir/target-x86_64_musl/miniupnpd-2.2.3/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/sp/lede/feeds/packages/net/miniupnpd'
time: package/feeds/packages/miniupnpd/compile#0.14#0.02#0.13
    ERROR: package/feeds/packages/miniupnpd failed to build.

moonerick, 2022-01-14:
两个源码包里面的内容不一样，单独只是修改里面的makefile文件无法编译

moonerick, 2022-01-15:
> 等待测试回馈

使用openwrt官方package里面的配置进行编译，成功编译，默认开启ipv6正常，由于我自己使用的是双wan，排序时pppoe-iptv先于pppoe-wan，出现默认ext_ifname重启后重新被配置为pppoe-iptv的问题，其它方面暂时没有问题

closed 2022-01-15 by moonerick