coolsnowwolf/lede #8675: 大佬，ath79里的机型编译出来的固件好多无法启动？？

jacelift, 2022-01-12:
之前的版本在ar71xx机型里编译出来的固件都没问题，也可以正常启动；而目前的版本里没有了ar71xx选项，但在ath79里有同样的机型，编译出来的固件是无法启动的，是固件头的问题？还是内核太新了？比如：tl-wdr6500-v2，tl-wr710n-v1我都试过了，都无法启动，有些是无限重启，有些是直接进breed。无法正常启动，换成以前ar71xx编译的固件就没问题！！请大佬指点！！

coolsnowwolf, 2022-01-12:
有 TTL 日志吗? 否则也不知道是哪里的问题

WYC-2020, 2022-01-12:
这车翻了？

jacelift, 2022-01-13:
> 
这是wdr6500-v2的TTL日志：
[   13.728506] jffs2: Further such events for this erase block will not be printed
[   13.745071] jffs2: Old JFFS2 bitmask found at 0x00544d58
[   13.750572] jffs2: You cannot use older JFFS2 filesystems with newer kernels

[   14.858171] jffs2: Further such events for this erase block will not be printed
[   14.894615] jffs2: Cowardly refusing to erase blocks on filesystem with no valid JFFS2 nodes
[   14.903343] jffs2: empty_blocks 0, bad_blocks 0, c->nr_blocks 93
[   14.909646] VFS: Cannot open root device "(null)" or unknown-block(31,3): error -5
[   14.917467] Please append a correct "root=" boot option; here are the available partitions:
[   14.926092] 1f00              64 mtdblock0 
[   14.926094]  (driver?)
[   14.932853] 1f01            8000 mtdblock1 
[   14.932855]  (driver?)
[   14.939607] 1f02            2039 mtdblock2 
[   14.939609]  (driver?)
[   14.946353] 1f03            5960 mtdblock3 
[   14.946354]  (driver?)
[   14.953110] 1f04              64 mtdblock4 
[   14.953112]  (driver?)
[   14.959865] 1f05              64 mtdblock5 
[   14.959867]  (driver?)
[   14.966610] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
[   14.975233] Rebooting in 1 seconds..
[无限重启TTL-0.log](https://github.com/coolsnowwolf/lede/files/7863879/TTL-0.log)
[无限重启TTL-1.log](https://github.com/coolsnowwolf/lede/files/7863881/TTL-1.log)

jacelift, 2022-01-14:
> 有 TTL 日志吗? 否则也不知道是哪里的问题

这是wdr6500-v2的TTL日志：
[   13.728506] jffs2: Further such events for this erase block will not be printed
[   13.745071] jffs2: Old JFFS2 bitmask found at 0x00544d58
[   13.750572] jffs2: You cannot use older JFFS2 filesystems with newer kernels

[   14.858171] jffs2: Further such events for this erase block will not be printed
[   14.894615] jffs2: Cowardly refusing to erase blocks on filesystem with no valid JFFS2 nodes
[   14.903343] jffs2: empty_blocks 0, bad_blocks 0, c->nr_blocks 93
[   14.909646] VFS: Cannot open root device "(null)" or unknown-block(31,3): error -5
[   14.917467] Please append a correct "root=" boot option; here are the available partitions:
[   14.926092] 1f00              64 mtdblock0 
[   14.926094]  (driver?)
[   14.932853] 1f01            8000 mtdblock1 
[   14.932855]  (driver?)
[   14.939607] 1f02            2039 mtdblock2 
[   14.939609]  (driver?)
[   14.946353] 1f03            5960 mtdblock3 
[   14.946354]  (driver?)
[   14.953110] 1f04              64 mtdblock4 
[   14.953112]  (driver?)
[   14.959865] 1f05              64 mtdblock5 
[   14.959867]  (driver?)
[   14.966610] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
[   14.975233] Rebooting in 1 seconds..

TTL日志文件附件：
[无限重启TTL-0.log](https://github.com/coolsnowwolf/lede/files/7863879/TTL-0.log)
[无限重启TTL-1.log](https://github.com/coolsnowwolf/lede/files/7863881/TTL-1.log)

coolsnowwolf, 2022-01-14:
大概知道了

jacelift, 2022-01-14:
> 大概知道了

大佬，有解决方法吗？

iotdoc58, 2022-01-19:
大佬这个有解决办法吗？ 一样的无法使用

iotdoc58, 2022-01-19:
> > 大概知道了
> 
> 大佬，有解决方法吗？

iotdoc58, 2022-01-20:
ress any key to interrupt autoboot ... 0

Trying to boot firmware from 0x9f020000 ...
TP-LINK firmware image header detected.
    Image Name:   08000001
    Data Size:    2081447 Bytes
    Load Address: 80060000
    Entry Point:  80060000
Uncompressing data (LZMA) ... done.
Flushing cache ... done.

Starting kernel at 0x80060000...

[    0.000000] Linux version 5.4.172 (dylan@vm12345-0509100) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r                        4051-1bf58b7e1)) #0 Mon Jan 17 14:31:40 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is TP-Link TL-WR800N v2
[    0.000000] SoC: Qualcomm Atheros QCA9533 ver 1 rev 1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=0004e4e0
[    0.000000] Readback ErrCtl register=0004e4e0
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 57092K/65536K available (4999K kernel code, 187K rwdata, 1140K rodata, 1216K                         init, 196K bss, 8444K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] random: get_random_bytes called from start_kernel+0x358/0x54c with crng_init=0
[    0.000000] CPU clock: 550.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6950037990 n                        s
[    0.000010] sched_clock: 32 bits at 275MHz, resolution 3ns, wraps every 7809031678ns
[    0.008884] Calibrating delay loop... 366.18 BogoMIPS (lpj=1830912)
[    0.075949] pid_max: default: 32768 minimum: 301
[    0.081381] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.089704] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.103846] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 191126044                        62750000 ns
[    0.115082] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.122948] pinctrl core: initialized pinctrl subsystem
[    0.130202] NET: Registered protocol family 16
[    0.171254] workqueue: max_active 576 requested for napi_workq is out of range, clamping between                         1 and 512
[    0.186422] clocksource: Switched to clocksource MIPS
[    0.193685] NET: Registered protocol family 2
[    0.198926] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.207999] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.217596] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.226307] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.234346] TCP: Hash tables configured (established 1024 bind 1024)
[    0.241756] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.249251] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.257542] NET: Registered protocol family 1
[    0.262517] PCI: CLS 0 bytes, default 32
[    0.271722] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.287721] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.294343] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red                         Hat, Inc.
[    0.325644] pinctrl-single 1804002c.pinmux: 576 pins, size 72
[    0.333446] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[    0.341620] printk: console [ttyS0] disabled
[    0.346609] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 1562500) is a 16550A
[    0.356435] printk: console [ttyS0] enabled
[    0.356435] printk: console [ttyS0] enabled
[... 70 lines cut ...]
[    4.932470] [<8021955c>] squashfs_cache_get+0x230/0x32c
[    4.937885] [<8018e368>] mem_cgroup_try_charge+0x7c/0x2ac
[    4.943486] [<8012ea74>] __add_to_page_cache_locked+0x1bc/0x2dc
[    4.949600] [<8021ae4c>] squashfs_readpage+0x4c8/0x7b4
[    4.954930] [<8015728c>] workingset_update_node+0x0/0x84
[    4.960423] [<8012ec54>] add_to_page_cache_lru+0xa0/0x120
[    4.966006] [<801390a0>] read_pages+0x118/0x168
[    4.970693] [<801a6634>] path_lookupat+0x144/0x26c
[    4.975645] [<801392b4>] __do_page_cache_readahead+0x1c4/0x20c
[    4.981665] [<8012ed38>] pagecache_get_page+0x64/0x3f8
[    4.986971] [<8012f5c4>] filemap_fault+0x4f8/0x9a0
[    4.991922] [<80536e5c>] xas_find+0x19c/0x1e4
[    4.996429] [<80165018>] mmap_region+0x18c/0x6fc
[    5.001198] [<8012c0b8>] filemap_map_pages+0x70/0x3ec
[    5.006419] [<80159a44>] __do_fault+0x3c/0x154
[    5.011016] [<8015e9b4>] handle_mm_fault+0x7c0/0xd00
[    5.016158] [<80071c10>] do_page_fault+0x108/0x4a8
[    5.021106] [<80162db4>] ksys_mmap_pgoff+0xd0/0x134
[    5.026147] [<8007701c>] tlb_do_page_fault_0+0x10c/0x114
[    5.031637]
[    5.033168] Code: (Bad address in epc)
[    5.037040]
[    5.038614] ---[ end trace 4aef6a3fc001094e ]---
[    5.043379] Kernel panic - not syncing: Fatal exception
[    5.048776] Rebooting in 1 seconds..

kkjoy, 2022-03-08:
6500v2 貌似新的分支里面没有适配 所以只能使用旧得版本

coolsnowwolf, 2026-09-05:
ath79 设备从 ar71xx 迁移后无法启动需要具体设备 DTS、镜像格式和串口日志；仅列举型号不足以安全修改，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf