coolsnowwolf/lede #8676: musl升级到1.2.2后fork-daapd（现在叫owntone）运行不正常，退回1.1.24就正常了

monsterzzzz, 2022-01-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：musl升级到1.2.2后fork-daapd（现在叫owntone）运行崩溃，musl退回1.1.24就正常了

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：
Wed Jan 12 16:55:00 2022 kern.info kernel: [  115.875793] httpd[6333]: segfault at ffffffff92181340 ip 00007efd94f7c2df sp 00007efd921c72d8 error 5 in libc.so[7efd94f3f000+49000]
Wed Jan 12 16:55:00 2022 kern.info kernel: [  115.876777] Code: f8 48 83 fa 08 72 14 f7 c7 07 00 00 00 74 0c a4 48 ff ca f7 c7 07 00 00 00 75 f4 48 89 d1 48 c1 e9 03 f3 48 a5 83 e2 07 74 05 <a4> ff ca 75 fb c3 48 89 f8 48 29 f0 48 39 d0 0f 83 bf ff ff ff 48

WYC-2020, 2022-01-12:
十有八九和time_t 类型变化有关，这插件听都没听过 不修了

WYC-2020, 2022-01-12:
刚我看我修复的库里面 有这个库的修复patch ，自己去拉去，前段时间专门修复musl 1.2的bug，修复了那些 我都懒得看了 @monsterzzzz

monsterzzzz, 2022-01-12:
> 十有八九和time_t 类型变化有关，这插件听都没听过 不修了

这个插件挺好的，可以在路由上放音乐

monsterzzzz, 2022-01-12:
> 刚我看我修复的库里面 有这个库的修复patch ，自己去拉去，前段时间专门修复musl 1.2的bug，修复了那些 我都懒得看了 @monsterzzzz

提交到官方修复吧，现在用官方的源也是崩溃

monsterzzzz, 2022-01-12:
> 刚我看我修复的库里面 有这个库的修复patch ，自己去拉去，前段时间专门修复musl 1.2的bug，修复了那些 我都懒得看了 @monsterzzzz

打了你的patch，依然运行时崩溃

WYC-2020, 2022-01-12:
@monsterzzzz 有时间再看下

monsterzzzz, 2022-01-13:
> @monsterzzzz 有时间再看下

这个补丁吗？
https://github.com/WYC-2020/packages/blob/master/sound/forked-daapd/patches/0001-fix-musl-1.2.x-build.patch
没什么变化啊

monsterzzzz, 2022-01-13:
之前一直拉代码再make，用musl 1.1.24正常，如果make clean再make，用musl一样报错

WYC-2020, 2022-01-13:
出错位置
[image]
这个要具体分析一下，暂时你加一个--disable-webinterface 先用看下 @monsterzzzz

WYC-2020, 2022-01-13:
@monsterzzzz  你这个bug和https://github.com/owntone/owntone-server/issues/885 这个类似

monsterzzzz, 2022-01-14:
> @monsterzzzz 你这个bug和[owntone/owntone-server#885](https://github.com/owntone/owntone-server/issues/885) 这个类似

加了--disable-webinterface之后确实不崩溃了，但在网页上控制的功能也没了

WYC-2020, 2022-01-14:
@monsterzzzz 只能暂时这样用了，我看这库官网有人再去除antl这个库的依赖，到时候你在试下吧

WYC-2020, 2022-01-18:
@monsterzzzz 我这边修复了，自己试下哈
[image]
依赖的相关库都看下我这边是否有对应的patch哈
https://github.com/WYC-2020/packages/commit/f377d97933598f42182cd9d1df6e93a0f14a5b42

monsterzzzz, 2022-01-30:
官方升级到了28.3，用bison flex代替了libantlr3，这边makefile需要改一下吗

monsterzzzz, 2022-01-30:
> @monsterzzzz 我这边修复了，自己试下哈 [image] 依赖的相关库都看下我这边是否有对应的patch哈 [WYC-2020/packages@f377d97](https://github.com/WYC-2020/packages/commit/f377d97933598f42182cd9d1df6e93a0f14a5b42)

打了这个patch之后确实没问题了，感谢

catsimple, 2022-06-13:
https://github.com/catsimple/packages
fork-daapd启动失败的可以编译这个，fork-daapd已经年久失修，建议同步到owntone

closed 2022-07-25 by aiamadeus