coolsnowwolf/lede #8681: 找到一个适配移动端很好的主题，能加入备选吗？

Pacio, 2022-01-13:
https://github.com/thinktip/luci-theme-neobird
我自己手动加入编译的话，编译一切正常，软件包也显示已安装，但是“语言和界面”里面不显示该主题

WYC-2020, 2022-01-13:
这玩意作者自己都说安卓没测，并且列出了很多问题，最起码等测试稳定了才决定加不加入，现在加入 要是又遇到问题，有是一堆issue

mitsukileung, 2022-01-13:
这个主题我试过编译进去，但好像显示不出来，不知道是什么鬼

mitsukileung, 2022-01-13:
> https://github.com/thinktip/luci-theme-neobird 我自己手动加入编译的话，编译一切正常，软件包也显示已安装，但是“语言和界面”里面不显示该主题

情况和你一样，不知为啥。不知道是不是少了什么依赖

Pacio, 2022-01-13:
> > https://github.com/thinktip/luci-theme-neobird 我自己手动加入编译的话，编译一切正常，软件包也显示已安装，但是“语言和界面”里面不显示该主题
> 
> 情况和你一样，不知为啥。不知道是不是少了什么依赖

不是少了依赖，如果少依赖编译的时候就会报错，我感觉就是bug，但是我单独编译出的主题ipk是可以正常使用的
https://wwi.lanzouw.com/iEGk1yqbqna

mitsukileung, 2022-01-13:
> > > https://github.com/thinktip/luci-theme-neobird 我自己手动加入编译的话，编译一切正常，软件包也显示已安装，但是“语言和界面”里面不显示该主题
> > 
> > 
> > 情况和你一样，不知为啥。不知道是不是少了什么依赖
> 
> 不是少了依赖，如果少依赖编译的时候就会报错，我感觉就是bug，但是我单独编译出的主题ipk是可以正常使用的 https://wwi.lanzouw.com/iEGk1yqbqna

单独编译出来也不行，安装后还是一样的情况

Pacio, 2022-01-13:
> > > > https://github.com/thinktip/luci-theme-neobird 我自己手动加入编译的话，编译一切正常，软件包也显示已安装，但是“语言和界面”里面不显示该主题
> > > 
> > > 
> > > 情况和你一样，不知为啥。不知道是不是少了什么依赖
> > 
> > 
> > 不是少了依赖，如果少依赖编译的时候就会报错，我感觉就是bug，但是我单独编译出的主题ipk是可以正常使用的 https://wwi.lanzouw.com/iEGk1yqbqna
> 
> 单独编译出来也不行，安装后还是一样的情况

不是吧，我单独编译出的ipk直接安装就用上了

mitsukileung, 2022-01-13:
> > > > > https://github.com/thinktip/luci-theme-neobird 我自己手动加入编译的话，编译一切正常，软件包也显示已安装，但是“语言和界面”里面不显示该主题
> > > > 
> > > > 
> > > > 情况和你一样，不知为啥。不知道是不是少了什么依赖
> > > 
> > > 
> > > 不是少了依赖，如果少依赖编译的时候就会报错，我感觉就是bug，但是我单独编译出的主题ipk是可以正常使用的 https://wwi.lanzouw.com/iEGk1yqbqna
> > 
> > 
> > 单独编译出来也不行，安装后还是一样的情况
> 
> 不是吧，我单独编译出的ipk直接安装就用上了

我也试过，还是不行，用的是5.15内核

Pacio, 2022-01-13:
> > > > > > https://github.com/thinktip/luci-theme-neobird 我自己手动加入编译的话，编译一切正常，软件包也显示已安装，但是“语言和界面”里面不显示该主题
> > > > > 
> > > > > 
> > > > > 情况和你一样，不知为啥。不知道是不是少了什么依赖
> > > > 
> > > > 
> > > > 不是少了依赖，如果少依赖编译的时候就会报错，我感觉就是bug，但是我单独编译出的主题ipk是可以正常使用的 https://wwi.lanzouw.com/iEGk1yqbqna
> > > 
> > > 
> > > 单独编译出来也不行，安装后还是一样的情况
> > 
> > 
> > 不是吧，我单独编译出的ipk直接安装就用上了
> 
> 我也试过，还是不行，用的是5.15内核
内核兼容性问题，我是5.10

thinktip, 2022-01-13:
已经修复了，第一次编译，第一次用github，自己漏了一个文件，现在再试吧

closed 2022-01-13 by Pacio