coolsnowwolf/lede #8684: 关于更新5.15内核之后shortcut-fe编译一直报错的问题

zd911222, 2022-01-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：更新5.15内核之后编译一直卡在shortcut-fe，以前5.10没这问题

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：make -j1 V=sc获取的日志如下：
make[3]: Entering directory '/home/zd/lede/package/lean/shortcut-fe/shortcut-fe'
rm -f /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built
touch /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built_check
make -j1 -C "/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12" KCFLAGS="-fmacro-prefix-map=/home/zd/lede/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/zd/lede/staging_dir/host/include -I/home/zd/lede/staging_dir/hostpkg/include -I/home/zd/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Jan 13 02:11:31 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/zd/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/zd/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=1  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8125.symvers /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8152.symvers /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8168.symvers /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/gl-mifi-mcu.symvers" KERNELRELEASE=5.15.12  M="/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe" EXTRA_CFLAGS+="-DSFE_SUPPORT_IPV6" SFE_SUPPORT_IPV6=y  modules
make[4]: Entering directory '/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
make -f ./scripts/Makefile.build obj=/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe \
single-build= \
need-builtin=1 need-modorder=1
  x86_64-openwrt-linux-musl-gcc -Wp,-MMD,/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.sfe_ipv4.o.d  -nostdinc -isystem /home/zd/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -O2 -fno-caller-saves --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -g -fno-var-tracking -femit-struct-debug-baseonly -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -fmacro-prefix-map=/home/zd/lede/build_dir/target-x86_64_musl=target-x86_64_musl -Werror -Wall -DSFE_SUPPORT_IPV6  -DMODULE  -DKBUILD_BASENAME='"sfe_ipv4"' -DKBUILD_MODNAME='"shortcut_fe"' -D__KBUILD_MODNAME=kmod_shortcut_fe -c -o /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.o /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c
/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_udp':
/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c:1477:5: error: 'struct sk_buff' has no member named 'fast_forwarded'
  skb->fast_forwarded = 1;
     ^~
/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_tcp':
/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c:2050:5: error: 'struct sk_buff' has no member named 'fast_forwarded'
  skb->fast_forwarded = 1;
     ^~
make[5]: *** [scripts/Makefile.build:277: /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.o] Error 1
make[4]: *** [Makefile:1868: /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe] Error 2
make[4]: Leaving directory '/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
make[3]: *** [Makefile:87: /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built] Error 2
make[3]: Leaving directory '/home/zd/lede/package/lean/shortcut-fe/shortcut-fe'
time: package/lean/shortcut-fe/shortcut-fe/compile#0.52#0.24#0.74
    ERROR: package/lean/shortcut-fe/shortcut-fe failed to build.
make[2]: *** [package/Makefile:116: package/lean/shortcut-fe/shortcut-fe/compile] Error 1
make[2]: Leaving directory '/home/zd/lede'
make[1]: *** [package/Makefile:110: /home/zd/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zd/lede'
make: *** [/home/zd/lede/include/toplevel.mk:230：world] 错误 2

lxl6125, 2022-01-13:
[image]

ZGNCZMY, 2022-01-13:
不支持啊

ccblinda, 2022-01-13:
515不支持sfe

WYC-2020, 2022-01-17:
那就顺手自己关闭issue撒，非要等别人来关闭

WYC-2020, 2022-01-20:
我修改了一下可以编译了5.15， 用起来没什么问题

zd911222, 2022-02-01:
> 我修改了一下可以编译了5.15， 用起来没什么问题

我刚刚有拉了一下大雕的库去编译还是出错，大佬说的可以没什么问题的是你名下的那个LEDE库吗？

现在编译报错和之前一致：

`make[3]: Entering directory '/home/zd/lede/package/lean/shortcut-fe/shortcut-fe'
rm -f /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built
touch /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built_check
make -j1 -C "/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.18" KCFLAGS="-fmacro-prefix-map=/home/zd/lede/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/zd/lede/staging_dir/host/include -I/home/zd/lede/staging_dir/hostpkg/include -I/home/zd/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Feb  1 02:50:41 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/zd/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/zd/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=1  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8125.symvers /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8152.symvers /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8168.symvers /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/gl-mifi-mcu.symvers" KERNELRELEASE=5.15.18  M="/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe" EXTRA_CFLAGS+="-DSFE_SUPPORT_IPV6" SFE_SUPPORT_IPV6=y  modules
make[4]: Entering directory '/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.18'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
make -f ./scripts/Makefile.build obj=/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe \
single-build= \
need-builtin=1 need-modorder=1
  x86_64-openwrt-linux-musl-gcc -Wp,-MMD,/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.sfe_ipv4.o.d  -nostdinc -isystem /home/zd/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -O2 -fno-caller-saves --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -g -fno-var-tracking -femit-struct-debug-baseonly -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -fmacro-prefix-map=/home/zd/lede/build_dir/target-x86_64_musl=target-x86_64_musl -Werror -Wall -DSFE_SUPPORT_IPV6  -DMODULE  -DKBUILD_BASENAME='"sfe_ipv4"' -DKBUILD_MODNAME='"shortcut_fe"' -D__KBUILD_MODNAME=kmod_shortcut_fe -c -o /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.o /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c
/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_udp':
/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c:1477:5: error: 'struct sk_buff' has no member named 'fast_forwarded'
  skb->fast_forwarded = 1;
     ^~
/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_tcp':
/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.c:2050:5: error: 'struct sk_buff' has no member named 'fast_forwarded'
  skb->fast_forwarded = 1;
     ^~
make[5]: *** [scripts/Makefile.build:277: /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.o] Error 1
make[4]: *** [Makefile:1868: /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe] Error 2
[... cut ...]
make[3]: Leaving directory '/home/zd/lede/package/lean/shortcut-fe/shortcut-fe'
time: package/lean/shortcut-fe/shortcut-fe/compile#0.27#0.78#1.03
    ERROR: package/lean/shortcut-fe/shortcut-fe failed to build.
make[2]: *** [package/Makefile:116: package/lean/shortcut-fe/shortcut-fe/compile] Error 1
make[2]: Leaving directory '/home/zd/lede'
make[1]: *** [package/Makefile:110: /home/zd/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zd/lede'
make: *** [/home/zd/lede/include/toplevel.mk:230：world] 错误 2
`

WYC-2020, 2022-02-06:
@zd911222 是我仓库下面那个，我没提pr

zd911222, 2022-02-06:
> 

大佬，我直接下载了你的仓库的LEDE，并覆盖了原package文件夹，在menuconfig里面打开了turboacc，以及里面的shortcut-fe
多线程编译shortcut-fe报错
单线程编译加V=s 看到了 fullconenat 报错……

make[3]: Entering directory '/home/zd/lede/package/lean/openwrt-fullconenat'
mkdir -p /home/zd/lede/dl
SHELL= flock /home/zd/lede/tmp/.fullconenat-2021-12-04-7fe02604.tar.xz.flock -c '  	/home/zd/lede/scripts/download.pl "/home/zd/lede/dl" "fullconenat-2021-12-04-7fe02604.tar.xz" "857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" "" || (  /home/zd/lede/scripts/dl_github_archive.py --dl-dir="/home/zd/lede/dl" --url="https://github.com/llccd/netfilter-full-cone-nat.git" --version="7fe02604e073945a3b081c18efbb4d3c799a31f6" --subdir="fullconenat-2021-12-04-7fe02604" --source="fullconenat-2021-12-04-7fe02604.tar.xz" --hash="857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/zd/lede/tmp/dl && cd /home/zd/lede/tmp/dl && rm -rf fullconenat-2021-12-04-7fe02604 && [ \! -d fullconenat-2021-12-04-7fe02604 ] && git clone  https://github.com/llccd/netfilter-full-cone-nat.git fullconenat-2021-12-04-7fe02604 && (cd fullconenat-2021-12-04-7fe02604 && git checkout 7fe02604e073945a3b081c18efbb4d3c799a31f6 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fullconenat-2021-12-04-7fe02604 && git log -1 --format='\''@%ct'\''` && rm -rf fullconenat-2021-12-04-7fe02604/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fullconenat-2021-12-04-7fe02604 | 	xz -zc -7e > /home/zd/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz && mv /home/zd/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz /home/zd/lede/dl/ && rm -rf fullconenat-2021-12-04-7fe02604; );  )    '
rm -f /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.built
touch /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.built_check
make -j1 -C "/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.19" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" M="/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604" EXTRA_CFLAGS="" modules
make[4]: Entering directory '/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.19'
  CC [M]  /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/xt_FULLCONENAT.o
/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/xt_FULLCONENAT.c: In function 'fullconenat_tg_check':
/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/xt_FULLCONENAT.c:1260:9: error: void value not ignored as it ought to be
     if (nf_conntrack_register_notifier(par->net, &ct_event_notifier) == 0) {
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[5]: *** [scripts/Makefile.build:277: /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/xt_FULLCONENAT.o] Error 1
make[4]: *** [Makefile:1868: /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604] Error 2
make[4]: Leaving directory '/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.19'
make[3]: *** [Makefile:70: /home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.built] Error 2
make[3]: Leaving directory '/home/zd/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.62#0.09#0.70
    ERROR: package/lean/openwrt-fullconenat failed to build.
make[2]: *** [package/Makefile:116: package/lean/openwrt-fullconenat/compile] Error 1
make[2]: Leaving directory '/home/zd/lede'
make[1]: *** [package/Makefile:110: /home/zd/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zd/lede'
make: *** [/home/zd/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-02-06:
@zd911222 什么覆盖不覆盖？你直接克隆clone按正常编译就行，不要去私自改源，源更新完毕，安装-f -a,我今天才编译了没问题

zd911222, 2022-02-13:
> @zd911222 什么覆盖不覆盖？你直接克隆clone按正常编译就行，不要去私自改源，源更新完毕，安装-f -a,我今天才编译了没问题

大佬，我直接拉了你的库，然后编译。现在提示
ERROR: module '/home/zd/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.100/lib/asn1_encoder.ko' is missing.
make[3]: *** [modules/lib.mk:283: /home/zd/lede/bin/targets/x86/64/packages/kmod-asn1-encoder_5.10.100-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/zd/lede/package/kernel/linux'
time: package/kernel/linux/compile#2.92#0.39#2.95
    ERROR: package/kernel/linux failed to build.

我查了一下，之前有人提过这个问题：https://github.com/coolsnowwolf/lede/issues/8551，但是好像没有一个合适的解决方案，应当如何处理

closed 2022-02-13 by zd911222