coolsnowwolf/lede #8695: 最新编译出现以下错误提示

osxdiy, 2022-01-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
 CC [M]  net/netfilter/nf_conntrack_netlink.o
net/netfilter/nf_conntrack_netlink.c:3773:3: error: 'struct notifier_block' has no member named 'exp_event'
  .exp_event = ctnetlink_expect_event,
   ^~~~~~~~~
net/netfilter/nf_conntrack_netlink.c:3773:15: error: initialization of 'struct notifier_block *' from incompatible pointer type 'int (*)(unsigned int,  const struct nf_exp_event *)' [-Werror=incompatible-pointer-types]
  .exp_event = ctnetlink_expect_event,
               ^~~~~~~~~~~~~~~~~~~~~~
net/netfilter/nf_conntrack_netlink.c:3773:15: note: (near initialization for 'ctnl_notifier.next')
net/netfilter/nf_conntrack_netlink.c: In function 'ctnetlink_net_pre_exit':
net/netfilter/nf_conntrack_netlink.c:3880:2: error: too few arguments to function 'nf_conntrack_unregister_notifier'
  nf_conntrack_unregister_notifier(net);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/net/netfilter/nf_conntrack_core.h:18,
                 from net/netfilter/nf_conntrack_netlink.c:38:
./include/net/netfilter/nf_conntrack_ecache.h:88:12: note: declared here
 extern int nf_conntrack_unregister_notifier(struct net *net, struct notifier_block *nb);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:277: recipe for target 'net/netfilter/nf_conntrack_netlink.o' failed
make[7]: *** [net/netfilter/nf_conntrack_netlink.o] Error 1
scripts/Makefile.build:540: recipe for target 'net/netfilter' failed
make[6]: *** [net/netfilter] Error 2
Makefile:1868: recipe for target 'net' failed
make[5]: *** [net] Error 2
make[5]: Leaving directory '/home/lubaoo/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
Makefile:27: recipe for target '/home/lubaoo/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.modules' failed
make[4]: *** [/home/lubaoo/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.modules] Error 2
make[4]: Leaving directory '/home/lubaoo/lede/target/linux/x86'
Makefile:11: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/lubaoo/lede/target/linux'
time: target/linux/compile#298.02#37.26#334.37
    ERROR: target/linux failed to build.
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 1
make[2]: Leaving directory '/home/lubaoo/lede'
target/Makefile:18: recipe for target '/home/lubaoo/lede/staging_dir/target-x86_64_musl/stamp/.target_compile' failed
make[1]: *** [/home/lubaoo/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/lubaoo/lede'
/home/lubaoo/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2
A：

### (2) 路由器型号和固件版本

A：n3150 
lean 5.15.12

### (3) 详细日志

A：

zcluo, 2022-01-15:
net/netfilter/nf_conntrack_netlink.c:3773:10: error: 'struct notifier_block' has no member named 'exp_event'
 3773 |         .exp_event = ctnetlink_expect_event,
      |          ^~~~~~~~~
net/netfilter/nf_conntrack_netlink.c:3773:22: error: initialization of 'struct notifier_block *' from incompatible pointer type 'int (*)(unsigned int,  const struct nf_exp_event *)' [-Werror=incompatible-pointer-types]
 3773 |         .exp_event = ctnetlink_expect_event,
      |                      ^~~~~~~~~~~~~~~~~~~~~~
net/netfilter/nf_conntrack_netlink.c:3773:22: note: (near initialization for 'ctnl_notifier.next')
net/netfilter/nf_conntrack_netlink.c: In function 'ctnetlink_net_pre_exit':
net/netfilter/nf_conntrack_netlink.c:3880:9: error: too few arguments to function 'nf_conntrack_unregister_notifier'
 3880 |         nf_conntrack_unregister_notifier(net);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/net/netfilter/nf_conntrack_core.h:18,
                 from net/netfilter/nf_conntrack_netlink.c:38:
./include/net/netfilter/nf_conntrack_ecache.h:88:12: note: declared here
   88 | extern int nf_conntrack_unregister_notifier(struct net *net, struct notifier_block *nb);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors

同样报错。

Beginner-Go, 2022-01-15:
删掉 [这个patch](https://github.com/coolsnowwolf/lede/commit/a30c8b94365cc646bcbce4ada2b4594bb8afe9c9)

coolsnowwolf, 2022-01-15:
API 参数改了 等修复下

closed 2022-01-16 by coolsnowwolf