coolsnowwolf/lede #8698: pve下安装openwrt无法保存配置

haohaoget, 2022-01-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：pve下安装openwrt无法保存配置（20+512m并没有溢出
[image]
）

### (2) 路由器型号和固件版本

A：N3450
R22.1.1 / LuCI Master (git-21.335.48743-5f363d9)

### (3) 详细日志

A：
[    0.000000] Linux version 5.4.162 (runner@fv-az133-964) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r1-b3ea4ae93)) #0 SMP Fri Jan 14 11:43:29 2022
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=3e644446-87f3-dae5-b236-b852c3a54e02 rootwait console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[3]:  576, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  640, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 704 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffd9fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffda000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.14.0-0-g155821a1990b-prebuilt.qemu.org 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 23ef001, primary cpu clock
[    0.000000] kvm-clock: using sched offset of 5901737876 cycles
[    0.000021] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000033] tsc: Detected 1094.400 MHz processor
[    0.003030] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.003033] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.003038] last_pfn = 0x7ffda max_arch_pfn = 0x400000000
[    0.003089] MTRR default type: write-back
[    0.003090] MTRR fixed ranges enabled:
[    0.003092]   00000-9FFFF write-back
[    0.003093]   A0000-BFFFF uncachable
[    0.003094]   C0000-FFFFF write-protect
[    0.003095] MTRR variable ranges enabled:
[    0.003097]   0 base 0080000000 mask 7F80000000 uncachable
[    0.003098]   1 disabled
[    0.003098]   2 disabled
[    0.003099]   3 disabled
[    0.003100]   4 disabled
[    0.003100]   5 disabled
[    0.003101]   6 disabled
[    0.003101]   7 disabled
[    0.003115] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.007413] found SMP MP-table at [mem 0x000f5a80-0x000f5a8f]
[    0.007511] Using GB pages for direct mapping
[    0.007838] ACPI: Early table checksum verification disabled
[... 580 lines cut ...]
[   13.994209] usbcore: registered new interface driver ums-usbat
[   14.028107] xt_time: kernel timezone is -0000
[   14.031895] usbcore: registered new interface driver ax88179_178a
[   14.039891] PPP generic driver version 2.4.2
[   14.041369] PPP MPPE Compression module registered
[   14.043082] NET: Registered protocol family 24
[   14.045217] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   14.046385] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[   14.051170] appfilter version:5.0.1
[   14.051716] open feature file failed
[   14.052204] error, feature buf is null
[   14.052780] load feature failed
[   14.081153] kmodloader: 1 module could not be probed
[   14.081927] kmodloader: - oaf - 0
[   39.890907] 8021q: adding VLAN 0 to HW filter on device eth0
[   39.892122] br-lan: port 1(eth0) entered blocking state
[   39.892806] br-lan: port 1(eth0) entered disabled state
[   39.893724] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   39.896065] device eth0 entered promiscuous mode
[   40.001024] br-lan: port 1(eth0) entered blocking state
[   40.001900] br-lan: port 1(eth0) entered forwarding state
[   40.002778] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   44.480194] overlayfs: filesystem on '/opt/check-overlayfs-support2788646186/upper' not supported as upperdir
[   44.492191] overlayfs: filesystem on '/opt/check-overlayfs-support3457493494/upper' not supported as upperdir
[  226.956750] sh (8003): drop_caches: 3

haohaoget, 2022-01-15:
系统在运行过程中修改的配置过一会也会消失，openwrt的reboot无效只有去pve里面停止再启动，启动后配置也会失效（所有的包括密码），感觉像是启动时和运行中系统出现故障默认把rom里面覆盖到overlay里面了

MrVicent, 2022-01-20:
感觉是你安装过程有问题吧
我这试过好几台pve装openwert都是正常跑的
最好把安装过程检查一下 或者再找别的教程参考下
也有可能是镜像转成虚拟磁盘那出问题了

haohaoget, 2022-01-20:
> 感觉是你安装过程有问题吧 我这试过好几台pve装openwert都是正常跑的 最好把安装过程检查一下 或者再找别的教程参考下 也有可能是镜像转成虚拟磁盘那出问题了

目前重新编译了个固件用一样的安装方法暂时稳定了，只是每次开机后实时监控和ttyd会消失过一段时间才会出来，还有passwall和解锁网易云等不会自启，平时还能正常用。

usazcx, 2022-02-14:
pve有时候关机不会正常关机，导致openwrt对硬盘写保护。
通过ssh连接到openwrt执行
`e2fsck -y /dev/sda2
`
重启后即可解决

coolsnowwolf, 2026-09-06:
PVE 下无法保存配置只有截图，缺少磁盘只读状态、挂载和 UCI 写入日志，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf