coolsnowwolf/lede #8700: 编译出错，go：download TCP连接被拒绝

lin1123, 2022-01-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译出错，go：download TCP连接被拒绝

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：touch /home/lin/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-i18n-ramfree-zh-cn_installed
echo "luci-i18n-ramfree-zh-cn" >> /home/lin/lede/staging_dir/target-x86_64_musl/pkginfo/luci-app-ramfree.default.install
make[3]: Leaving directory '/home/lin/lede/package/lean/luci-app-ramfree'
time: package/lean/luci-app-ramfree/compile#0.14#0.05#0.18
make[3]: Entering directory '/home/lin/lede/package/lean/rclone'
mkdir -p /home/lin/lede/dl
SHELL= flock /home/lin/lede/tmp/.rclone-1.57.0.tar.gz.flock -c '  	/home/lin/lede/scripts/download.pl "/home/lin/lede/dl" "rclone-1.57.0.tar.gz" "294f7a6b0874509997d3a9ffae7c74f0c45b687df0ac7d7742f284ad3814fe55" "" "https://codeload.github.com/rclone/rclone/tar.gz/v1.57.0?"    '
rm -f /home/lin/lede/build_dir/target-x86_64_musl/rclone-1.57.0/.built
touch /home/lin/lede/build_dir/target-x86_64_musl/rclone-1.57.0/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/lin/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/lin/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/rclone/rclone" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/rclone/rclone/..." GO_EXCLUDES="test" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/lin/lede/build_dir/target-x86_64_musl/rclone-1.57.0" GO_BUILD_DIR="/home/lin/lede/build_dir/target-x86_64_musl/rclone-1.57.0/.go_work/build" GO_BUILD_BIN_DIR="/home/lin/lede/build_dir/target-x86_64_musl/rclone-1.57.0/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/lin/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOARM="" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/lin/lede/build_dir/target-x86_64_musl/rclone-1.57.0=rclone-1.57.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/lin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/lin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/lin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/lin/lede/build_dir/target-x86_64_musl/rclone-1.57.0=rclone-1.57.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/lin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/lin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/home/lin/lede/build_dir/target-x86_64_musl/rclone-1.57.0/.go_work/build" GOCACHE="/home/lin/lede/tmp/go-build" GOMODCACHE="/home/lin/lede/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/lin/lede/feeds/packages/lang/golang//golang-build.sh build -v -trimpath -ldflags "all=-buildid '1642226050' -linkmode external -extldflags '-L/home/lin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/lin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-s -w -X github.com/rclone/rclone/fs.Version=v1.57.0 -X main.Version=v1.57.0 -X main.BuildUser=openwrt -X main.BuildHost=openwrt -buildid '1642226050' -linkmode external -extldflags '-L/home/lin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/lin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets
go: bazil.org/fuse@v0.0.0-20200524192727-fb710f7dfd05: Get "https://proxy.golang.org/bazil.org/fuse/@v/v0.0.0-20200524192727-fb710f7dfd05.mod": dial tcp 142.251.43.17:443: connect: connection refused
go: downloading github.com/ncw/go-acd v0.0.0-20201019170801-fe55f33415b1
go: downloading github.com/pkg/errors v0.9.1
go: downloading golang.org/x/oauth2 v0.0.0-20210819190943-2bc19b11175f
go: downloading github.com/youmark/pkcs8 v0.0.0-20201027041543-1326539a0a0a
go: downloading golang.org/x/sync v0.0.0-20210220032951-036812b2e83c
go: downloading github.com/Azure/azure-pipeline-go v0.2.3
go: downloading github.com/Azure/azure-storage-blob-go v0.14.0
go: downloading github.com/Azure/go-autorest/autorest/adal v0.9.14
go: downloading github.com/dropbox/dropbox-sdk-go-unofficial/v6 v6.0.3
go: downloading github.com/buengese/sgzip v0.1.1
go: downloading github.com/patrickmn/go-cache v2.1.0+incompatible
go: downloading github.com/gabriel-vasile/mimetype v1.3.1
go: downloading go.etcd.io/bbolt v1.3.6
go: downloading github.com/rfjakob/eme v1.1.2
go: downloading google.golang.org/api v0.54.0
go: downloading golang.org/x/net v0.0.0-20210813160813-60bc85c4be6d
go: downloading golang.org/x/crypto v0.0.0-20210817164053-32db794688a5
go: downloading github.com/rclone/ftp v1.0.0-210902f
go: downloading golang.org/x/time v0.0.0-20210723032227-1f47c861a9ac
go: downloading github.com/colinmarc/hdfs/v2 v2.2.0
go: downloading github.com/ncw/swift/v2 v2.0.1
go: downloading github.com/jcmturner/gokrb5/v8 v8.4.2
go: downloading github.com/koofr/go-httpclient v0.0.0-20200420163713-93aa7c75b348
go: downloading github.com/t3rm1n4l/go-mega v0.0.0-20200416171014-ffad7fcb44b8
go: downloading golang.org/x/sys v0.0.0-20210820121016-41cdb8703e55
go: downloading github.com/koofr/go-koofrclient v0.0.0-20190724113126-8e5366da203a
go: downloading golang.org/x/text v0.3.7
go: downloading github.com/putdotio/go-putio/putio v0.0.0-20200123120452-16d982cac2b8
go: downloading github.com/yunify/qingstor-sdk-go/v3 v3.2.0
go: downloading github.com/coreos/go-semver v0.3.0
go: downloading github.com/aws/aws-sdk-go v1.40.27
go: downloading github.com/pkg/sftp v1.13.2
go: downloading github.com/xanzy/ssh-agent v0.3.1
go: downloading github.com/google/uuid v1.3.0
go: downloading storj.io/uplink v1.4.6
go: downloading github.com/Azure/go-ntlmssp v0.0.0-20200615164410-66371956d46c
go: downloading github.com/spf13/cobra v1.2.1
go: downloading github.com/spf13/pflag v1.0.5
go: downloading bazil.org/fuse v0.0.0-20200524192727-fb710f7dfd05
[... 15 lines cut ...]
go: downloading github.com/sirupsen/logrus v1.8.1
go: downloading github.com/prometheus/client_golang v1.11.0
go: downloading github.com/Unknwon/goconfig v0.0.0-20200908083735-df7de6a44db8
go: downloading github.com/a8m/tree v0.0.0-20210414114729-ce3525c5c2ef
go: downloading github.com/aalpar/deheap v0.0.0-20200318053559-9a0c2883bd56
go: downloading gopkg.in/yaml.v2 v2.4.0
go: downloading github.com/shirou/gopsutil/v3 v3.21.8
go: downloading github.com/klauspost/compress v1.13.4
go: downloading github.com/mattn/go-colorable v0.1.8
go: downloading github.com/dop251/scsu v0.0.0-20200422003335-8fadfb689669
go: bazil.org/fuse@v0.0.0-20200524192727-fb710f7dfd05: Get "https://proxy.golang.org/bazil.org/fuse/@v/v0.0.0-20200524192727-fb710f7dfd05.mod": dial tcp 142.251.43.17:443: connect: connection refused

Building targets
go: bazil.org/fuse@v0.0.0-20200524192727-fb710f7dfd05: Get "https://proxy.golang.org/bazil.org/fuse/@v/v0.0.0-20200524192727-fb710f7dfd05.mod": dial tcp 142.251.43.17:443: connect: connection refused
go: bazil.org/fuse@v0.0.0-20200524192727-fb710f7dfd05: Get "https://proxy.golang.org/bazil.org/fuse/@v/v0.0.0-20200524192727-fb710f7dfd05.mod": dial tcp 142.251.43.17:443: connect: connection refused

make[3]: *** [Makefile:109: /home/lin/lede/build_dir/target-x86_64_musl/rclone-1.57.0/.built] Error 1
make[3]: Leaving directory '/home/lin/lede/package/lean/rclone'
time: package/lean/rclone/compile#0.56#1.55#698.15
    ERROR: package/lean/rclone failed to build.
make[2]: *** [package/Makefile:116: package/lean/rclone/compile] Error 1
make[2]: Leaving directory '/home/lin/lede'
make[1]: *** [package/Makefile:110: /home/lin/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lin/lede'
make: *** [/home/lin/lede/include/toplevel.mk:230：world] 错误 2

minlang112, 2022-01-15:
网络问题，全局代理未发现该现象

lin1123, 2022-01-15:
> 网络问题，全局代理未发现该现象
用的VMware创建Ubuntu虚拟机编译的，通过V2rayN开启全局代理，虚拟机下linux系统内，设置网络代理手动指定到Windows下socks5端口及电脑本地IP，虚拟机内测试IP地址，显示为代理的IP。测试过多次都还是这种错误。

closed 2022-01-16 by AmadeusGhost