coolsnowwolf/lede #8703: 每次刚开机wireguard连接到路由器不过ss

xinxijishuwyq, 2022-01-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：每次刚开机或重启wireguard连接到路由器不过ss
[image]
如上图，其中cnm为wireguard的接口，接口控制一设置好。但是刚开机的时候wiretuard连接的不过ss代理，必须手动保存应用一次才可以。
查看防火墙，发现刚开记得时候相关接口少了两个，提交后恢复正常

### (2) 路由器型号和固件版本

A：x64虚拟机

### (3) 详细日志

A：

xinxijishuwyq, 2022-01-16:
刚开机的时候nat表和mangle表的ss相关项都没有wireguard对应接口的规则，重新提交应用luci-app-ssr才有-

coolsnowwolf, 2022-01-16:
wg 成功建立接口顺序比插件后，你可以参照官方wiki写个hotplug 脚本自动重启，参考wiki

xinxijishuwyq, 2022-01-16:
> wg 成功建立接口顺序比插件后，你可以参照官方wiki写个hotplug 脚本自动重启，参考wiki

创建文件
/etc/hotplug.d/iface/50-wg

`#!/bin/sh
if [ "ifup" = "$ACTION" ] && [ "$INTERFACE" = "cnm" ]; then
        /etc/init.d/shadowsocksr restart
fi`

确实可以了，不过只是这样升级后就没了吧==

zhangguanzhang, 2022-01-17:
[image]
接口名字亮了

coolsnowwolf, 2026-09-06:
WireGuard 开机时防火墙接口缺失、手动保存后恢复，缺少 network/firewall 配置和启动时序日志，无法在现行代码定位，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf