coolsnowwolf/lede #8706: 5.15 在线编译内核kmod报错

Cptmacmillan2022007, 2022-01-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在线编译到kmod-rxrpc报错

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：

Packaged contents of /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-fs-fscache into /workdir/openwrt/bin/targets/x86/64/packages/kmod-fs-fscache_5.15.12-1_x86_64.ipk
rstrip.sh: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-dnsresolver/lib/modules/5.15.12/dns_resolver.ko: relocatable
Packaged contents of /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-dnsresolver into /workdir/openwrt/bin/targets/x86/64/packages/kmod-dnsresolver_5.15.12-1_x86_64.ipk
Package kmod-rxrpc is missing dependencies for the following libraries:
ip6_udp_tunnel.ko
udp_tunnel.ko
make[3]: *** [modules/netsupport.mk:1154: /workdir/openwrt/bin/targets/x86/64/packages/kmod-rxrpc_5.15.12-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#9.51#2.39#14.19
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-01-17:
又选了什么驱动嘛，我这边编译没任何问题

Cptmacmillan2022007, 2022-01-17:
> 又选了什么驱动嘛，我这边编译没任何问题

貌似是和ipv6还有udp tunnel有关的组件

WYC-2020, 2022-01-17:
@Cptmacmillan2022007 https://github.com/WYC-2020/lede/commit/d1b0bfbf68a8079e1e75e5d1eed2cc55def5a730
测试下功能

Cptmacmillan2022007, 2022-01-17:
好的

Cptmacmillan2022007, 2022-01-19:
> @Cptmacmillan2022007 [WYC-2020/lede@d1b0bfb](https://github.com/WYC-2020/lede/commit/d1b0bfbf68a8079e1e75e5d1eed2cc55def5a730) 测试下功能

kmod-rxrpc没有报错

新的错误出现了

Package kmod-usb-net-smsc95xx is missing dependencies for the following libraries:
libphy.ko
make[3]: *** [modules/usb.mk:1265: /workdir/openwrt/bin/targets/x86/64/packages/kmod-usb-net-smsc95xx_5.15.15-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#51.55#20.33#79.31
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-01-19:
后面再看应该是usb-net-smsc95xx 少了依赖，我现在在搞5.15的sfe

WYC-2020, 2022-01-20:
@Cptmacmillan2022007 
[image]
这样修改

shu0802, 2022-01-22:
> @Cptmacmillan2022007 [image] 这样修改

看来5.15的问题还多，得修复才能编译

WYC-2020, 2022-01-22:
ksmbd我修复了的不再说，驱动那么多有错很正常，不可能一次性就能把所有驱动都测试完了

Cptmacmillan2022007, 2022-01-23:
> > @Cptmacmillan2022007 [image] 这样修改
> 
> 看来5.15的问题还多，得修复才能编译。今天编译遇到了关于ksmbd报错。 [image]

那5.10编译正常吗？不行我就退回5.10编译了。

closed 2022-02-28 by AmadeusGhost

ndevel, 2022-12-21:
> 

make[3]: Entering directory '/home/dev/dev/lede_20220412/lede/package/firmware/prism54-firmware'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/dev/dev/lede_20220412/lede/package/firmware/prism54-firmware'
time: package/firmware/prism54-firmware/compile#0.15#0.02#0.40
make[3]: Entering directory '/home/dev/dev/lede_20220412/lede/package/kernel/linux'
mkdir -p /home/dev/dev/lede_20220412/lede/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/dev/dev/lede_20220412/lede/tmp/.root-copy.flock -c 'cp -fpR /home/dev/dev/lede_20220412/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/dev/dev/lede_20220412/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/dev/dev/lede_20220412/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-rxrpc is missing dependencies for the following libraries:
ip6_udp_tunnel.ko
udp_tunnel.ko
make[3]: *** [modules/netsupport.mk:1154: /home/dev/dev/lede_20220412/lede/bin/targets/x86/64/packages/kmod-rxrpc_5.15.34-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/dev/dev/lede_20220412/lede/package/kernel/linux'
time: package/kernel/linux/compile#1.93#0.23#7.69
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/dev/dev/lede_20220412/lede'
make[1]: *** [package/Makefile:110: /home/dev/dev/lede_20220412/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dev/dev/lede_20220412/lede'
make: *** [/home/dev/dev/lede_20220412/lede/include/toplevel.mk:230: world] Error 2

出现了同样的错误，有解决方案么，
https://github.com/WYC-2020/lede/commit/d1b0bfbf68a8079e1e75e5d1eed2cc55def5a730  这个链接无法访问了