coolsnowwolf/lede #8707: 使用Action编译报错

lon91ong, 2022-01-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 有类似的issue #8619 但是没有得到解决

## 2. 详细叙述

### (1) 具体问题

A：fullconenat-2021-12-04-7fe02604/xt_FULLCONENAT.c:1350:7: error: void value not ignored as it ought to be
ret = nf_nat_masquerade_ipv4_register_notifier();

查找了一番，似乎nf_nat_masquerade_ipv4_register_notifier()函数定义返回值为void，无法直接赋值给ret

### (2) 路由器型号和固件版本

A：4.14内核源码，#8619 是5.4源码， 应该和内核版本关系不大

### (3) 详细日志

A：
```
make[3]: Entering directory '/workdir/openwrt/feeds/custom/openwrt-fullconenat'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.fullconenat-2021-12-04-7fe02604.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "fullconenat-2021-12-04-7fe02604.tar.xz" "skip" "" || (  /workdir/openwrt/scripts/dl_github_archive.py --dl-dir="/workdir/openwrt/dl" --url="https://github.com/llccd/netfilter-full-cone-nat.git" --version="7fe02604e073945a3b081c18efbb4d3c799a31f6" --subdir="fullconenat-2021-12-04-7fe02604" --source="fullconenat-2021-12-04-7fe02604.tar.xz" --hash="skip" || ( 	echo "Checking out files from the git repository..."; mkdir -p /workdir/openwrt/tmp/dl && cd /workdir/openwrt/tmp/dl && rm -rf fullconenat-2021-12-04-7fe02604 && [ \! -d fullconenat-2021-12-04-7fe02604 ] && git clone  https://github.com/llccd/netfilter-full-cone-nat.git fullconenat-2021-12-04-7fe02604 && (cd fullconenat-2021-12-04-7fe02604 && git checkout 7fe02604e073945a3b081c18efbb4d3c799a31f6 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fullconenat-2021-12-04-7fe02604 && git log -1 --format='\''@%ct'\''` && rm -rf fullconenat-2021-12-04-7fe02604/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fullconenat-2021-12-04-7fe02604 | 	xz -zc -7e > /workdir/openwrt/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz && mv /workdir/openwrt/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz /workdir/openwrt/dl/ && rm -rf fullconenat-2021-12-04-7fe02604; );  )    '
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604/.built
touch /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604/.built_check
make -j1 -C "/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.14.261" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" M="/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604" EXTRA_CFLAGS="" modules
make[4]: Entering directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.14.261'
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604/xt_FULLCONENAT.o
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604/xt_FULLCONENAT.c: In function 'fullconenat_tg_init':
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604/xt_FULLCONENAT.c:1350:7: error: void value not ignored as it ought to be
   ret = nf_nat_masquerade_ipv4_register_notifier();
       ^
scripts/Makefile.build:335: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604/xt_FULLCONENAT.o' failed
make[5]: *** [/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604/xt_FULLCONENAT.o] Error 1
make[4]: *** [_module_/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604] Error 2
Makefile:1550: recipe for target '_module_/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604' failed
make[3]: *** [/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604/.built] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.14.261'
Makefile:65: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/fullconenat-2021-12-04-7fe02604/.built' failed
make[3]: Leaving directory '/workdir/openwrt/feeds/custom/openwrt-fullconenat'
time: package/feeds/custom/openwrt-fullconenat/compile#1.37#0.46#1.68
package/Makefile:113: recipe for target 'package/feeds/custom/openwrt-fullconenat/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [package/feeds/custom/openwrt-fullconenat/compile] Error 2
make[1]: *** [/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
package/Makefile:109: recipe for target '/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:221: recipe for target 'world' failed
make: *** [world] Error 2
Error: Process completed with exit code 2.
```

AmadeusGhost, 2022-01-17:
你用 openwrt 那个库不行吗？为什么要在 5.x 的源码里面杂交 4.14。。。

WYC-2020, 2022-01-17:
组合怪就自己解决哈哈

lon91ong, 2022-01-18:
> 你用 openwrt 那个库不行吗？为什么要在 5.x 的源码里面杂交 4.14。。。

没有杂交内核源码， #8619 的问题是采用5.4源码编译遇到的，我的问题是编译[4.14内核源码](https://github.com/Lienol/openwrt/tree/19.07)遇到的，问题的相同点都是出在同一处：fullconenat-2021-12-04-7fe02604

WYC-2020, 2022-01-20:
4.14内核源码可能对于nf_nat_masquerade_ipv4_register_notifier的函数定义为返回值为void,与5.4以后的不同，5.4到5.15 我这边编译都没问题，具体要下载4.14内核来看一下 @lon91ong

lon91ong, 2022-01-27:
> 

> 4.14内核源码可能对于nf_nat_masquerade_ipv4_register_notifier的函数定义为返回值为void,与5.4以后的不同，5.4到5.15 我这边编译都没问题，具体要下载4.14内核来看一下 @lon91ong

早前一样的编译都没问题，难道现在还有针对4.14代码的**大变动**？有没有方法指定用fullconenat之前的代码版本？

WYC-2020, 2022-01-27:
年后有时间看下

lon91ong, 2022-01-27:
> 年后有时间看下

深入搜索了一番，找到了这个[Makefile](https://github.com/Lienol/openwrt/blob/19.07/package/network/fullconenat/Makefile), 其中明确指定了`fullconenat`的文件版本，从现在报错的信息看，似乎这个指定设置没有生效
```
PKG_NAME:=fullconenat
PKG_RELEASE:=1

PKG_SOURCE_DATE:=2020-05-09
PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/llccd/netfilter-full-cone-nat.git
PKG_MIRROR_HASH:=bec75e2c0dc1103928861f8ea8f3f9ada63731103099a884222e15eab0a2fff8
PKG_SOURCE_VERSION:=cc30ca6031ef5dca5bdb10cfd5ffc0c79425af9e

PKG_LICENSE:=GPL-2.0
PKG_LICENSE_FILES:=LICENSE
```

lon91ong, 2022-01-28:
找到错误原因了，错误的位置是'/openwrt/**feeds/custom**/openwrt-fullconenat', 这个是自己在编译前添加引用的包
`sed -i '$a src-git custom https://github.com/kiddin9/openwrt-packages' feeds.conf.default`
然后找到了这个包的更新历史：
[image]
Makefile变化：
[image]
原因已经显而易见了！

closed 2022-01-28 by lon91ong