coolsnowwolf/lede #8710: 5.15编译错误

GreatMichaelLee, 2022-01-16:
最新代码，谁能帮我看看，这是什么情况

touch /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.configured
rm -f /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/vmlinux /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/System.map
make -C /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12 KCFLAGS="-fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/builder/lede_x86/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Jan 16 06:01:17 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/builder/lede_x86/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/builder/lede_x86/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.12 CC="x86_64-openwrt-linux-musl-gcc" bzImage modules
make[5]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  DESCEND objtool
  CHK     include/generated/compile.h
  UPD     include/generated/compile.h
  CC      init/version.o
  AR      init/built-in.a
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
  MODPOST vmlinux.symvers
  MODINFO modules.builtin.modinfo
  GEN     modules.builtin
  LD      .tmp_vmlinux.kallsyms1
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/kernel/head_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/entry/entry_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/kernel/irqflags.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/hweight.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/clear_page_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/copy_page_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/copy_user_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/getuser.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/memcpy_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/memmove_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/memset_64.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/putuser.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: warning: orphan section `__ksymtab_strings' from `arch/x86/lib/retpoline.o' being placed in section `__ksymtab_strings'
x86_64-openwrt-linux-musl-ld: drivers/dma-buf/dma-buf.o: in function `dma_buf_release':
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/drivers/dma-buf/dma-buf.c:79: undefined reference to `dma_buf_stats_teardown'
x86_64-openwrt-linux-musl-ld: drivers/dma-buf/dma-buf.o: in function `dma_buf_export':
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/drivers/dma-buf/dma-buf.c:575: undefined reference to `dma_buf_stats_setup'
x86_64-openwrt-linux-musl-ld: drivers/dma-buf/dma-buf.o: in function `dma_buf_init':
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/drivers/dma-buf/dma-buf.c:1480: undefined reference to `dma_buf_init_sysfs_statistics'
x86_64-openwrt-linux-musl-ld: drivers/dma-buf/dma-buf.o: in function `dma_buf_deinit':
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/drivers/dma-buf/dma-buf.c:1499: undefined reference to `dma_buf_uninit_sysfs_statistics'
make[5]: *** [Makefile:1183: vmlinux] Error 1
make[5]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12'
make[4]: *** [Makefile:28: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.12/.modules] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/target/linux'
time: target/linux/compile#20.24#6.38#26.34
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [target/Makefile:18: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230: world] Error 2
builder@gcp-hk:~/lede_x86$

WYC-2020, 2022-01-17:
你一天到晚哪那么多问题哦，都懒得给你修了哈哈，有时间再看

Wilson-coder, 2022-01-18:
> 你一天到晚哪那么多问题哦，都懒得给你修了哈哈，有时间再看

好像代码库200多位贡献者里面并没有你啊？看你的回复，好像是你主要负责Debug？问题是你真正帮忙解决了哪个问题？

WYC-2020, 2022-01-18:
行，我不多说

AdoShan, 2022-02-20:
我也是这个错误

sjkhsl, 2022-03-04:
我也是，你解决了吗？

coolsnowwolf, 2026-09-05:
5.15 编译日志只显示链接阶段告警并在实际失败前截断，缺少首个错误，无法定位源码问题，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf