coolsnowwolf/lede #8712: R8000编译错误

skyeyc, 2022-01-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
是的
## 2. 详细叙述

### (1) 具体问题

A：R8000编译错误，x64固件编译没问题，提示补丁错误

### (2) 路由器型号和固件版本

A：R8000

### (3) 详细日志

A：
Applying /home/donadel/r8000/target/linux/bcm53xx/patches-5.4/302-ARM-dts-BCM5301X-Update-Northstar-pinctrl-binding.patch using plaintext: 
patching file arch/arm/boot/dts/bcm5301x.dtsi
patch: **** malformed patch at line 21: @@ -450,10 +450,9 @@

Patch failed!  Please fix /home/donadel/r8000/target/linux/bcm53xx/patches-5.4/302-ARM-dts-BCM5301X-Update-Northstar-pinctrl-binding.patch!
make[4]: *** [Makefile:40: /home/donadel/r8000/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.171/.prepared_af02e8c29fc0adde62d82f071fcb993d] Error 1
make[4]: Leaving directory '/home/donadel/r8000/target/linux/bcm53xx'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/donadel/r8000/target/linux'
time: target/linux/compile#10.04#9.54#13.23
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/donadel/r8000'
make[1]: *** [target/Makefile:18: /home/donadel/r8000/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/donadel/r8000'
make: *** [/home/donadel/r8000/include/toplevel.mk:230：world] 错误 2

Beginner-Go, 2022-01-16:
等会我看看

lxhao61, 2022-01-16:
类似出错。
https://github.com/lxhao61/Actions-Lean-OpenWrt/runs/4832360601?check_suite_focus=true

emsvip, 2022-01-17:
[image]
编译k3也是类似的错误，朋友们问题解决了吗？

Beginner-Go, 2022-01-17:
别催 在修了

closed 2022-01-17 by coolsnowwolf