coolsnowwolf/lede #8713: lede有啥文件作为脚本的判断依据吗

zhangguanzhang, 2022-01-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

我用lede，天灵和官方的openwrt 的源码编译了固件，然后写了个升级脚本，脚本会下载新的固件。比如lede的就下载lede，天灵就下载天灵的。
lede的固件rootfs内部有啥文件或者命令能判断当前固件是 `lede 的

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：1

### (2) 路由器型号和固件版本

A：1

### (3) 详细日志

A：1

WYC-2020, 2022-01-17:
你自己定义名字不就可以确定是哪个固件了

zhangguanzhang, 2022-01-17:
> 你自己定义名字不就可以确定是哪个固件了

你哪次能仔细看下我说的文字，我都说了是一个脚本升级所有固件，假设脚本在lede的固件里执行，怎么判断当前是lede固件

AmadeusGhost, 2022-01-17:
不要浪费别人的时间。

closed 2022-01-17 by AmadeusGhost

WYC-2020, 2022-01-17:
这玩意就是想搞自动升级，都能自己加功能了，再固件里面加一个标识是很难吗？然后用脚本读取哪个标识，去下载不就行了，转不弯伤不起，组合怪

zhangguanzhang, 2022-01-17:
你们两个不会我可以教你俩
```
        REPO=lede
        # lede immortalwrt openwrt 的 os-release 都包含 lede，非 lede 判断放后面
        grep -qw 'immortalwrt' /etc/os-release && REPO=immortalwrt
```