coolsnowwolf/lede #8716: no declaration for Atheros 5XXX WLAN card

GreatMichaelLee, 2022-01-17:
编译5.15，遇到这个错误，哪位大佬看看？

/drivers/net/wireless/ath/ath5k/reset.o
  CC [M]  /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/attach.o
  CC [M]  /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/base.o
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/base.c:93:25: error: expected declaration specifiers or '...' before string constant
 MODULE_SUPPORTED_DEVICE("Atheros 5xxx WLAN cards");
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
make[11]: *** [scripts/Makefile.build:277: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/base.o] Error 1
make[10]: *** [scripts/Makefile.build:540: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k] Error 2
make[9]: *** [scripts/Makefile.build:540: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath] Error 2
make[8]: *** [scripts/Makefile.build:540: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1868: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
make[3]: *** [Makefile:584: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#13.91#2.74#16.43
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-01-17:
https://github.com/WYC-2020/lede/commit/08a66611083f4fc61de545b98d0e4064978730a6

GreatMichaelLee, 2022-01-17:
> [WYC-2020/lede@08a6661](https://github.com/WYC-2020/lede/commit/08a66611083f4fc61de545b98d0e4064978730a6)

Is this implemented?

WYC-2020, 2022-01-18:
相信就用不相信就不用，要不然等会又有喷子来喷我，这个宏MODULE_SUPPORTED_DEVICE linux已经废除了，所以只需要定义一个空的宏，让他编译通过即可，其他我就不说了 @GreatMichaelLee

closed 2022-03-29 by AmadeusGhost