coolsnowwolf/lede #8718: Qualcomm Technologies, Inc. IPQ8064/AP-148，启动报错

flexyangjie, 2022-01-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [是的 ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
A：启动时日志报错，然后重启。也偶尔能正常启动，但只能坚持几分钟或在WEB界面做一些操作之后也会再现故障自动重启
关键日志： Unable to handle kernel NULL pointer dereference at virtual address 00000040（该处地址每次都不一样）

有尝试过最新固件和早两个月的固件，有尝试过默认配置编译和去掉APP编译，也有试过连无线相关NSS相关都去掉编译。编译了二三十次。基本上都上出现上面的报错，然后再报Kernel panic - not syncing: Fatal exception，自动重启。

不确定是不是UBOOT环境设置不对或刷机姿势不对。硬件应该没问题，原机固件没有这个故障，有两台相同的机器都一样的问题。

### (2) 路由器型号和固件版本
A：高通AP148兼容板，PCB板上型号sr5005 http://www.dunchongnet.com/upload/2019/03/30/155391302405jgae0.pdf
ipq8064,32M-NOR,512M-NAND,没有无线相关硬件。可能使用openwrt原版机型AP148，RAMFS文件启动，能正常识别各接口硬件
原机系统是任子行改造过的3.14内核版本，在NOR上。
B：没有该机型的刷机资料，使用网上找到的兄弟机型AP161的刷机方式。操作如下：
tftpboot openwrt-ipq806x-generic-qcom_ipq8064-ap148-fit-uImage.itb
nand erase  0x320000 0x400000
nand write  0x44000000 0x320000 0x400000
----------上面刷引导文件，下面刷固件----------------------------------
tftpboot openwrt-ipq806x-generic-qcom_ipq8064-ap148-squashfs-nand-factory.bin
nand erase  0x720000 0x1f8e0000
nand write  0x44000000 0x1340000 $文件尺寸
C: printenv 如下：
(IPQ) # printenv
autostart=no
baudrate=115200
bootargs=console=ttyMSM0,115200n8 ubi.mtd=ubi root=/dev/ubiblock0_1 rootfstype=squashfs
bootcmd=nboot 0x44000000 0 0x320000;bootm
bootdelay=3
ethact=eth0
fileaddr=44000000
filesize=1120000
ipaddr=192.168.1.2
machid=1260
nand_erasesize=20000
nand_oobsize=80
nand_writesize=800
serverip=192.168.1.22
stderr=serial
stdin=serial
stdout=serial

### (3) 详细日志

A：原机日志参考
U-Boot 2012.07 [Standard IPQ806X.LN,r187] (Mar 21 2016 - 15:22:08)

smem ram ptable found: ver: 0 len: 5
DRAM:  1003 MiB
NAND:  SF: Detected S25FL256S with page size 64 KiB, total 32 MiB
ipq_spi: page_size: 0x100, sector_size: 0x10000, size: 0x2000000
544 MiB
MMC:
In:    serial
Out:   serial
Err:   serial
Net:   MAC1 addr:94:88:5e:8:9:95
S17 chip id: 0x1302
athrs17_reg_init: complete
athrs17_vlan_config ...done
S17c init  done
MAC2 addr:94:88:5e:8:9:96
eth0, eth1
[... 831 lines cut ...]
[   33.887916] [<c0425a00>] (exit_mmap) from [<c031aac4>] (mmput+0x58/0x118)
[   33.887933] [<c031aac4>] (mmput) from [<c03219b8>] (do_exit+0x334/0xa3c)
[   33.887947] [<c03219b8>] (do_exit) from [<c032213c>] (do_group_exit+0x48/0xd4)
[   33.887956] [<c032213c>] (do_group_exit) from [<c03221dc>] (__wake_up_parent+0x0/0x18)
[   33.887980] Code: e34c30c3 e5932874 e0811102 e59105b4 (e5902040)
[   33.888094] ---[ end trace d598fa800905d988 ]---
[   33.888103] Kernel panic - not syncing: Fatal exception
[   33.888890] CPU1: stopping
[   33.888911] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D WC        5.4.172 #0
[   33.888919] Hardware name: Generic DT based system
[   33.888961] [<c030f974>] (unwind_backtrace) from [<c030b968>] (show_stack+0x14/0x20)
[   33.889001] [<c030b968>] (show_stack) from [<c08ff598>] (dump_stack+0x94/0xa8)
[   33.889035] [<c08ff598>] (dump_stack) from [<c030eba0>] (handle_IPI+0x184/0x1b8)
[   33.889070] [<c030eba0>] (handle_IPI) from [<c05dc59c>] (gic_handle_irq+0xb4/0xb8)
[   33.889100] [<c05dc59c>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[   33.889112] Exception stack(0xef08ff18 to 0xef08ff60)
[   33.889137] ff00:                                                       00000000 00000007
[   33.889189] ff20: 2ebe1000 ef722a80 ee8f2c00 00000001 ef721e30 00000007 00000007 00000000
[   33.889244] ff40: e3eed8c0 e3b94520 00000015 ef08ff68 c0730748 c073074c 80000013 ffffffff
[   33.889282] [<c0301a8c>] (__irq_svc) from [<c073074c>] (cpuidle_enter_state+0x94/0x498)
[   33.889311] [<c073074c>] (cpuidle_enter_state) from [<c0730b94>] (cpuidle_enter+0x30/0x4c)
[   33.889338] [<c0730b94>] (cpuidle_enter) from [<c034ae34>] (do_idle+0x1d8/0x240)
[   33.889364] [<c034ae34>] (do_idle) from [<c034b144>] (cpu_startup_entry+0x1c/0x20)
[   33.889385] [<c034b144>] (cpu_startup_entry) from [<423024cc>] (0x423024cc)
[   34.052473] Rebooting in 1 seconds..

flexyangjie, 2022-01-18:
收集了一些报错信息，请大佬们参考
[   38.750171] br-lan: port 1(eth1.1) entered forwarding state
[   38.757639] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[  712.468224] 8<--- cut here ---
[  712.468261] Unable to handle kernel paging request at virtual address 00001900
[  712.470184] pgd = 14cb833d
[  712.477376] [00001900] *pgd=6e751835, *pte=00000000, *ppte=00000000
[  712.480081] Internal error: Oops: 17 [#1] SMP ARM
----------------------------------
[ 1453.848579] Unable to handle kernel paging request at virtual address 00100ea0
[ 1453.850501] pgd = 959edf6e
[ 1453.857699] [00100ea0] *pgd=00000000
[ 1453.860485] Internal error: Oops: 17 [#1] SMP ARM
[ 1453.864121] Modules linked in: ecm shortcut_fe_drv xt_FUL
----------------------------------------
 80.462715] Unable to handle kernel NULL pointer dereference at virtual address 00000240
[   80.464640] pgd = b4119e8c
[   80.473028] [00000240] *pgd=00000000
[   80.475405] Internal error: Oops: 5 [#1] SMP ARM
--------------------------------------
[   20.706396] 8<--- cut here ---
[   20.706450] Unable to handle kernel paging request at virtual address 000012c0
[   20.708493] pgd = 2d272de4
[   20.715571] [000012c0] *pgd=00000000
[   20.718270] Internal error: Oops: 5 [#1] SMP ARM
------------------------------------
   32.300895] 8<--- cut here ---
[   32.300915] Unable to handle kernel paging request at virtual address f0e337d0
[   32.302839] pgd = 2cffd42b
[   32.310100] [f0e337d0] *pgd=712ef811, *pte=00000000, *ppte=00000000
[   32.312736] Internal error: Oops: 7 [#1] SMP ARM
------------------------------------
10.087905] 8<--- cut here ---
[   10.087934] Unable to handle kernel paging request at virtual address 6269ece4
[   10.089858] pgd = 41aa889e
[   10.097056] [6269ece4] *pgd=00000000
[   10.099754] Internal error: Oops: 5 [#1] SMP ARM
[   10.103481] Modules linked in:
[   10.108083] CPU: 0 PID: 73 Comm: kmodloader Tainted: G        W         5.4.172 #0
[   10.110948] Hardware name: Generic DT based system
[   10.118512] PC is at _raw_spin_lock_irqsave+0x10/0x4c
[   10.123284] LR is at try_to_wake_up+0x50/0x53c
------------------------------------------
 11.668679] 8<--- cut here ---
[   11.668720] Unable to handle kernel paging request at virtual address 0800d000
[   11.670647] pgd = 288ea929
[   11.677942] [0800d000] *pgd=00000000
[   11.680624] Internal error: Oops: 805 [#1] SMP ARM
[   11.684263] Modules linked in:
[   11.688872] CPU: 0 PID: 73 Comm: kmodloader Tainted: G        W         5.4.172 #0
[   11.691913] Hardware name: Generic DT based system
---------------------------------------------
 [  122.461183] 8<--- cut here ---
[  122.461230] Unable to handle kernel paging request at virtual address 00008900
[  122.463162] pgd = a1e7dde4
[  122.470470] [00008900] *pgd=6de1e835, *pte=00000000, *ppte=00000000
[  122.473033] Internal error: Oops: 17 [#1] SMP ARM

closed 2022-02-11 by coolsnowwolf

lhc1993, 2024-11-09:
后来有解决么

flexyangjie, 2025-01-16:
有解决了，用23.05的版本可以正常使用，已正常使用一件多。找一找23.05带NSS的，然后DTB文件要小改编译出来才能正常使用

flexyangjie, 2025-01-16:
忘记说了，要用D7800机型，然后在uboot里改为从nand启动，就可以刷入用D7800机型编译的固件了

PR #8810 (2022-02-11): ipq806x: add support for SURF G-NAT200
```
Hardware specs:
  SoC: Qualcomm IPQ8064
  RAM: 1GB DDR3
  SPI: 32MB S25FL256S
  NAND: 512MB S34MS04G2 (unused)
  Ethernet: 5x 10/100/1000 Mbps QCA8337
  LED: internet, usb, sata
  Button: Reset
  USB: 1 x 3.0

Installation:
  Interrupt U-Boot, tftpboot initramfs image,
  than sysupgrade to openwrt.
```
Closed: #8718