coolsnowwolf/lede #8719: R2C编译后wan口没反应

wazituzi, 2022-01-17:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R2C最新版本编译后刷入，出现wan口没反应的问题

### (2) 路由器型号和固件版本

A：友善R2C，版本R22.1.1

### (3) 详细日志

A：

Gabrielxzx, 2022-01-17:
同无法驱动，明天重刷一遍看看有没有日志

LPDDR6-10000MHz, 2022-01-18:
上次有个人提交OrangePi R1 Plus LTS 带Motorcomm驱动，然后覆盖，有点乱。解决方法就是退回之前的驱动 
 https://github.com/coolsnowwolf/lede/blob/master/target/linux/rockchip/patches-5.4/600-net-phy-Add-driver-for-Motorcomm-YT85xx-PHYs.patch  ---> https://github.com/coolsnowwolf/lede/blob/b1b36c86af6d9fa3ba57eb2140025d2039996b81/target/linux/rockchip/patches-5.4/600-net-phy-Add-driver-for-Motorcomm-YT85xx-PHYs.patch

Gabrielxzx, 2022-01-19:
> 上次有个人提交OrangePi R1 Plus LTS 带Motorcomm驱动，然后覆盖，有点乱。解决方法就是退回之前的驱动 [...]

感谢，有效果

closed 2022-01-25 by AmadeusGhost