coolsnowwolf/lede #8721: 最近编译的几个版本中，AMD GX-420CA SOC 被识别为了1核1线程

TioaChan, 2022-01-17:
我正在使用惠普T620plus瘦机做x86软路由，在最近编译的几个版本中，cpu被识别为了1核1线程。

该瘦机使用了AMD GX-420CA SOC，是一颗4核4线程的u，见[ CPU Benchmarks](https://www.cpubenchmark.net/cpu.php?cpu=AMD+GX-420CA+SOC&id=2121)

[image]
[image]

请问该如何解决？

TioaChan, 2022-01-17:
这是最近一次启动时的内核日志
```
[    0.000000] Linux version 5.10.91 (tioachan@TioaChan-PC) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r4091-fabb20c32) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Fri Jan 14 07:13:10 2022
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=687d4ce8-ec70-3f71-5cf7-28a5dc4d3402 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000009c862fff] usable
[    0.000000] BIOS-e820: [mem 0x000000009c863000-0x000000009c892fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000009c893000-0x000000009c8c0fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000009c8c1000-0x000000009c9b8fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000009c9b9000-0x000000009dff9fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000009dffa000-0x000000009e0b2fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000009e0b3000-0x000000009e0b3fff] usable
[    0.000000] BIOS-e820: [mem 0x000000009e0b4000-0x000000009e0bbfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000009e0bc000-0x000000009e217fff] usable
[    0.000000] BIOS-e820: [mem 0x000000009e218000-0x000000009e8aafff] reserved
[    0.000000] BIOS-e820: [mem 0x000000009e8ab000-0x000000009e8effff] usable
[    0.000000] BIOS-e820: [mem 0x000000009e8f0000-0x000000009eff0fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000009eff1000-0x000000009effffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100001000-0x000000013effffff] usable
[    0.000000] Notice: NX (Execute Disable) protection missing in CPU!
[    0.000000] efi: EFI v2.31 by American Megatrends
[    0.000000] efi: ESRT=0x9dff8c98 ACPI=0x9c89d000 ACPI 2.0=0x9c89d000 SMBIOS=0x9dff8a98 
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: Hewlett-Packard HP t620 PLUS Quad Core TC/213D, BIOS L40 v02.19 02/19/2019
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 1996.226 MHz processor
[    0.000195] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000200] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000215] last_pfn = 0x13f000 max_arch_pfn = 0x400000000
[    0.000222] MTRR default type: uncachable
[    0.000224] MTRR fixed ranges enabled:
[    0.000226]   00000-9FFFF write-back
[    0.000228]   A0000-BFFFF write-through
[    0.000230]   C0000-DFFFF uncachable
[    0.000231]   E0000-FFFFF write-protect
[    0.000233] MTRR variable ranges enabled:
[    0.000235]   0 base 0000000000 mask FF80000000 write-back
[    0.000238]   1 base 0080000000 mask FFE0000000 write-back
[    0.000240]   2 base 009F000000 mask FFFF000000 uncachable
[    0.000241]   3 disabled
[    0.000242]   4 disabled
[    0.000243]   5 disabled
[    0.000244]   6 disabled
[    0.000245]   7 disabled
[    0.000247] TOM2: 000000013f000000 aka 5104M
[    0.000977] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001303] e820: update [mem 0x9f000000-0xffffffff] usable ==> reserved
[    0.001315] last_pfn = 0x9f000 max_arch_pfn = 0x400000000
[    0.008759] esrt: Reserving ESRT space from 0x000000009dff8c98 to 0x000000009dff8cd0.
[    0.008943] Using GB pages for direct mapping
[    0.010966] Secure boot disabled
[    0.010972] ACPI: Early table checksum verification disabled
[    0.010979] ACPI: RSDP 0x000000009C89D000 000024 (v02 HPQOEM)
[    0.010987] ACPI: XSDT 0x000000009C89D088 00008C (v01 HPQOEM SLIC-CPC 01072009 AMI  00010013)
[    0.010998] ACPI: FACP 0x000000009C8AB1D8 00010C (v05 HPQOEM SLIC-CPC 01072009 AMI  00010013)
[    0.011008] ACPI BIOS Warning (bug): Optional FADT field Pm2ControlBlock has valid Length but zero Address: 0x0000000000000000/0x1 (20200925/tbfadt-624)
[    0.011017] ACPI: DSDT 0x000000009C89D1A0 00E032 (v02 HPQOEM SLIC-CPC 00000000 INTL 20051117)
[    0.011025] ACPI: FACS 0x000000009C9AE080 000040
[... 955 lines cut ...]
[  764.116922]  vsnprintf+0x62/0x590
[  764.119738]  sprintf+0x46/0x50
[  764.122525]  proc_put_long+0x4f/0x130
[  764.125311]  ? proc_put_char.part.4+0x10/0x10
[  764.128111]  __do_proc_dointvec+0xf7/0x320
[  764.130890]  ? __kmalloc+0x3ba/0x4c0
[  764.133630]  proc_dointvec+0x22/0x30
[  764.136363]  proc_sys_call_handler+0x14d/0x220
[  764.139134]  proc_sys_read+0xb/0x10
[  764.141838]  new_sync_read+0xfc/0x170
[  764.144520]  vfs_read+0xda/0x160
[  764.147166]  ksys_read+0x4d/0xc0
[  764.149788]  __x64_sys_read+0x15/0x20
[  764.152383]  do_syscall_64+0x37/0x50
[  764.154910]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  764.157440] RIP: 0033:0x7f4f53cc9044
[... cut ...]
[  764.165075] RSP: 002b:00007ffc47df4068 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
[  764.167591] RAX: ffffffffffffffda RBX: 00007f4f53ce7ac8 RCX: 00007f4f53cc9044
[  764.170055] RDX: 0000000000002000 RSI: 00007ffc47df40c8 RDI: 0000000000000024
[  764.172477] RBP: 0000000000000000 R08: 0000000000000000 R09: 0000000000000000
[  764.174837] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000024
[  764.177132] R13: 0000000000002000 R14: 0000000000000003 R15: 00007f4f53b9f950
[  764.179371] ---[ end trace f4bd88e41fb3d305 ]---
```

minlang112, 2022-01-17:
我的T620 四核版 GX415-GA被正确识别为 Hewlett-Packard /21B4 - AMD GX-415GA SOC with Radeon(tm) HD Graphics : 4 Core 4 Thread

TioaChan, 2022-01-18:
> 我的T620 四核版 GX415-GA被正确识别为 Hewlett-Packard /21B4 - AMD GX-415GA SOC with Radeon(tm) HD Graphics : 4 Core 4 Thread
> 
> [ 0.315783] smp: Bringing up secondary CPUs ... [ 0.316175] x86: Booting SMP configuration: [ 0.316329] .... node #0, CPUs: #1 #2 #3 [ 0.318788] smp: Brought up 1 node, 4 CPUs [ 0.319108] smpboot: Max logical packages: 1 [ 0.319262] smpboot: Total of 4 processors activated (11978.04 BogoMIPS)

请问你遇到过开机时卡在”booting openwrt“的情况吗

minlang112, 2022-01-18:
> 请问你遇到过开机时卡在”booting openwrt“的情况吗

暂未发现此现象（5.10内核，U盘引导）

TioaChan, 2022-01-20:
```
root@OpenWrt:~# dmidecode |grep -A100 "Processor"
Processor Information
        Socket Designation: P0
        Type: Central Processor
        Family: G-Series
        Manufacturer: AuthenticAMD
        ID: FF FB 8B 17 01 0F 70 00
        Signature: Family 11, Model 15, Stepping 15
        Flags:
                FPU (Floating-point unit on-chip)
                CX8 (CMPXCHG8 instruction supported)
                APIC (On-chip APIC hardware supported)
                SEP (Fast system call)
                DS (Debug store)
                ACPI (ACPI supported)
        Version: AMD GX-420CA SOC with Radeon(tm) HD Graphics
        Voltage: 1.3 V
        External Clock: 100 MHz
        Max Speed: 2000 MHz
        Current Speed: 2000 MHz
        Status: Populated, Enabled
        Upgrade: None
        L1 Cache Handle: 0x002C
        L2 Cache Handle: 0x002D
        L3 Cache Handle: Not Provided
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Part Number: Not Specified
        Core Count: 1
        Core Enabled: 1
        Thread Count: 1
        Characteristics:
                64-bit capable

Handle 0x0031, DMI type 208, 5 bytes
OEM-specific Type
        Header and Data:
                D0 05 31 00 00

Handle 0x0034, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Abbreviated
        Installable Languages: 13
                enUS
                frFR
                esES
                deDE
                itIT
                daDK
                fiFI
                nlNL
                noNO
                ptPT
                svSE
                jaJP
                zhCN
        Currently Installed Language: zhCN

Handle 0x0035, DMI type 127, 4 bytes
End Of Table
```

TioaChan, 2022-01-20:
已恢复正常

closed 2022-01-20 by TioaChan