coolsnowwolf/lede #8724: 今天编译出来的固件在型号这里显示和之前不一样，为啥？

ohbaby30, 2022-01-18:
这是今天编译出来的固件显示的
型号：QEMU Standard PC (i440FX + PIIX - Intel(R) Core(TM) i3-10100T CPU @ 3.00GHz

这是之前编译出来的固件显示的
型号：Intel(R) Core(TM) i3-10100T CPU @ 3.00GHz

一样的pve环境 一样的pve虚拟机配置 固件内容没有更改过

虽然不影响使用，但是我想知道为啥，，

lxl6125, 2022-01-18:
应该是这个
https://github.com/coolsnowwolf/lede/commit/dced95e70556960363c366b8783e5695290186f2

ohbaby30, 2022-01-18:
> 应该是这个 [dced95e](https://github.com/coolsnowwolf/lede/commit/dced95e70556960363c366b8783e5695290186f2)

照着红字改回去吗？应该不影响使用的吧？

hcym, 2022-01-18:
10100应该没影响吧，我的6600t和n6005都正常识别，6005应该也是十代

ohbaby30, 2022-01-18:
> 10100应该没影响吧，我的6600t和n6005都正常识别，6005应该也是十代

识别是正常识别 只是前面多了一串QEMU什么什么的，

lxl6125, 2022-01-18:
> 

只是增加了机型识别而已，这么介意自己改回去就行了呗

ohbaby30, 2022-01-18:
> > 
> 
> 只是增加了机型识别而已，这么介意自己改回去就行了呗

QEMU Standard PC (i440FX + PIIX 这一串代码是为了增加机型识别吗？

我不是介意，只是和之前不一样了，所以想问一下，既然不影响使用那就没问题。

lxl6125, 2022-01-18:
> 

野生软路由识别成什么都不奇怪😂

lxl6125, 2022-01-18:
> > > 
> > 
> > 
> > 只是增加了机型识别而已，这么介意自己改回去就行了呗
> 
> QEMU Standard PC (i440FX + PIIX 这一串代码是为了增加机型识别吗？
> 
> 我不是介意，只是和之前不一样了，所以想问一下，既然不影响使用那就没问题。

野生软路由识别成什么都不奇怪😂

VergilGao, 2022-01-22:
> > 
> 
> 只是增加了机型识别而已，这么介意自己改回去就行了呗

我能否这样理解，这个程序可以在luci界面里显示更详细的机型信息？我目前的路由器在机型那一行显示的是default string default string

closed 2022-02-12 by Beginner-Go