coolsnowwolf/lede #8726: 5.10.92 x86-64版本,upnp使用3-4个小时会崩

guhill, 2022-01-19:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：5.10.92 x86-64版本,miniupnpd 2.0.20170421-3,使用3-4个小时会崩,本人使用网心云和甜糖的时候发现,upnp不可用
3-4个小时必然崩,再用电脑工具等测试也发现不了upnp设备,
以前内核5.9的时候没有此现象.

### (2) 路由器型号和固件版本

A：
5.10.92 x86-64版本,miniupnpd 2.0.20170421-3

### (3) 详细日志

A：崩溃以后查不到该进程和日志
崩溃之前最后的日志:
Wed Jan 19 04:07:51 2022 daemon.info miniupnpd[7610]: SOAPAction: urn:schemas-upnp-org:service:WANIPConnection:1#DeletePortMapping
Wed Jan 19 04:07:51 2022 daemon.info miniupnpd[7610]: DeletePortMapping: external port: 59853, protocol: TCP
Wed Jan 19 04:07:51 2022 daemon.info miniupnpd[7610]: Returning UPnPError 606: Action not authorized
Wed Jan 19 04:07:56 2022 daemon.notice miniupnpd[7610]: shutting down MiniUPnPd
Wed Jan 19 04:07:56 2022 daemon.notice miniupnpd[32652]: HTTP listening on port 5000
Wed Jan 19 04:07:56 2022 daemon.notice miniupnpd[32652]: Listening for NAT-PMP/PCP traffic on port 5351

Wed Jan 19 04:55:05 2022 daemon.warn miniupnpd[32652]: HTTP Connection from 192.168.2.151 closed unexpectedly

Wed Jan 19 06:08:28 2022 kern.info kernel: [20998.895745] traps: miniupnpd[32652] general protection fault ip:7f00466160eb sp:7ffc0d78da20 error:0 in libc.so[7f0046605000+49000]
================================================================
测试进程
root@OpenWrt:~# ps |grep upnp
 6345 root      1104 R    grep upnp

请大佬关注一下此问题,谢谢

minlang112, 2022-01-19:
对的，5.4内核的时候没有这个问题

swxk521, 2022-01-19:
Upnp是不是也该升级了？

Kennylaogu, 2022-01-20:
我自己试了好久才发现upnp会导致路由器断网，现在已经不开upnp了

hxlls, 2022-01-20:
[image]
[image]
[image]
我的已经运行了好久了，好像没见过UPNP崩溃，是不是你跑的东西会不断请求upnp 开端口，把端口撸满了，造成的崩溃？你试试在高级设置那定时清除规则试试

minlang112, 2022-01-22:
> 

并没有，5.4版本的时候是正常的，升到5.10之后就变成了在全锥形和其他之间隔几个小时波动N次的状态（看网心云app的状态），不知道是哪儿的问题……

minlang112, 2022-01-22:
> [image] [image] [image] 我的已经运行了好久了，好像没见过UPNP崩溃，是不是你跑的东西会不断请求upnp 开端口，把端口撸满了，造成的崩溃？你试试在高级设置那定时清除规则试试

并没有，5.4版本的时候是正常的，升到5.10之后就变成了在全锥形和其他之间隔几个小时波动N次的状态（看网心云app的状态），不知道是哪儿的问题……

moonerick, 2022-01-26:
同样的问题，自己编译了miniupnp2.2.3，全锥变映射，但是在线状态不能24小时在线，网络类型像心电图一样，群里面说是软路由的性能太弱了，J5005+8G+1T机械硬盘，现在难得管了，每天8毛到1.5元，了胜于无

guhill, 2022-01-26:
> 同样的问题，自己编译了miniupnp2.2.3，全锥变映射，但是在线状态不能24小时在线，网络类型像心电图一样，群里面说是软路由的性能太弱了，J5005+8G+1T机械硬盘，现在难得管了，每天8毛到1.5元，了胜于无

5005不弱了, 不是软路由的问题, 5.4就没这问题

ostrichii, 2022-05-06:
同样的问题，重启Upnp之后一段时就崩了，使用UPnpc 添加端口失败

guhill, 2022-05-06:
> 同样的问题，重启Upnp之后一段时就崩了，使用UPnpc 添加端口失败

没想到这么久还没人解决,我现在是用lienol的固件自编译,使用几个月了,没有此问题.

yongman, 2022-07-01:
同+1

yongman, 2022-09-30:
miniupnpd在 kernel 5.15.69 x86版本下还是一直崩。使用crontab定时检查，崩溃后自动拉起，每隔10分钟就崩。。。。更严重了
kern.info kernel: [212734.599454] traps: miniupnpd[6005] general protection fault ip:7f8389725bbe sp:7ffd71af7490 error:0 in libc.so[7f8389715000+49000]

yongman, 2022-09-30:
@guhill 你现在用的哪个版本？目前miniupnpd是否稳定？

guhill, 2022-10-01:
> 

我现在用的是5.19版本,miniupnpd已经稳定了,5.18版本也稳定,5.15不清楚目前没有在用了.

guhill, 2022-10-01:
> @guhill 你现在用的哪个版本？目前miniupnpd是否稳定？

我现在用的是lean 5.19版本,miniupnpd已经稳定了,5.18版本也稳定,5.15不清楚目前没有在用了.

yongman, 2022-10-07:
@guhill 还是原来5.15内核版本，貌似关闭bbr后miniupnpd就没有再挂了。

guhill, 2022-10-21:
> @guhill 还是原来5.15内核版本，貌似关闭bbr后miniupnpd就没有再挂了。

好的,有机会测试一下

iLmessi, 2022-12-12:
> @guhill 还是原来5.15内核版本，貌似关闭bbr后miniupnpd就没有再挂了。

还真是，我把BBR关了之后重启一下upnp qbittorrent上传就立马跑上去了

coolsnowwolf, 2026-09-05:
报告针对 2022 年 miniupnpd 2.0.20170421；当前代码树已不再保留该旧源码/版本，原 libc 崩溃问题不对应当前包，关闭为 completed。

closed 2026-09-05 by coolsnowwolf