coolsnowwolf/lede #8727: davfs2 编译失败

zhangguanzhang, 2022-01-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：1

```
CONFIG_PACKAGE_davfs2=y
```

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：1

```
(cd /workdir/openwrt/build_dir/target-x86_64_musl/davfs2-1.6.0/./; if [ -x ./configure ]; then find /workdir/openwrt/build_dir/target-x86_64_musl/davfs2-1.6.0/ -name config.guess | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-x86_64_musl/davfs2-1.6.0/ -name config.guess | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.guess; find /workdir/openwrt/build_dir/target-x86_64_musl/davfs2-1.6.0/ -name config.sub | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-x86_64_musl/davfs2-1.6.0/ -name config.sub | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/davfs2-1.6.0=davfs2-1.6.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/davfs2-1.6.0=davfs2-1.6.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/davfs2-1.6.0=davfs2-1.6.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --with-neon="/workdir/openwrt/staging_dir/target-x86_64_musl/usr" --disable-rpath ; fi; )
configure: loading site script /workdir/openwrt/include/site/x86_64
checking for a BSD-compatible install... /workdir/openwrt/staging_dir/host/bin/install -c
checking whether build environment is sane... yes
checking for x86_64-openwrt-linux-strip... x86_64-openwrt-linux-musl-strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for x86_64-openwrt-linux-gcc... x86_64-openwrt-linux-musl-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether x86_64-openwrt-linux-musl-gcc accepts -g... yes
checking for x86_64-openwrt-linux-musl-gcc option to accept ISO C89... none needed
checking whether x86_64-openwrt-linux-musl-gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of x86_64-openwrt-linux-musl-gcc... gcc3
checking whether ln -s works... yes
checking whether NLS is requested... no
checking for msgfmt... /usr/bin/msgfmt
checking for gmsgfmt... /usr/bin/msgfmt
checking for xgettext... /usr/bin/xgettext
checking for msgmerge... /usr/bin/msgmerge
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-openwrt-linux-gnu
checking for ld used by x86_64-openwrt-linux-musl-gcc... x86_64-openwrt-linux-musl-ld
checking if the linker (x86_64-openwrt-linux-musl-ld) is GNU ld... yes
checking for shared library run path origin... done
checking how to run the C preprocessor... x86_64-openwrt-linux-musl-gcc -E
checking for grep that handles long lines and -e... /workdir/openwrt/staging_dir/host/bin/grep
checking for egrep... /workdir/openwrt/staging_dir/host/bin/grep -E
checking for CFPreferencesCopyAppValue... no
checking for CFLocaleCopyCurrent... no
checking whether to use NLS... no
checking for neon library in /workdir/openwrt/staging_dir/target-x86_64_musl/usr... found
checking linking against neon... yes
configure: incompatible neon library version 0.32.1: wanted 0.27 28 29 30 31
configure: error: could not find neon
make[3]: *** [Makefile:76: /workdir/openwrt/build_dir/target-x86_64_musl/davfs2-1.6.0/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/davfs2'
time: package/feeds/packages/davfs2/compile#3.43#0.50#4.79
    ERROR: package/feeds/packages/davfs2 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/davfs2/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
```

WYC-2020, 2022-01-19:
error: could not find neon

zhangguanzhang, 2022-01-19:
> error: could not find neon

look, human txt

Beginner-Go, 2022-01-19:
更新下feeds 再编译试试

zhangguanzhang, 2022-01-19:
> 更新下feeds 再编译试试

action里编译的，拉的master代码，今天上午11点的时候开始编译的

zhangguanzhang, 2022-01-19:
看到你的commit了，我试试

zhangguanzhang, 2022-01-20:
@Beginner-Go 昨天是arm64的固件开了davfs2编译出来的没有，然后就编译x86_64的固件，结果编译出错，在大佬更新了feeds后试了下能编译成功，并且在arm64 编译的也带davfs2了

closed 2022-01-20 by zhangguanzhang