coolsnowwolf/lede #8728: 防火墙显示“500 内部服务器错误”

xydche, 2022-01-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：网络--防火墙--常规设置，下面区域<->转发那里显示如下，感觉一星期前编译的版本也是这样的，使用上没感觉有什么异常，不知道如何让它恢复正常，求助
```
500 内部服务器错误
对不起，服务器遇到未知错误。

/usr/lib/lua/luci/dispatcher.lua:474: Failed to execute function dispatcher target for entry '/admin/network/firewall'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:474: Failed to execute arcombine dispatcher target for entry '/admin/network/firewall/zones'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/tblsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/firewall_zoneforwards'.
A runtime error occured: /usr/lib/lua/luci/model/firewall.lua:21: bad argument #2 to 'get' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:474: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:126: in function 
```

### (2) 路由器型号和固件版本

A：OpenWrt R22.1.1 / LuCI Master (git-21.335.48743-5f363d9)

### (3) 详细日志

A：
```
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: /usr/lib/lua/luci/dispatcher.lua:474: Failed to execute function dispatcher target for entry '/admin/network/firewall'.
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: The called action terminated with an exception:
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: /usr/lib/lua/luci/dispatcher.lua:474: Failed to execute arcombine dispatcher target for entry '/admin/network/firewall/zones'.
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: The called action terminated with an exception:
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/tblsection'.
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/firewall_zoneforwards'.
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: A runtime error occured: /usr/lib/lua/luci/model/firewall.lua:21: bad argument #2 to 'get' (string expected, got nil)
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: stack traceback:
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: 	[C]: in function 'assert'
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: 	/usr/lib/lua/luci/dispatcher.lua:474: in function 'dispatch'
Wed Jan 19 19:40:34 2022 daemon.err uhttpd[4471]: 	/usr/lib/lua/luci/dispatcher.lua:126: in function </usr/lib/lua/luci/dispatcher.lua:125>
```

xydche, 2022-01-20:
搞定了
firewall配置config zone 缺少了vpn，手动加了一行option name vpn 解决了。我记得之前防火墙这里是有VPN配置的，ipsec还在，只是最近几次编译把openvpn server取消了，可能是影响到firewall配置了

closed 2022-01-20 by xydche

xydche, 2022-01-20:
应该是大小写的问题，刚才仔细看了一下firewall配置里有 “VPN” zone，没有'vpn' zone，而下面好多规则里有的是大写，有的是小写。

xydche, 2022-01-24:
取消勾选openvpn-server后，配置还在firewall的config里，导致的

SudSuo, 2022-02-15:
请问这个是在哪里改配置，我编译的时候没有选 openvpn-server，也是出现了这样的提示

SudSuo, 2022-02-15:
非常感谢，终于解决了相同的问题

leonleexin, 2022-08-05:
大神，我也遇到相同问题了，能详细说说怎么解决么？

leonleexin, 2022-08-05:
能详细说说怎么处理，你说的firewall配置的config zone在哪啊？

SudSuo, 2022-08-05:
[image]
用Winscp连接路由器，进配置文件更改

leonleexin, 2022-08-05:
谢谢。
是加一行还是把这行删除掉呢，大大

leonleexin, 2022-08-05:
> [image] 用Winscp连接路由器，进配置文件更改

leonleexin, 2022-08-05:
搞定，我把带vpn的全删除就好了

liudaoguiguzi, 2022-12-15:
多谢指教，终于把这个有问题的配置删掉了