coolsnowwolf/lede #8729: IPQ60XX系统没有Gl.iNet AX1800的配置

LarkinZero, 2022-01-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新master分支代码，make menuconfig时，没有没有Gl.iNet AX1800的配置

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

[image]
[image]

1054009064, 2022-01-19:
还没稳定，暂时是为了编译相关插件，大佬们应该快搞好了

TurboBlaze, 2022-01-20:
@LarkinZero 需要更正。 阅读这些主题： 
https://github.com/coolsnowwolf/lede/commit/a32407cdaf218ff9f55bd41f6b1b4667740e55f4
https://github.com/coolsnowwolf/lede/issues/8659#issuecomment-1014592564
https://forum.gl-inet.com/t/gl-inet-ax1800-openwrt-firmware/19447/6

closed 2022-01-23 by AmadeusGhost