coolsnowwolf/lede #8732: MT7621的NAT可以跑满千兆吗？

MageDelfador, 2022-01-20:
联通千兆宽带，路由器K2P(A1)测试如下：
1.光猫设置路由模式，路由器仅作为交换机使用，电脑网线连接，测速下载带宽900M以上
2.光猫设置桥接模式，路由器pppoe拨号，打开Flow Offloading和hwnat，电脑网线连接，测速下载带宽500M左右。路由器CPU使用3%
3.光猫设置桥接模式，路由器pppoe拨号，关闭Flow Offloading和hwnat，电脑网线连接，测速下载带宽300M左右。路由器CPU使用97%
请问各位大佬，MT7621这个速率正常吗？是否还有提升空间？

ZqinKing, 2022-01-20:
正常，光猫桥接后限速了，只能跑500不是路由器的原因。

MageDelfador, 2022-01-20:
还有一步测试忘了说了
4.光猫设置路由模式，光猫网段192.168.1.0/24，路由器LAN网段192.168.2.0/24，使路由器走NAT，打开Flow Offloading和hwnat，电脑网线连接，测速下载带宽仍然是500M左右。
之前看到网上有人说桥接限速，所以专门测试了一下，确认了桥接没有限速

zhanddkk, 2022-01-20:
我用Openwrt 21.02 ipv4模式下，开启HW Flow Offloading可以再MT7621上跑满千兆，下载945Mbit/s，上传930Mbit/s。CPU占用率很低
这个lede在我的开发板(HLK-7621A)上无法运行，应该是我的bootloader太老了。

测试方式：
主路由是J4125软路由，开启iperf3服务。再接开发板再由开发板的LAN接Win11 PC.

MageDelfador, 2022-01-20:
刚刚用iperf3测试了一下：
5.路由器LAN口接一台电脑，WAN口接另外一台电脑，WAN网段192.168.1.1/24，LAN网段192.168.2.1/24，简单改下防火墙后使用iperf3测试，结果两边都是940M左右。关闭Flow Offloading和hwnat，速率变成310M左右。
这个结果跟第四步测试的结果冲突，所以重新测试了一下第四步，结果800M左右。猜测原因是测速节点不稳定，当时测第四步跟前三步不在同一时间。
所以结论是光猫桥接确实限速，路由器NAT可以跑满。
各位大佬有好办法解决桥接限速问题吗？开路由模式感觉会受到很多限制。

WYC-2020, 2022-01-20:
现在光猫都是软件桥接，不像以前的猫直接硬件转发，如果运营商猫没有绑定，自己换猫吧，或者换找下看有没光电转换这种

MageDelfador, 2022-01-20:
> 现在光猫都是软件桥接，不像以前的猫直接硬件转发，如果运营商猫没有绑定，自己换猫吧，或者换找下看有没光电转换这种

有推荐的用来桥接的光猫型号吗

zhanddkk, 2022-01-23:
我同事把光猫改桥接，依旧可以千兆跑满。光猫是电信自带的SDN智能网关

closed 2022-02-12 by Beginner-Go