coolsnowwolf/lede #8735: 动态dns没有ipv6

sungamma, 2022-01-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
根据P3TERX/Actions-OpenWrt教程正常编译，但是服务里面的动态dns，不能选择ipv6，出现不了ipv6这个选项
### (1) 具体问题

A：动态dns，不能选择ipv6，出现不了ipv6这个选项。即使手动改配置为ipv6，luci界面会提示ipv6不支持

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：

WYC-2020, 2022-01-23:
[image]
 是这个？

sungamma, 2022-01-23:
是的，就是这个地方没有显示ipv6地址这个选项。自己编译了很多次，都没有。网络上找了别人的也发现没有。
[image]
我编译的参数以及固件地址如下:https://github.com/sungamma/lean_OpenWrt/releases/tag/2022.01.22-2039
。

coolsnowwolf, 2026-09-06:
当前 `luci-app-ddns` 已提供 IPv6 地址选项，并根据系统 IPv6 能力显示 `IPv6-Address`；旧版界面缺少 IPv6 选择的问题已解决，关闭为“已完成”。

closed 2026-09-06 by coolsnowwolf