coolsnowwolf/lede #8736: luci-app-verysync 不能运行（暂时有解决办法）

icecubi, 2022-01-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
A：luci-app-verysync 不能运行 verysync， 进过研究发现stop() {	kill -9 verysync >/dev/null 2>&1} 把脚本进程也kill了，无法继续执行脚本，导致启动失败(很奇怪，2022.1.12之前的代码却可以运行)
解决办法：把启动脚本/init.d/verysync 名字改为 verysync_init 脚本里停止verysync进程 改为killall -9 verysync >/dev/null 2>&1 解决

### (2) 路由器型号和固件版本

A： x86

### (3) 详细日志

A：

WYC-2020, 2022-01-23:
是因为他们名字冲突了，脚本自己吧自己杀死了，后面的启动都走不了

WYC-2020, 2022-01-23:
你这样改了 luci 界面上是不能控制的

WYC-2020, 2022-01-23:
https://github.com/WYC-2020/lede/commit/8ea8089584e9a2cbe1f45b2466a22567d67a0357 
我这边这样修改的 @icecubi

icecubi, 2022-01-24:
异曲同工吧，我改启动脚本名称，你改服务名称，你那个方法不容易出问题些

WYC-2020, 2022-01-24:
只改启动脚本名称，应该是不行的，luci上面应该控制不了，你还要改luci-default你可以试下，我按你这个方法改了界面上没法控制

shanhai2333, 2022-01-27:
照着大佬的改了，还是起不来

WYC-2020, 2022-01-27:
应该是你那里没对，我那个方法我是验证过得，没问题，建议不保留配置

shanhai2333, 2022-01-28:
> 应该是你那里没对，我那个方法我是验证过得，没问题，建议不保留配置

找到原因了，，我改github编译源那个url，看日志没改成功。

WYC-2020, 2022-02-10:
@xixijojo https://github.com/WYC-2020/lede/commit/f9ecf0529073904a8468efaa0d0f20ee95a45c72

unifreq, 2022-03-11:
经测试把/etc/init.d/verysync里的 verysync加上绝对路径就可以了。
#!/bin/sh /etc/rc.common
  
START=50
STOP=10

VERYSYNC="/usr/bin/verysync"

start() {
        local enabled="$(uci get verysync.config.enabled)"
        local port="$(uci get verysync.config.port)"
        local profile="$(uci get verysync.config.profile)"
        stop
        [ "${enabled}" == "1" ] || exit 0
        export HOME="/root/"
        "${VERYSYNC}" -gui-address="0.0.0.0:${port}" -logfile="/var/log/verysync.log" -home="${profile}" -no-browser >/dev/null 2>&1 &
}

stop() {
        kill -9 `pgrep "${VERYSYNC}"` >/dev/null 2>&1
}

coolsnowwolf, 2026-09-05:
报告针对旧版 luci-app-verysync 启停脚本，当前 feeds 中已不再提供该 LuCI 应用实现，无法在现行代码中修复，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf